mcguirepr89/BirdNET-Pi #2: add sensitivity to parameters

DD4WH, 2021-10-01:
Hi Patrick,

not sure about this, but my impression is, that the BirdNET-Lite system is started without setting the "sensitivity" parameter, which I think is a quite important one for the user to set.

So, I propose adding this to the config file.

I made a test with adding the SENSITIVITY variable to the birdnet_analysis.sh and according to the log it seems to work fine:

  SENSITIVITY=1.25	
  cd ${HOME}/BirdNET-Lite || exit 1
  for i in "${files[@]}";do
    if [ -f ${1}/${i} ] && [ ! -f ${CUSTOM_LIST} ];then
      set -x
      python3 analyze.py \
        --i "${1}/${i}" \
        --o "${1}/${i}.csv" \
        --lat "${LATITUDE}" \
        --lon "${LONGITUDE}" \
        --week "${WEEK}" \
        --overlap "${OVERLAP}" \
        --min_conf "${CONFIDENCE}" \
	--sensitivity "${SENSITIVITY}"

mcguirepr89, 2021-10-01:
I'm working on it right now :)

mcguirepr89, 2021-10-01:
Done :) Please test and let me know when you can if it is doing the trick. I set it to the default you suggested.

mcguirepr89, 2021-10-02:
Tested as working -- closing this issue

closed 2021-10-02 by mcguirepr89