mcguirepr89/BirdNET-Pi #3: Calculation of week

DD4WH, 2021-10-02:
it seems that BirdNET-Lite has a different logic of calculating the "week"
* For us, a year has 52 weeks.
* For BirdNET-Lite, a year has 48 weeks (4 weeks a month)
"--week, Week of the year when the recording was made. Values in [1, 48] (4 weeks per month)." https://github.com/kahst/BirdNET-Lite

I would propose to calculate "week" as follows:
* Jan day 1-7 --> week 1
* Jan day 8-14 --> week 2
* Jan day 15-21 --> week 3
* Jan day 21-x --> week 4 
* Feb day 1-7 --> week 5
* etc.

mcguirepr89, 2021-10-02:
Very good observation -- I had wondered why the python script only accepted week values between 1 and 48.
I will work on this and will let you know when it is testable.
Thank you as always!

mcguirepr89, 2021-10-02:
Hello again -- I have made changes for testing. This type of testing will be kind of long-term, since we'll need to keep track of the weeks as they change to determine that the new WEEK calculations are correct. I will explain how it is calculated below so that you can hopefully spot any flaws in my logic that would lead to these values being incorrect.

```
  WEEK_OF_YEAR="$(echo "($(date +%m)-1) * 4" | bc -l)"

  DAY_OF_MONTH="$(date +%d)"

  if [ ${DAY_OF_MONTH} -le 7 ];then
    WEEK="$(echo "${WEEK_OF_YEAR} + 1" |bc -l)"

  elif [ ${DAY_OF_MONTH} -le 14 ];then
    WEEK="$(echo "${WEEK_OF_YEAR} + 2" |bc -l)"

  elif [ ${DAY_OF_MONTH} -le 21 ];then
    WEEK="$(echo "${WEEK_OF_YEAR} + 3" |bc -l)"

  elif [ ${DAY_OF_MONTH} -ge 22 ];then
    WEEK="$(echo "${WEEK_OF_YEAR} + 4" |bc -l)"

  fi
```

To break that down with today (October 2nd, 2021) as the example, it does the following:
1. It first calculates a base _WEEK_OF_THE_YEAR_  value. That is calculated by finding out how many weeks have already passed in the year. It does this by getting the current month's decimal value and subtracting 1. So, using the example, this calculation gets a decimal value for the current month (October=10) and subtracts 1 (now 9). So that means that there have been 9 full months that have passed, with 4 weeks in each, so the value is multiplied by 4 to get the total weeks so far, 36.

2. It then creates a value for the _DAY_OF_THE_MONTH_. This value will be used in the if statements directly below it to test whether the current day of the month indicates that more weeks have passed and need to be added to the base _WEEK_OF_THE_YEAR_ value.

3. If the day of the month is 7 or less, 1 is added to the base _WEEK_OF_THE_YEAR_ value. If that check fails, it will see if the value is less than or equal to 14, for which it will add 2 weeks to the base _WEEK_OF_THE_YEAR_ value. If that check fails, it will see if the value is less than or equal to 21, for which it will add 3 weeks to the base _WEEK_OF_THE_YEAR_ value. And finally, if those checks all fail, it will ensure that the date is greater than or equal to 22, and will add 4 weeks to the base _WEEK_OF_THE_YEAR_ value.

I know that's a little stringy to follow, but hopefully you can follow and let me know if that seems to make sense or if I'm on the wrong track.

Thanks!

DD4WH, 2021-10-04:
Hi Patrick, excellent! 
With your explanation, the code is crystal-clear and exactly what it should be for the purpose of BirdNET-Lite!
Also, in the current rpialpha of 4th October, the week is calculated correctly!
Thanks a lot for this!
From my point of view, this issue is solved.

closed 2021-10-04 by mcguirepr89