mcguirepr89/BirdNET-Pi #32: Pushed.co notifications not working

mcguirepr89, 2021-10-15:
@Christoph-Lauer has noticed that the pushed co is sending messages every three minutes -- sounds like a bug

mcguirepr89, 2021-10-15:
I agree

Christoph-Lauer, 2021-10-15:
Got the same error again also for a clean installation.

mcguirepr89, 2021-10-15:
I'm testing now so I'll update you when I have this working

mcguirepr89, 2021-10-16:
Looks like the issue was with how the crontab handles the script. I put it into a systemd service that restarts every 3 seconds on-success and it is working like a charm so far. No duplicates, no empty notifications, and quicker, more responsive notification timing.

I will push this as an update that you can test with the new update button soon, probably tomorrow when I get up. I'll let you know after more testing when to do an update. Thanks for using this feature so it doesn't die!

Christoph-Lauer, 2021-10-16:
the update went well and the messaging service runns no correct despite the fact that i one time get a message with 3 or more species. Still continue testing. 

Greets from the german ccc --> https://www.youtube.com/watch?v=_m3Y-u0lKkQ

mcguirepr89, 2021-10-16:
systemd service still has issues, but I came up with a solution in my sleep last night that I think will work very very well. I will update you when I have the new idea in place.

Christoph-Lauer, 2021-10-16:
Yes it stoped sending messages after reboot and clear. Rest is working fine. Got my outdoor mik setup and cannot see what flying around here.

mcguirepr89, 2021-10-16:
did you have any nomachine issues on the clean install?

Christoph-Lauer, 2021-10-16:
Nomachine works

mcguirepr89, 2021-10-16:
Great thank you for checking -- if you need to disable the species notifier
while I'm working on it, you can either remove your pushed.co creds from
the birdnet.conf or issue
sudo systemctl disable --now pushed_notifications

mcguirepr89, 2021-10-16:
> Scripting god must fixit again.

Yeah :/

mcguirepr89, 2021-10-16:
This is working now -- I've completely rewritten both the update_species.sh and species_notifier.sh to read from the database. It is working and I feel very confident it will work for you without issue.
It is available as an update -- give the update about 4-5 minutes to complete, then you can test and let me know what you think.

mcguirepr89, 2021-10-16:
Sorry I made those comments before pushing the changes --they've been pushed now though

Christoph-Lauer, 2021-10-16:
got this message from the journalctl:

```
pi@raspberrypi:~/BirdNET-Pi $ journalctl -fu pushed\*
-- Logs begin at Sat 2021-10-16 12:45:18 CEST. --
Okt 16 18:34:37 raspberrypi systemd[1]: Stopped BirdNET-Pi Pushed.co Notifications.
Okt 16 18:34:37 raspberrypi systemd[1]: Started BirdNET-Pi Pushed.co Notifications.
Okt 16 18:34:37 raspberrypi species_notifier.sh[20100]: /usr/local/bin/species_notifier.sh: Zeile 13: /home/pi/BirdNET-Pi/scripts/new_update_species.sh: Datei oder Verzeichnis nicht gefunden
Okt 16 18:34:37 raspberrypi systemd[1]: pushed_notifications.service: Succeeded.
Okt 16 18:34:40 raspberrypi systemd[1]: pushed_notifications.service: Service RestartSec=3s expired, scheduling restart.
Okt 16 18:34:40 raspberrypi systemd[1]: pushed_notifications.service: Scheduled restart job, restart counter is at 209.
Okt 16 18:34:40 raspberrypi systemd[1]: Stopped BirdNET-Pi Pushed.co Notifications.
Okt 16 18:34:40 raspberrypi systemd[1]: Started BirdNET-Pi Pushed.co Notifications.
Okt 16 18:34:40 raspberrypi species_notifier.sh[20116]: /usr/local/bin/species_notifier.sh: Zeile 13: /home/pi/BirdNET-Pi/scripts/new_update_species.sh: Datei oder Verzeichnis nicht gefunden
Okt 16 18:34:40 raspberrypi systemd[1]: pushed_notifications.service: Succeeded.
Okt 16 18:34:43 raspberrypi systemd[1]: pushed_notifications.service: Service RestartSec=3s expired, scheduling restart.
Okt 16 18:34:43 raspberrypi systemd[1]: pushed_notifications.service: Scheduled restart job, restart counter is at 210.
Okt 16 18:34:43 raspberrypi systemd[1]: Stopped BirdNET-Pi Pushed.co Notifications.
Okt 16 18:34:43 raspberrypi systemd[1]: Started BirdNET-Pi Pushed.co Notifications.
Okt 16 18:34:43 raspberrypi species_notifier.sh[20147]: /usr/local/bin/species_notifier.sh: Zeile 13: /home/pi/BirdNET-Pi/scripts/new_update_species.sh: Datei oder Verzeichnis nicht gefunden
Okt 16 18:34:43 raspberrypi systemd[1]: pushed_notifications.service: Succeeded.
^C
pi@raspberrypi:~/BirdNET-Pi $ 
```

Christoph-Lauer, 2021-10-16:
LOL

Christoph-Lauer, 2021-10-16:
Ok, works, got messages. Make an test with the iPhone.

mcguirepr89, 2021-10-16:
phew! thank you for hanging in there. I'm glad the update feature is working as it is very helpful for someone like me who has to fix my mistakes often