mcguirepr89/BirdNET-Pi #36: BirdNet-Pi 0.2 Update Issue

CaiusX, 2021-10-16:
Hi @mcguirepr89 

Trying the 0.2 update script I get the attached in the console,

stops after OS update with:

error: pathspec 'testing' did not match any file(s) known to git.

Tried again, same error.

Will send the log files by e-mail.

Best
[BirdNetPi_0.2_install.txt](https://github.com/mcguirepr89/BirdNET-Pi/files/7357114/BirdNetPi_0.2_install.txt)

CaiusX, 2021-10-16:
Hi @mcguirepr89 

Update runs through fine.

After update-

1. :Recognized Birds link still shows the old stats page
2. Database seems to be a new empty one, starts to fill up after a while.  Best to keep the old one and append new readings? 
3. BirdsDB.txt csv file remains intact and continues getting written to.
4. Up and running on birds.naturestation.net easily.

I'll send the install log by mail.

Best 
Laconic African Penguin :)

mcguirepr89, 2021-10-16:
Thanks, L.A.P.!

I think the issue is that I haven't gotten the update working for the main branch yet -- I'm going to test it out thoroughly in the testing branch first. It seems that the updater didn't want to proceed because you were "checked out" on the main branch. That was a failure in my documentation for my plan on how to use, test, and ultimately deploy the update button. I'm sending you an email now so that I might be able to take a deeper look!