mcguirepr89/BirdNET-Pi #66: Cat it run on RPi 3?

joknjokn, 2021-11-06:
Hi,

I'm wondering if you think this could run on RPi 3? If so, would it require difficult changes?

Thanks!

mcguirepr89, 2021-11-06:
Hello! Thanks for your interest! I'm sorry I cannot answer this with any real certainty since I've not been able to test on a RPi3 myself yet and do not know.

I do _think_ BirdNET-Pi would run on the RPi 3B+. So long as the ARM64 version of RaspiOS is the operating system, the TFLite pre-built binaries aught to run, which is the only part that might fuss.

You _will_ want to enable the zRAM swapping to compensate for the 3B+'s small memory. There is a tool for this, `birdnet-pi-config` available in the _newinstaller_ branch under system settings. I was actually considering pulling that branch into the main branch now that I have tested it so thoroughly in the last few days, so that tool will likely be available in the _main_ branch soon.

See the screen shots below for how to use the tool on the _newinstaller_ branch:
In the Raspberry Pi's terminal (after installing BirdNET-Pi), OR from an SSH terminal session with the BirdNET-Pi, issue `birdnet-pi-config`:
[image]
Use your arrow keys to select "System Options"
[image]
Then select "Configure zRAM"
[image]
Choose yes
[image]
I recommend using 2G for the RPi3
[image]
Then you'll see this
[image]

You'll have to reboot for the swap partition to be available.

Please git it a try, OR you are welcome to test the new pre-installed image [HERE](https://github.com/mcguirepr89/BirdNET-Pi/discussions/63) to save a little time. (You'll still want to do the zRAM options above if you choose this method.)

Please let me know if you encounter any trouble or have any other questions and I will be more than happy to assist you!

mcguirepr89, 2021-11-08:
Hello, @joknjokn -- I hope this update finds you well.

We have been working on this for you and tests are looking quite promising, so this is the update:
- **YES** BirdNET-Pi _does_ run on the Raspberry Pi 3B
- It **DOES** require the zRAM swaping to be configured
- It would likely benefit from disabling the Live Audio Stream, and/or disabling the desktop altogether so that BirdNET-Pi only boots into the command line.
- Adequate cooling is a must!! 
- Once you get a reliable method of cooling, you can experiment with overclocking the arm, which could yield results comparable to the RPi4

I am more than happy to help out with any of the above suggestions, simply let me know where you'd like guidance and I will provide what I can!

My best regards,
Patrick

PS 
I am keeping this issue open while I tinker with things a bit more and will provide any updates here that could be helpful.

joknjokn, 2021-11-08:
@mcguirepr89 , thank you so much! I hope to get time for it soon, but since I can't find my RPi3-fan (ordering new one), I can't try yet. If the RPi3 gives me too much trouble, I'll probably order a RPi4.

Also, I was actually inspired by your project and managed to make a Flutter-app (Android only at the moment) that uses the BirdNet-model and also runs "live" and collects the results,  without saving recordings though. Kind of like a portable mini-version of yours. Not nearly done, and I'm unsure if it will be, but the live-BirdNet functionality is really fun!

mcguirepr89, 2021-11-08:
> Also, I was actually inspired by your project and managed to make a Flutter-app (Android only at the moment) that uses the BirdNet-model and also runs "live" and collects the results, without saving [...]

That's awesome! I don't have any Android devices (as much as I love me some Linux), otherwise I would be eager to check this out. Please feel invited to share that project on this page's Discussions if you'd like -- I think it sounds very cool. 

I actually have been able to successfully integrate a live detection stream from the [BMBF InnoTruck demonstration source code](https://github.com/kahst/BirdNET-Demo) that can utilize the same TFLite pre-built binaries that BirdNET-Pi uses for BirdNET-Lite. The full stack offers a live audio stream, a live spectorgram, and photos, but I've only been able to get the live detections and photos to work properly. [Please take a look here](https://github.com/mcguirepr89/BirdNET-Pi/discussions/24) in hopes that you might be able to help out with getting this firmly integrated into BirdNET-Pi, for which I would be very grateful!!!

My best regards

mcguirepr89, 2021-11-12:
Hello again -- I haven't had the time to work on this, so please let me know how it goes for you. I'm happy to help you figure out how to get the resource requests to match up to what the RPi3 can offer, but I didn't want you to expect more updates from me at this moment as I'm focused elsewhere for the time being.

My best,
Patrick

joknjokn, 2021-11-12:
Sorry, forgot to reply. 

The flutter/android-project is too messy to share at the moment. I had to mess up a third tflite-plugin to get it to work.

Regarding BirdNet-PiRPi3, it's more of a "long term" project for me, with slow progress. I think my next step is to see if I can grab a cheap RPi4 at black friday, before I spend too much time experimenting. But thanks a lot for the help - I'm sure I'm not the only one interested in getting it to work on RPi3 also. Really appreciate the help :)

mcguirepr89, 2021-11-12:
It may be worth considering [the new Raspberry Pi 2 Zero W ](https://www.raspberrypi.com/products/raspberry-pi-zero-2-w/)-- it has a comparable ARM processor to the RPi3B with less power-consumption and MUCH cheaper. If I could get BirdNET-Pi running on one of those, then we would be in pretty exciting territory being able to edge compute deep neural networks on a $15 computer!!

I'll certainly keep you posted on any headway I make for either device.
Please accept my best regards -- Patrick

joknjokn, 2021-11-12:
That's interesting. Also probably better for portability with powerbank+solar, if that's ever going to happen!

mcguirepr89, 2021-11-12:
> Also probably better for portability with powerbank+solar, if that's ever going to happen!

I surely hope someone has the resources and know-how to put that together -- it would be really exciting to Frankenstein a BirdNET-Pi that could live forever in the woods (like Frankenstein's real monster). :zombie:

gitalm, 2021-11-25:
HI, thanks for the hints. I tried to install without desktop and everything went fine. I also activated zram. But i cannot connect to http://birdnet.local 
I can connect via ssh from the same PC, but also the IP adress in the browser isn't showing something.

Thanks for the help

gitalm, 2021-11-25:
It is obviously working, but i cannot access the webpage 
[image]

mcguirepr89, 2021-11-26:
Hello, @gitalm and thank you so much for your interest in BirdNET-Pi. Sorry I wasn't able to get back to you yesterday, but it was Thanksgiving here in the US and I had a lot going on.

Running on a RPi3 is somewhat tricky due to resource constraints, but I'm happy to work with you on getting things working as best as possible.

To begin, I wanted to be sure you're attempting to access the correct web page. Your comment mentions not being able to reach the web interface at "http://birdnet.local," but the web interfaces is actually at http://birdnetpi.local -- notice the 'pi' tucked in between _.local_ and _birdnet_.

Could you check for me to see if you are able to access the web interface at http://birdnetpi.local?

Thank you and I will be around today to help out as much as I can -- 
My best,
Patrick

mcguirepr89, 2021-11-27:
Hi, @gitalm -- I just wanted to follow up with you to see if you've had an opportunity to revisit the web interface at http://birdnetpi.local with any success?

Wishing you a great weekend,
Patrick

gitalm, 2021-11-27:
Dear Patrick 

it is working. I think i a problem with my USB Microphone. Is a microphone for the 3.5mm Jack not applicable?

Thanks for helping

But it is not processing any files. See attached screenshoot. What ca  i do?
[image]

[image]

mcguirepr89, 2021-11-27:
Could you show a screenshot of the "BirdNET-Lite Log" under "System Links"? That will likely tell a bit more about what is going on.

With that said, I believe the problem will be due no audio data input. I suspect that the spectrograms are black? Please confirm this for me. This is because the 3.5mm audio jack that is built into the pi does not support audio input. 

What I would recommend to see what is going on are the following, which I will describe better below.
1. Temporarily stopping the `birdnet_recording.service`
1. Removing all of the recordings your system has created so far 
1. Grabbing some extracted sound files from @CaiusX's [installation](https://birds.naturestation.net/) to populate today's directory on your system **OR** Plug in a USB audio input device
1. Watching the "BirdNET-Lite Log" found under "System Links" on http://birdnetpi.local
1. Reporting results here

To **Temporarily stop the `birdnet_recording.service`**, issue:
```
sudo systemctl stop birdnet_recording.service
```
To **Remove all of the recordings your system has created so far**, issue:
```
rm -rfv ~/BirdSongs/November-2021/27-Samstag/*
```
To ***Grab some extracted sound files from @CaiusX's [installation](https://birds.naturestation.net/) to populate today's directory on your system**, find a sound from a bird that has a very high confidence score (they're embedded in the file name, so you can sort by file name to get a file with a high confidence score) and right-click the file to save it. Place it in the `/home/pi/BirdSongs/November-2021/27-Samstag/`

*Note: @CaiusX's installation is set for very different latitude and longitude than you are likely to have, so your system might not have the best chance of making a detection with exotic sounding birds. _If_ using one of those files does not lead to a detection, we should still see it at least get processed successfully, so the test is not worthless.

**Note also that the RPi3 cannot keep up with realtime running all services, so you will still see files accumulate where the same installation will have no more than 2-3 files at a time on an RPi4.

I'll be out for the rest of the day, but I will certainly respond as soon as I have the chance.

gitalm, 2021-11-27:
Sorry for the delay. Sometimes the RPi3 Server is not accesible 

Seems like Tensorflow has a problem

[image]

gitalm, 2021-11-27:
I installed Tensorflow for Python 3.9 with the following command,
`python -m pip install --upgrade https://storage.googleapis.com/tensorflow/mac/cpu/tensorflow-1.12.0-py3-none-any.whl`
and then the next issue appeared
[image]
[image]

[image]

I have also some issues about analyze.py
First mysql.connector was missing
`pip install mysql.connector`
[image]

But i can load the model
[image]

mcguirepr89, 2021-11-27:
Hi, @gitalm. I can shed a lot of light on what you are seeing and hopefully you'll have a better understanding of how BirdNET-Pi is put together.

Before anything, though, I will need you to verify with no uncertainly that you are using the ARM64 version of RaspiOS built on Debian Bullseye that can only be found [here](https://downloads.raspberrypi.org/raspios_arm64/images/). Please don't continue troubleshooting until you can be sure that you are using the 64-bit version of the OS.

>I installed Tensorflow for Python 3.9 with the following command,
python -m pip install --upgrade https://storage.googleapis.com/tensorflow/mac/cpu/tensorflow-1.12.0-py3-none-any.whl
and then the next issue appeared

This method, although it does not fail, does not install the site packages where BirdNET-Pi needs them to be. BirdNET-Pi's services utilize a python venv that is specifically built for the services. Also, that version of Tensorflow is not what is needed here and will not work. You will need to use the pre-built TFLite binaries that are included in the wheels in the BirdNET-Pi directory:

With that in mind, you can understand why your later interactions with the OS's site-packages seemed to indicate that those site-packages are not installed on the system . . . 
>I have also some issues about analyze.py
First mysql.connector was missing
pip install mysql.connector
But i can load the model . . .

They are installed, but need to be accessed from within the 'birdnet' python venv located in the BirdNET-Pi directory. To _activate_ the appropriate python virtual environment, you will need to issue `source ~/BirdNET-Pi/birdnet/bin/activate`.

From that venv prompt, your `pip` interactions will actually be affection the correct site-packages. Issue `pip list` to see the site-packages that BirdNET-Pi will actually be using.

Before we troubleshoot too much more, could you please relay to me your installation procedure?

My best

gitalm, 2021-11-28:
Thanks for all your help
i followed the instructions from [Headless](https://github.com/mcguirepr89/BirdNET-Pi/wiki/%22Headless%22-installation-using-VNC) installtion, but didn't avtivated the VNC and used ssh instead. I installed a Lite version from https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2021-11-08/2021-10-30-raspios-bullseye-arm64-lite.zip  I wanted to use as less services as available, to get enough speed.
After that i increased the Z-Ram by 2G like described above.

[image]

mcguirepr89, 2021-11-28:
I'm happy to help and appreciate you trying to get this up and running on the Lite version of the OS. I used to only run things on the Lite version for my personal installations, but have done a lot of developing since then and it seems like there may be some dependency issues when using Lite now. Thank you for your help in pin pointing those things for me and others.

I'm scratching my head on how to best investigate what is going on. Not having the "activate" file in the python venv does not make sense to me, so I can only conclude that there were some glaring errors during installation that need to be addressed.

Is there any chance you would be willing to host a r[emote desktop session via AnyDesk](https://anydesk.com/en/downloads/) on the device you are using to access the BirdNET-Pi via SSH?

I would really like to take a look through some logs and basically redo the installation while increasing the installer's verbosity.

If a remote session is out of the picture, I can recommend the following, which is what I would personally do:

1. Issue `/home/pi/BirdNET-Pi/scripts/uninstall.sh` to remove everything that has been installed so far
2. Issue `rm -drfv /home/pi/BirdSongs` to remove the data directory
3. Issue `cp ~/BirdNET-Pi/birdnet.conf-defaults ~/BirdNET-Pi/birdnet.conf` to create a new, generic `birdnet.conf`
4. Edit `birdnet.conf` to NOT select NoMachine installation
5. Increase verbosity in the `install_birdnet.sh` and `install_services.sh` scripts. Near the top of each script, there is a line for debugging:
   ```
   . . .
   # set -x
   . . .
   ```
   Remove the '#' to enable debugging.
6. Issue `/home/pi/BirdNET-Pi/scripts/install_birdnet.sh`

Very carefully examine the installation for any errors to investigate further.

Again, I'm more than happy to take a look myself with AnyDesk -- this information is quite valuable to me, so either way I appreciate your persistence.

My best

mcguirepr89, 2021-11-28:
I had another thought: 
You may try using the pre-installed Beta image, which we know works, and then disable the desktop environment and all of its services by changing the system's default target. You would still need to enable the zRAM, but you know how to do that.

To use the pre-installed image and then disable the graphical.target:

1. Write the [Pre-installed image you can find here ](https://pmcgui.xyz/sdm_downloads/birdnet-pi-v0-9-beta/)to the SD card and insert it into the Pi to boot.
2. Wait at least 5 minutes for the system to configure itself after the first boot. Better wait 7 just to be sure. It will reboot when it has finished.
3. After 7 minutes (the reboot), you can connect to the "BirdNET-Pi_v0.9" WiFi network from the host machine you've been using to access BirdNET-Pi via SSH.
4. Once on the "BirdNET-Pi_v0.9" network, you can use a VNC client to connect to **192.168.16.1**
5. Before starting the configuration wizard, open the terminal and issue `sudo systemctl set-default multi-user.target` to disable the desktop for all subsequent sessions.
6. Also issue
   ```
   sudo sed -i 's/ExecStart=.*/ExecStart=echo/g' /etc/systemd/system/network-online.target.wants/systemd-networkd-wait-online.service && sudo reboot
   ```
   to fix a bug with the image.
7. After those terminal adjustments, you can walk through the BirdNET-Pi Configuration Wizard to setup the system.
8. When the http://birdnetpi.local page automatically opens upon completion, reboot the BirdNET-Pi.

Let me know if you would like to pursue this option forward and/or if you have any questions about the process.

PS, please read the "Welcome ReadMe" on the desktop before you begin the Wizard.

gitalm, 2021-11-28:
Dear Patrick, thanks for your help.
I have two problems. I reinstalled the whole with the version mentioned in the Headless Installation and did exactly all steps. The Problem remains
[image]

I would try to use you 0.9 image, but the server is not found at the moment. 

[image]

mcguirepr89, 2021-11-28:
Do try once more
[image]

gitalm, 2021-11-29:
Dear Patrick,

i could now download your image and installed it on the RPi3. I can connect to the WIFI, but i cannot reach the RPi, neither SSH nor VNC (see screenshot) 
Is there another password?

Best wishes Johannes
[image]

mcguirepr89, 2021-11-29:
That "BirdNET-Pi_v0.9" access point is the way into the system. Could you please show me what kind of error you are seeing when trying to connect with VNC and with SSH?

gitalm, 2021-12-11:
Dear Patrick,

today my RPi 4 arrived and it is working. My problem with the RPi 3 was, that i used RPImager and set automatically a password.
Thanks for the .0.10 version. Will try RPi3 soon, just need another SD Card.

mcguirepr89, 2021-12-11:
That's great news! I just got a new job that starts Monday, so I will get myself a RPi3 in the next couple of months and will be able to do much more thorough testing for that model.

May I ask where you BirdNET-Pi will be listening?

gitalm, 2021-12-11:
My BirdNetPi should listen at lake of chiemsee https://www.openstreetmap.de/karte.html?zoom=12&lat=47.86819&lon=12.45941&layers=B00TT
I'm really excited

mcguirepr89, 2021-12-11:
Very exciting!! You'll have a lot of great results, I'm sure!

I hope you might [consider making the installation public](https://github.com/mcguirepr89/BirdNET-Pi/wiki/Sharing-Your-BirdNET-Pi) (if it will have internet access out there)

Also, it would be wonderful if you would consider contributing your data to https://app.birdweather.com

You can see my installation in Charlotte, North Carolina, United States (it's the green circle on the Eastern coast of the United States) to get an idea of how your installation will contribute data.

If you are interested, please email Tim to request a BirdWeather ID (which you will then simply input in `birdnet.conf` to start using) -- "thewebsite.com" is "birdweather.com" and you can reach Tim via email at "hisname"@"thewebsite.com" (trying to obscure the email address from site-crawlers).

Thanks again for installing BirdNET-Pi!! :)

gitalm, 2021-12-12:
Dear Patrcik,

it is running with a RPi3!!!!! Thanks for your patience and help

joknjokn, 2021-12-12:
@gitalm Great! Would share an installation/setup guide for the RPi3? :)

mcguirepr89, 2021-12-12:
Let me know how it performs compared to the RPi4!

gitalm, 2021-12-12:
I do my best. But the installation was just the same with your image 0.10.
I changed now zram and gonna try both systems. 
How can i measure this? `htop`

mcguirepr89, 2021-12-12:
>I changed now zram and gonna try both systems.

The zRAM swapping will be a big help

>How can i measure this?

The easiest way to compare the two would be only run one system at a time and compare the **`birdnet_analysis.service: Consumed TIME(s) CPU time`** line in the "BirdNET-Lite Log" page. 

Do this for each (after configuring zRAM for the RPi3):

1. Find a youtube video of local birdsongs, at least 3 minutes or so.
2. While playing the video into the microphone, watch the "BirdNET-Lite Log" page under "System Links" on http://birdnetpi.local
3. Between each analysis cycle, you will see a report from `systemd` as to how much time the service used the CPU. It will look something like this:
   ```
   ...
   Date-time: 2021-12-12 10:57:43
   ANALYZING AUDIO... DONE! Time 1.6 SECONDS
   WRITING RESULTS TO /home/pi/BirdSongs/December-2021/12-Sunday/2021-12-12-birdnet-10:57:43.wav.csv ... DONE! 
   WROTE 0 RESULTS.
     ('Junco hyemalis_Dark-eyed Junco', 0.09362676)
     ('Junco hyemalis_Dark-eyed Junco', 0.20422241)
     ('Junco hyemalis_Dark-eyed Junco', 0.08576561)
     ('Dryobates borealis_Red-cockaded Woodpecker', 0.11782787)
     ('Euphagus carolinus_Rusty Blackbird', 0.08103995)
    Database Entry
    Database Entry
    Database Entry
    Database Entry
    Database Entry
   birdnet_analysis.service: Succeeded.
   birdnet_analysis.service: Consumed 7.923s CPU time.                                            <-------------------- This line
   birdnet_analysis.service: Scheduled restart job, restart counter is at 19141.
   Stopped BirdNET Analysis.
   birdnet_analysis.service: Consumed 7.923s CPU time.
   Started BirdNET Analysis.
   Sun 12 Dec 2021 10:58:07 AM EST
   Starting run_birdnet() for December-2021/12-Sunday
   get_files() for December-2021/12-Sunday
   Files loaded
   Starting move_analyzed() for December-2021/12-Sunday
   Moving analyzed files to new directory
   Starting run_analysis() for December-2021/12-Sunday
   ...
   ```

Do that on both and compare that value and we'll have a decent idea of how their core function compare to one another

mcguirepr89, 2021-12-12:
Also, [take a look at this](https://github.com/mcguirepr89/BirdNET-Pi/wiki/Advanced-Configuration) for how you can run both BirdNET-Pis on the same network at the same time. My instructions above specify only having one system on the network at a time, but it is certainly possible for them to co-exist on your network by setting up DHCP reservations on your router and configuring the `BIRDNETPI_URL` in `birdnet.conf` to use their respective LAN IP addresses.

gitalm, 2021-12-12:
Dear joknjokn,

i followed the instruction from Patrick
To use the pre-installed image and then disable the graphical.target:

* Write the Pre-installed image you can find here to the SD card and insert it into the Pi to boot.
 * Wait at least 5 minutes for the system to configure itself after the first boot. Better wait 7 just to be sure. It will reboot when it has finished.
* After 7 minutes (the reboot), you can connect to the "BirdNET-Pi_v0.10" WiFi network from the host machine you've been using to access BirdNET-Pi via SSH. The "BirdNET-Pi_v0.10" Access Point password is 'birdnetpi'
* Once on the "BirdNET-Pi_v0.10" network, i used SSH client to connect to 192.168.16.1. The user is 'pi' and the default password is 'birdnetpi'
* `sudo systemctl set-default multi-user.target` to disable the desktop for all subsequent sessions.
* Also issue
    `sudo sed -i 's/ExecStart=.*/ExecStart=echo/g' /etc/systemd/system/network-online.target.wants/systemd-networkd-wait-online.service`
* First run `sudo raspi-config` to change the password `1 System Options  -> S3 Password`
* i also changed the Localisation within raspi-config `5 Localisation Option -> L1 - L4`
 for your country
* After those terminal adjustments, you can walk through the BirdNET-Pi Configuration Wizard to setup the system with `birdnet-pi-config `
* When the http://birdnetpi.local page automatically opens upon completion, reboot the BirdNET-Pi.
* in the end i added ZRAM in the console `System Options -> Configure zRAM`

mcguirepr89, 2021-12-12:
It looks like your steps above might be omitting running through the graphical BirdNET-Pi Configuration Wizard before disabling the `graphical.target`?

gitalm, 2021-12-12:
But still there is a small problem with my RPi3 
My Audio is not recognized, although it is the same software
[image]

mcguirepr89, 2021-12-12:
Hmm that _is_ very weird?

what is your output for `aplay -L`?

gitalm, 2021-12-12:
```
null
    Discard all samples (playback) or generate zero samples (capture)
default
    Playback/recording through the PulseAudio sound server
lavrate
    Rate Converter Plugin Using Libav/FFmpeg Library
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
hw:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct hardware device without any conversions
plughw:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Hardware device with all software conversions
sysdefault:CARD=Headphones
    bcm2835 Headphones, bcm2835 Headphones
    Default Audio Device
dmix:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct sample mixing device
usbstream:CARD=Headphones
    bcm2835 Headphones
    USB Stream Output
hw:CARD=E205U,DEV=0
    Superlux E205U, USB Audio
    Direct hardware device without any conversions
plughw:CARD=E205U,DEV=0
    Superlux E205U, USB Audio
    Hardware device with all software conversions
sysdefault:CARD=E205U
    Superlux E205U, USB Audio
    Default Audio Device
front:CARD=E205U,DEV=0
    Superlux E205U, USB Audio
    Front output / input
surround21:CARD=E205U,DEV=0
    Superlux E205U, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=E205U,DEV=0
    Superlux E205U, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=E205U,DEV=0
    Superlux E205U, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=E205U,DEV=0
    Superlux E205U, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=E205U,DEV=0
    Superlux E205U, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=E205U,DEV=0
    Superlux E205U, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=E205U,DEV=0
    Superlux E205U, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=E205U,DEV=0
    Superlux E205U, USB Audio
    Direct sample mixing device
usbstream:CARD=E205U
    Superlux E205U
    USB Stream Output
hw:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM i2s-hifi-0
    Direct hardware device without any conversions
plughw:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM i2s-hifi-0
    Hardware device with all software conversions
sysdefault:CARD=vc4hdmi
    vc4-hdmi, MAI PCM i2s-hifi-0
    Default Audio Device
hdmi:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM i2s-hifi-0
    HDMI Audio Output
dmix:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM i2s-hifi-0
    Direct sample mixing device
usbstream:CARD=vc4hdmi
    vc4-hdmi
    USB Stream Output

```

mcguirepr89, 2021-12-12:
try issuing `alsamixer` -- it should appear like this 
[image]

Then press `F6` so that it brings up a list like this:
[image]

and see if your sound card is listed there. If so, choose it and then press `F4` to view the "capture" devices associated with the sound card.
[image]

gitalm, 2021-12-12:
All works here 
[image]

mcguirepr89, 2021-12-12:
Then I'm not quite convinced that there is an issue with the sound _yet_ -- it looks like we might just have services that haven't been started.

Let's try these things and let me know what you see:

1. `restart_services.sh`
1. `journalctl -n10 --no-pager -u birdnet_recording.service`
1. `journalctl -n30 --no-pager -u birdnet_analysis.service`
1. `journalctl -n30 --no-pager -u livestream.service`

gitalm, 2021-12-12:
Thanks, it helps and it started recognizing

mcguirepr89, 2021-12-12:
My pleasure! I really appreciate you experimenting with BirdNET-Pi on the RPi3 -- looking forward to learning your results!

gitalm, 2021-12-12:
RPi with that file https://xeno-canto.org/sounds/uploaded/BLMSIUFTFU/XC690635-200614_1865-3.30_TM-%C5%9Bp_KampPN.mp3 
[image]

gitalm, 2021-12-12:
RPi4 with the same sound and the same microphone without zRAM
(but i cannot see this with my PC over birdnetpi.local (see screenshoot in the same WIFI)
[image]

mcguirepr89, 2021-12-12:
Yep it looks like the RPi3 is a bit backed up, even with the desktop disabled.

Some other things to try:

1. Adequate cooling will help a lot -- definitely use a heatsink and fan is also great if you have one. 
2. You can try disabling the live-stream: `sudo systemctl disable --now livestream icecast2`
3. You can tell `analyze.py` to use all cores for the analysis by editing this part of `/home/pi/BirdNET-Pi/analyze.py`:
   ```
   ...   
   print('LOADING TF LITE MODEL...', end=' ')
   
   # Load TFLite model and allocate tensors.
   interpreter = tflite.Interpreter(model_path='model/BirdNET_6K_GLOBAL_MODEL.tflite',num_threads=2)
   interpreter.allocate_tensors()
   ...
   ```
Change **`num_threads=2`** to **`num_threads=4`**
   ```
   ...   
   print('LOADING TF LITE MODEL...', end=' ')
   
   # Load TFLite model and allocate tensors.
   interpreter = tflite.Interpreter(model_path='model/BirdNET_6K_GLOBAL_MODEL.tflite',num_threads=2)
   interpreter.allocate_tensors()
   ...
   ```

### Disclaimer -- this can break your pi if you push it too hard, or at least void your warranty
So long as you are confident that the power-supply is adequate, you can also try overclocking the system. Be aware, though, that if this fails, the image won't boot until you fix things, which isn't difficult, but if it's you're first time doing it, it can seem like a real nuisance.

To overclock the RPi3, edit the `/boot/config.txt` file, and add these lines:
```
arm_freq=1500
over_voltage=4
```

To see the overclocking and ensure that there is no over-voltage warnings, I put together a small tmux pane that displays clock speed info and voltage "throttling":
[image]
The new image has an old tmux, so run this to see the pane properly:
```
sudo rm /usr/local/bin/tmux
sudo apt -y install tmux
/usr/bin/tmux
```
Those commands "fix" the broken tmux, so after it's fixed, (in subsequent shell sessions), you'll just be able to call `tmux` to see those values.

If all goes well, just increase the clock value gradually. But don't try to exceed 1750, as you most certainly won't boot at that point . . . I don't think.

If after overclocking you find your pi won't boot, put the SD card in an SD card reader and insert it into your laptop/PC and edit that same `/boot/config.txt` file back to whatever you know worked.

gitalm, 2021-12-12:
I tested with your values
* i have a fan for RPI3
* stopped icecast
* increased to 4
* added arm_freq=1300
core_freq=500
sdram_freq=500
over_voltage=2

[image]

mcguirepr89, 2021-12-12:
I'm afraid arm_freq=1300 will actually *throttle* the arm, since its
default clock speed is 1400 -- I may be wrong about that, but that is my
understanding

mcguirepr89, 2021-12-26:
Hi, @gitalm @joknjokn --

I hope this finds you both doing quite well.

I wanted to share with you the news that I received a Raspberry Pi 0W2 as a gift yesterday and plan to test out ways that BirdNET-Pi might be able to run on that hardware. It's not an RPi3, but it's Broadcom chipset has the same core type, Cortex-A53 64-bit, but with a much lower clock speed (1.4GHz for RPi3, 1GHz for RPi0W2) and significantly less RAM.

I should be able to mitigate the memory constrains with well-configured swap space, but will let you know how I fare with the other obstacles.

My best regards to you both,
Patrick

oliverk71, 2022-01-07:
Interesting topic, since I would like to use one of my older 3B (not 3B+) for birdpinet. I think I would use an aluminium case which is a cooler, too. But I don't like these small fans since there is always trouble with them. Downside of a aluminium cooler case is that I would have to add an external wifi antenna.

Svardsten53, 2022-01-07:
Try to not mount the top of the mounting box. It must be the best way to maximize both airflow and WiFi signal. I have done so on my Raspberry Pi 4B because it is inside another box.

oliverk71, 2022-01-07:
If a aluminium cooler case is being used, the top must be mounted for optimal cooling. That's why an external antenna is necessary. But that is not a big problem, since ready-to-use 'pigtail' SMA cables were available.

mcguirepr89, 2022-01-26:
@gitalm @oliverk71 @joknjokn @Svardsten53 Howdy, gang -- just wanted to let you know that I've gotten my hands on a Raspberry Pi 3B+ that I will test with today. I will let you all know my results once I've had enough time to tinker with settings.

My best!

DD4WH, 2022-01-26:
Patrick, thats really excellent!
First trial result with my RPi3B+:
* installation worked perfectly without any problem exactly as you described
* everything I tried worked:
* Live stream
* Changing language to German
* Changing settings: overlap, recording length, extraction length
* Changing costum URL and subsequent "Update"
* Edit scripts (restrict spectrogram y-axis to 10kHz) with tiny file manager --> very nice tool!
* I can reach the BirdNET-Pi with my Windows PC, with a historical model Ipod Touch and with an Android phone

Excellent work!!!

mcguirepr89, 2022-01-26:
@DD4WH Yes it is working hard -- ironically it takes about 20~ seconds to build the model, and then about 2-5 seconds to process the 30 second clip :) -- I'm sure there are even better results that can be had, but these are pretty great for such an inexpensive computer with only 2G RAM and no additional swapping (it doesn't seem to need it at all?)

joknjokn, 2022-01-26:
Awesome!
So, I just realized my RPi3 is 1GB. 

I have it up and running-ish. Installed fine (hours I guess), and the website works. But:
The "Currently analyzing" spectrogram states a timestamp that is 25 minutes late or so. I guess it's struggling to catchup with a queue of recordings, or something like that?

Any suggestions to what settings I should try? I don't expect it to actually end up working well with 1GB, but I think it's worth the try now it's actually running.

oliverk71, 2022-01-27:
@joknjokn   I use 2 GB zRAM, moderate overclocking and 30 s audio on my 3B. Seems to work.
You can set zRAM use in birdnet-pi-config, overclocking in boot.conf and audio recording time in the advanced settings of the birdnet-pi's local website.

mcguirepr89, 2022-01-27:
Thanks, @oliverk71 

@joknjokn , could you compare what you see on your "System Links" > "System Info" page around where it looks likes this and let me know what you see?

[image]

[51 more replies]