mcguirepr89/BirdNET-Pi #87: Second System Crash - Low priority

CaiusX, 2021-12-03:
Hi @mcguirepr89 

My "second system"'s BirdNET-Pi crashed yesterday at around 14h27 my time.  

The system is on a 32GB SD card - & seems to have around 12GB free disk space so memory isn't the problem.  Its the original 0.9 installation with PiSound HAT mods and with PIV2 camera attached & live video feed window open.  

It was running unattended headless with video streaming.  

I don't think this is a high priority to get right before 0.10 release as its probably something to do with my "special needs" PiSound HAT & Video.

I have left it exactly as is if you want to dig through logs to see why.

nx://169.0.72.134:22684

Mail me for the pw if needed :).

Best
Dennis

mcguirepr89, 2021-12-03:
THanks for letting me know, Dennis -- I definitely suspect it was a RAM/free-memory crash. `free -h` may indicate no more memory. When I get a chance, I'll hop in and will probably enable the zRAM swapping (`birdnet-pi-config`--->System Options) for the 4G to see if that lets the system run without a crash.

mcguirepr89, 2021-12-03:
I found that the issue was actually _not_ related to resources, but a fight between between the `livestream`/`icecast2` service and `birdnet_recording.service` over the PiSound input channel.

I don't think I mentioned to you that the PiSound card input channel was not getting abstracted properly by `pulseaudio`, so I had to set the `REC_CARD` variable to the ALSA representation of the PiSound input, which does not allow for sharing. So, at some point the `livestream` or `icecast2` service had a restart that was timed just right and was able to close the `birdnet_recording.service` input channel, which stopped creating recordings, which made the analysis service eventually fail as well.

The solution for now was to disable both the `livestream` and `icecast2` services so that `birdnet_recording.service` gets to use the PiSound card exclusively.

It's kind of a bummer that the PiSound card is not getting handled by `pulseaudio` in typical fashion, as this makes the live stream a dud. The good news, though, would be that you could possibly replace the "Play Live Audio" link with a "View Live Video Stream" and then no one would need to know :)

CaiusX, 2021-12-03:
Thanks @mcguirepr89 

Think I'll go back to the USB solution for this test unit.  Whilst a dedicated audio hat sound like a good idea, the plug n play nature of USB audio is best I suspect until the Raspberry PI foundation puts analogue in pins.

My testing indicates that having a dedicated tweakable pre-amp allows me to get more recognitions but in the main of the same birds.  I'll do some daily comparisons "phenology" charts between the 2 systems & report in the show and tell section.

Best
AP

closed 2021-12-03 by CaiusX

mcguirepr89, 2021-12-03:
Sure thing -- thanks for having the second system for additional testing -- very cool and helpful!!

I wrote a little Crash Report and left it on the desktop for you as well