mcguirepr89/BirdNET-Pi #93: Can not find module "pytz" when analyzing spectrograms

Svardsten53, 2021-12-08:
My newly installed system is now up and running and I get spectrograms. However, no analysis results and when I checked the BirdNET-Lite log I see the following error message:
```
Traceback (most recent call last):
  File "analyze.py", line 25, in <module>
    import pytz
ModuleNotFoundError: No module named 'pytz'
RECORDING_LENGTH set to 15
python3 analyze.py --i /home/pi/BirdSongs/december-2021/08-onsdag/2021-12-08-birdnet-13:59:57.wav --o /home/pi/BirdSongs/december-2021/08-onsdag/2021-12-08-birdnet-13:59:57.wav.csv --lat 59.253 --lon 18.248 --week 46 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
```
Google says there is something with Python time zone stuff and I have tried to run the install commands "sudo pip install pytz" and "sudo pip3 install pytz" without any result, not even after reboot. I do not want to take a chance and dig too deep into the system because I am now eager to make this exciting system work.

Svardsten53, 2021-12-08:
Perfect, that helped, error message is gone, but got a new message: 
```
from tzlocal import get_localzone
ModuleNotFoundError: No module named 'tzlocal'
```

mcguirepr89, 2021-12-08:
Haha same solution, then only use `tzlocal` where you formerly used `pytz` and that should do it for you

mcguirepr89, 2021-12-08:
If there are any more errors messages, you can cover all your bases with this one:

```
source /home/pi/BirdNET-Pi/birdnet/bin/activate
pip3 install -U -r /home/pi/BirdNET-Pi/requirements.txt
```

and that will ensure that ALL required packages are installed. (Not sure why this didn't happen for you during installation?)

Svardsten53, 2021-12-08:
Bingo! Now the log looks completely different. I get a lot of very strange species listed, but with very low confidence value.
It is dark here in Sweden now, so have to wait until tomorrow when the birds have woken up again.
Thanks again, you respond really fast!

mcguirepr89, 2021-12-08:
My pleasure! So glad to have you up and running. I hope you will considering [sharing some of your results with our little community of BirdNET-Pi lovers](https://github.com/mcguirepr89/BirdNET-Pi/discussions/69) -- you may also be interested in having your installation POST its results to https://app.birdweather.com (you can view mine in Charlotte, North Carloina, United States) -- if so, reach out to @timsterc [on this discussion page](https://github.com/mcguirepr89/BirdNET-Pi/discussions/82) and he will be able to provide a BirdWeather ID, and I will be more than happy to help make sure it works well for you.

My best regards and thanks again for using BirdNET-Pi! I hope you love yours as much as I love mine :)

closed 2021-12-08 by mcguirepr89