mcguirepr89/BirdNET-Pi #94: minor issue: format of no. of detections

DD4WH, 2021-12-09:
[image]

I noticed that the no. of detections is formatted as "2e+x" when it exceeds 100.
It would be desirable to have the no. of detections (for each hour) stated as "124", instead of "2e+024"

mcguirepr89, 2021-12-09:
I'll get with @CaiusX and I bet we'll be able to adjust that

mcguirepr89, 2021-12-09:
@CaiusX I found [this](https://michael-fuchs-python.netlify.app/2019/04/28/how-to-suppress-scientific-notation-in-pandas/) and will see what kind of results I can come up with

DD4WH, 2021-12-09:
May I suggest something else that is related to rounding  scientific notation ?
Could the confidence scores be expressed as % and rounded to one digit after the decimal point [eg. 92.4%] ? Three digits are more than enough and we do not need that kind of precision, I think ;-)

mcguirepr89, 2021-12-09:
>May I suggest something else that is related to rounding / scientific notation ?
Could the confidence scores be expressed as % and rounded to one digit after the decimal point [eg. 92.4%] ? Three digits are more than enough and we do not need that kind of precision, I think ;-)

Noted!

mcguirepr89, 2021-12-09:
[How's this?](https://birdnetpi.pmcgui.xyz/viewdb.php)

mcguirepr89, 2021-12-09:
I've pulled the new percentages view and _suppressing scientific notation_ into main, so you should be able to use the "Update BirdNET-Pi" "Tool" to get those changes into your system

mcguirepr89, 2021-12-09:
@CaiusX's results:
[image]

closed 2021-12-09 by mcguirepr89