mcguirepr89/BirdNET-Pi #95: minor issue: calculation of "Last Hour"

DD4WH, 2021-12-09:
[image]

It seems the no. of detections in the last hour is not always calculated correctly. Not sure why this is so. Maybe a cause of different time zones and the software not taking account of this!?

DD4WH, 2021-12-09:
Also, it would be good to having the same time format (24h) at all places. At the moment database uses 24h, graph showing recent detections uses AM/PM format.

mcguirepr89, 2021-12-09:
Hi @DD4WH 
I'm not sure what I'm looking at for the issue with the "no. of detections in the last hour" -- what were you expecting the image you shared to show? I'm sorry, I'm just trying to pinpoint the issue so I can try to look into it further, but am unsure where to look at the moment.

Also, I will revert the chart time to 24h format (I'm so American)

mcguirepr89, 2021-12-09:
I'm sorry, I understand now. 

What are the timestamps for the extractions? Are they accurate?

DD4WH, 2021-12-09:
Thank you, Patrick for taking care of this!
* timestamps are OK, but they are in the 24h format --> maybe that is the reason? Comparing AM time stamps with 24h time stamps?
* However, if I look at the second BirdNET-Pi that I have running, everything seems OK . . . 
[image]

Hmm, not sure if I messed something up and that I am the cause !?

mcguirepr89, 2021-12-09:
hahaha Hard to say!! I don't see that issue on my system nor @CaiusX's -- but that doesn't necessarily mean that you are the root cause!

The AM/PM only shows up on the chart and is in no way related to Extractions nor database entries. All database entries (and BirdDB.txt entries) are 24h formatted, so we can safely say that is not the issue here. 

Perhaps running an "Update BirdNET Pi" will get things working again as needed? (This won't change your `birdnet.conf` file at all, but would replace your German `labels.txt`, so make sure you back that up before running an update, should you choose to do so.)

Are there any customizations that you can recall on the system that is having the trouble?

mcguirepr89, 2021-12-09:
>Also, it would be good to have the same time format (24h) at all places. At the moment database uses 24h, graph showing recent detections uses AM/PM format.

I have made this change now -- you can see how it will look on [my installation](https://birdnetpi.pmcgui.xyz)

DD4WH, 2021-12-09:
Thank you! Looks very good now!

I tried your suggestion to do an update, but still too many detections for "Last Hour". But probably I will have to wait until tomorrow until I see any change?

I am not convinced any more that this is a real bug, so maybe better close this issue! Probably this is connected to my messing up of the system! Sorry for causing any trouble!

mcguirepr89, 2021-12-09:
No trouble at all! I'm more than happy to hunt down any bug, whether real or phantom! Let me know what you see tomorrow -- sounds strange, but perhaps it was a system-time issue of some type. Do look regularly in the BirdNET-Lite and Extraction logs for anything that seems unusual.

I'll keep this open until we see what the system is up to tomorrow. If all is well after the Update, we'll close this. If there are still issues and you'd like to hunt them down, I am 100% ready!

My best,
Patrick

CaiusX, 2021-12-09:
Hi @mcguirepr89  & @DD4WH 

Putting fmt="g" into the heatmap plot will force integer readings 

i.e. in line 94 of daily_plot.py (which is in  /BirdSongs/Extracted/scripts) 

`plot = sns.heatmap(heat, norm=LogNorm(), annot=True,  fmt="g",  annot_kws={"fontsize":7}, cmap = pal , square = False, cbar=False, linewidths = 0.5, linecolor = "Grey", ax=axs[1], yticklabels = False)
`

I'll learn myself how to publish changes into Github soon :)

I'm dabbling with ideas on how to incorporate Confidence in the phenology graphic - hopefully inspiration will hit me on the beach!!

Best

mcguirepr89, 2021-12-09:
Then I have put that in place and it will be a part of the `main` branch once it's been tested and verified 👍 Thanks @CaiusX for your speedy solution :)

DD4WH, 2021-12-10:
OK, tested for one day after update:
* everything works perfectly! Thanks Patrick for your help!
* Last hour no. of detections is calculated correctly
* Time stamp of graphical overview is in 24h format
* Confidence scores are in % and limited to one digit after the decimal point!

Wow, all requests already implemented! Superb! 

Thanks a lot!

mcguirepr89, 2021-12-10:
: ) very happy to help -- thank you for your suggestions and thanks @CaiusX for his speedy insights!

I'll go ahead and close this issue -- please let me know if you see anything else unusual and I'll be more than happy to see what is the matter.

My best regards as always!!!
Patrick

closed 2021-12-10 by mcguirepr89