mcguirepr89/BirdNET-Pi #96: DHCPCD Probleme

gitalm, 2021-12-12:
Dear Patrick, i have another problem with the internet.

The RPi says 'connection to dhcpcd lost' and woth the command 'sudo dhcpcd - 4' i can connect with the network manager.
With the control 'journalctl -fu birdnet_analysis' i see the RPi working.

[image]
[image]

mcguirepr89, 2021-12-12:
Please read through the image's included "Welcome ReadMe" to better understand the networking for the system.

The pre-installed image does not use DHCPCD by default, and instead uses systemd-networkd. To setup the network, please go through the steps on the BirdNET-Pi Configuration Wizard.

Let me know if you still have trouble after reviewing that document and following the configuration wizard.

My best,
Patrick

mcguirepr89, 2021-12-12:
Also, you will never be able to reach "https://birdnetpi.local" as http is over 443 and would require an SSL certificate, which cannot be issued to ".local" domains.

gitalm, 2021-12-12:
Thanks for the help. I have to IP adresses 
Should i remove some software?
[image]

mcguirepr89, 2021-12-12:
No need to remove any software -- when you reboot, DHCPCD will be stopped (it is disabled on the image).

When you reboot, the BirdNET-Pi ought to connect to the network just fine, though that desktop panel icon will _appear_ to show that things are NOT connected. Use the _far right_ network monitor from your most recent screen shot to view networking information.

With this setup, if you take your BirdNET-Pi somewhere that it cannot connect to your home's WiFi network, it will automatically present the BirdNET-Pi_v0.10 access point for you to be able to connect to it in the field without any internet, set the system clock using the BirdNET-Pi Configuration Wizard, and be able to collect data fully offline.

Anytime the BirdNET-Pi _is_ able to reach your home network, it will be connected.

I have been searching for a good desktop panel replacement for systemd-networkd that would function like the DHCPCD panel icon, but I have not found anything yet.

Please let me know if BirdNET-Pi is able to connect without trouble after a reboot (though the desktop panel icon will be misleading and pretty useless).

If you ever want to add another network to the BirdNET-Pi's list of "known networks" it will try before presenting its Access Point, you can manually start the dhcpcd client in the command line and then you will be able to use the desktop panel icon again as you are used to doing.
```
sudo systemctl start dhcpcd
```

But only do that when you want to add networks to the BirdNET-Pi's list of known networks -- it shouldn't be necessary to do otherwise.

gitalm, 2021-12-12:
Thanks,  this idea is awesome.birdnetpi
It is working now

mcguirepr89, 2021-12-12:
I'm so glad you appreciate why it is different.

It is a little frustrating at first to have something familiar taken away (like that wifi desktop panel being so easy to use), but the ability to have that hotspot appear anytime the BirdNET-Pi cannot find its known networks is a really great feature for field researchers.

I'll go ahead and close this issue, but do let me know if you encounter any trouble.

My best,
Patrick

closed 2021-12-12 by mcguirepr89