mcguirepr89/BirdNET-Pi #102: Local webserver

oliverk71, 2022-01-05:
Hi, I have just set up birdnet-pi and there seems to be a problem with the local webserver. It does not show anything any more. At first startup it was okay, but still no recordings (of course). After updating and restart the problem occured. It's the same from my PC and on birdnet-pi, so it shouldn't be a network issue. I did not investigate much by now.

mcguirepr89, 2022-01-05:
Hi, Oliver, and thank you so very much for your interest in BirdNET-Pi. I'm really sorry that you're having trouble with your system, but I'm sure we'll get to the bottom of things. Thanks also for creating a new issue for this discussion, as it is always helpful for others.

To get started, I will definitely need more information in order to determine what is going on. 

The best place for us to start would be allowing me to take a look through your logs. There is a script installed on your BirdNET-Pi called `dump_logs.sh`. If you could be so kind as to run that script on the BirdNET-Pi, and then [email me](mailto:mcguirepr89@gmail.com) the zipped up tarball of logs it creates, it will give me a general overview of your system and may reveal the issue. You will find the zipped and tarred logs in the `/home/pi/BirdNET-Pi` directory and it will be entitled "`logs.tar.gz`".
Note, the log-dump contains one very minor password, which is why I request you email that file. If you are committed to changing your passwords after troubleshooting, you're more than welcome to try to post that file here instead -- whatever works for you.

After I've been able to look through those files, I may request more information, or may have a solution. Either way, we'll know where to go from there.

My best regards,
Patrick
PS -- You probably won't hear back from me until after work this evening after 6pm Eastern, but you'll definitely hear back asap!

closed 2022-01-06 by mcguirepr89

nilspupils, 2022-01-06:
Please keep us updated, it could be the same issue that prevents me to open birdnetpi.local on my Windows pc (and my Android phone btw.)

mcguirepr89, 2022-01-08:
Hi, Nils!! I hope you're well :)

The issue you might be encountering based on what you mentioned could be due to your router's "mDNS" or "Multi-Cast DNS" settings. It will need to allow mDNS for the 'birdnetpi.local' service to be announced on the network and therefore reachable throughout. Also, you may take a look at the router's "Domain" settings. Sometimes the "Domain" is set to "lan" or something other than "local" -- in rare instances, this will disallow the _".local"_ portion of the BirdNET-Pi.local from resolving correctly. 

The situation with Oliver was due to trying to reach the web interface via the Pi's IP address, but Caddy was not configured to listen on the IP address. When I guided him through adjusting the Caddyfile to respond to requests to the IP address, it is now reachable at that IP.

I hope that info might be helpful -- but if it is not, please don't hesitate to provide some more information and you know I will be very willing to help troubleshoot :)

My best regards as always!!
Patrick

nilspupils, 2022-01-08:
Hi Patrick, thank you all is fine here. I hope you are well and you new job is going all right? 
Thanks so much for offering help again with this issue! I had finally some time to play with my Pi again today. My problem was that i could never reach the web interface at "birdnetpi.local" but also not at the Pi's IP address. I first thought this was a problem with my router, but then i realized it happened also when i connected directly to the Pi's access point. I then thought it was a problem with my computer but it also happend from my phone and another computer. 
Today i changed the custom Birdnetpi-URL in birdnet.conf and after updating i can now finally reach the web interface with the IP address!! The fact that it still doesn't work with "birdnetpi.local" is funny but probably has to do with Windows. So thanks very much for asking but i think this is resolved now. (The GPIO did not yet work but i replied to it in the original thread).
Best regards, 
Nils

mcguirepr89, 2022-01-09:
Thanks for the update! 

Having an easy way to make the installation addressable using its LAN IP address has proven to be a very commonly needed feature for a lot of users. Home routers and individual devices have so many configuration settings that can prevent certain network services from playing nicely with one another. Using IP addresses in network environments is a really great way to bypass individual peculiarities that might be lurking in router settings, laptop/desktop settings, general network routing (that seems to always work . . . until you need something special), and DNS caching (which is a really tricky one to troubleshoot since there are many links in the DNS daisy-chain).

I feel the project is past-due for an updated pre-installed image version, but I'm consistently running out of time and resources on my BirdNET-Pi project goals as they accumulate and never seem to warp space/time long enough for me to get to everything. Fingers crossed that available time presents itself soon so that I can keep some of the project's momentum.

TTFN!