mcguirepr89/BirdNET-Pi #104: Run on Docker?

Swiftnesses, 2022-01-16:
Hello,

Looking to install this to learn and have fun with the kids :-)

I have a couple of docker stacks running on my server and some unifi cameras with feeds available - would it be possible to deploy this on a custom stack, based on something like Debian perhaps?

Many thanks in advance.

mcguirepr89, 2022-01-16:
Hello, @Swiftnesses and thank you so much for your interest in BirdNET-Pi and for your thoughtful question.

I have not tried anything like that and am not super sure what would be needed to achieve that. I'm not great with Docker from scratch.
BUT, through my testing and trying to keep the stack Debian-based, and not necessarily RaspiOS-based, there are a few things I _can_ say.
### Possible Difficulties
##### (But not impossible hurdles by any means)
- The '`pi`' user is hard-coded in a few places. < `sed` can take care of that without too much hassle if for some reason you were not able to simply set the user to `pi`
- The pre-built TFLite binaries in the typical installation are for aarch64. Those binaries fun about 4 times faster on aarch64 over  armhf, and require the full package support, not just the kernel to install correctly. < There are versions in this repo for python 3.7 and 3.9, but I'm dropping support for 3.7, as 3.9 is the default python for Debian Bullseye. @PINTO0309 has [great options for armhf and various python versions](https://github.com/PINTO0309/TensorflowLite-bin), so these things can be experimented with. I simply know that these values work well for the system now -- feel free to experiment.
### The stack now
#### PHP
- Version is detected during installation
- `php` `php-fpm` `php-mysql` `php-xml`
- DB password stored in

mcguirepr89, 2022-01-16:
I did not mean to close this issue!!! I meant to keep this open in hopes this conversation can lead to a docker!

mcguirepr89, 2022-01-26:
Hello, again, @Swiftnesses  -- I just wanted to see if this was still something you are interested in?

mcguirepr89, 2022-02-07:
Closing from no response, but please get back to me if you'd like to reopen this.

closed 2022-02-07 by mcguirepr89