mcguirepr89/BirdNET-Pi #105: Microphone recommendation

reglund, 2022-01-16:
Hi! Great project! We are setting up a pi here in Norway.

I was wondering if you have a recommendation on a good enough usb microphone to use for capturing bird song.

Thank you!

nilspupils, 2022-01-16:
Hi there is a whole discussion about recommended microphones here: 
https://github.com/mcguirepr89/BirdNET-Pi/discussions/39

mcguirepr89, 2022-01-16:
Greetings, @reglund, from North Carolina, US to Norway! Thank you so much for your interest in the project and in birds!

@nilspupils is correct and has made some really great contributions in that discussion thread :)

See you over there!

Let me know if you have any trouble at all during installation and I will be more than happy to help you out! I am sorry there are not yet any language options for Norwegian (yet), but perhaps you may be able to help with that???

My best regards,
Patrick

closed 2022-01-16 by mcguirepr89

reglund, 2022-01-16:
Thank you for the quick reply! We got it up and running easy as pie!

That thread looks like a great starting point - we will have to do some reading to find good candidates.

We retrieved a list of Norwegian names for known Norwegian birds from the Norwegian ornithological society. Unfortunately that is not an exhaustive list, but it includes all the birds we are likely to encounter. Attached an updated labels.txt file with our edits (birds we did not find names for are still in English). We can of course provide our list e.g. as a csv with Norwegian/Latin/English names if that is of interest.

[labels.txt](https://github.com/mcguirepr89/BirdNET-Pi/files/7877923/labels.txt)

nilspupils, 2022-01-17:
Hi @reglund, 
there is a custom species list for Norwegian in the other thread which is almost complete!
You can find it here: 
https://github.com/kahst/BirdNET-Lite/issues/4#issuecomment-886724712

reglund, 2022-01-17:
> kahst#4 (comment)

Thank you :-)

mcguirepr89, 2022-01-17:
@nilspupils !
Thank you for that link -- I have updated the `README.md`'s Localization section per your wonderful suggestion :)
I'm working on some big changes for setup and installation and part of that will be selecting your language for both the `labels.txt` but also for the web interface :) -- cross your fingers that I am successful in my tinkerings

Patrick

mcguirepr89, 2022-01-17:
Thanks for letting me know and please do send screen shots once you are able. I think this is something I've just completely overlooked, if my hunch is correct.

- **Hunch**: The following files have `<html lang="en">` at the indicated line numbers, which could very possibly be causing issues with the allowed character-set
   Line 52 in `viewdb.php`
   Line 46 in `overview.php`
   Line 44 in `viewday.php`

The spectrograms, however, should be labeled according to what is found in `model/labels.txt` -- the script that creates those grabs the species name directly from that file. If there is some issue with those characters getting used properly, I would have expected the issue to also be visible for the Caddy directories that are created to house the files, but you're saying that the extractions that are created are labeled properly.

Quick question, is the system locale set to `de_DE.UTF-8`?

:thinking:

Svardsten53, 2022-01-17:
@mcguirepr89: We have a lot of special characters in Sweden and the other Nordic countries as well, but I have not had any problems with this. As you may remember I use English as locale, but Swedish setup for keyboard and time zone.

I look forward to the changes you are working on and do not hesitate to ask me for help if in the future help with customization or translation into Swedish is needed.

mcguirepr89, 2022-01-17:
@Svardsten53 thank you for chiming in -- it is helpful to know your settings. We'll get to the bottom of what is going on in @nilspupils' installation with this information and all of these helpful observations.

nilspupils, 2022-01-17:
First of all thanks for the attention to this problem. I am quite sure i messed the system up in some way by playing with the settings and it is not a general problem of BirdNET-Pi. Would be glad to find a solution anyways.
I have made some screenshots of the problem, in this case the bird "Rabenkrähe".

nilspupils, 2022-01-17:
[image]
[image]
[image]

nilspupils, 2022-01-17:
@mcguirepr89 : I use de_DE.UTF-8 right now. Might have been en_GB.UTF-8 before though...

mcguirepr89, 2022-01-17:
Hmmmm, very interesting.

I would be very interested to see what the actual database entries look like within the database to determine whether the correct characters are in place but not rendering properly in the web page, or if they appear messed up within the actual database.

Would you please navigate into "Tools" > "Database Maintenance" and log into the database that way?

`Server` should be `localhost`
`Username` should be `birder`
`Password` is the database password you set during installation (can be found in `birdnet.conf`)
`Database` is `birds`

Once you are on this page:
[image]
Follow the steps below:
1. Click the blue link within the left menu where it says "select" just left of "detections." When you click "select," it should look something like the screenshot below:
    [image]
2. When you are able to see parts of the database as in the screenshot above, click the "Select" option near the top and use the drop down box on the right to "Select" "Com_Name"
    [image]
3. After choosing "Select" and then "Com_Name", use the next option to the right labeled "Search." Use the boxes that you see after clicking "Search" to select "Com_Name" "LIKE%%" and then "Raben"
    [image]
4. Use the button labeled "Select" within the "Action" box to submit that search.

If I was clear enough in the steps above, that should show you _exactly_ what the database has stored for its name on this example species.

Please provide any screenshots that you can on that database view so that I might have a better idea where the data is being entered incorrectly (or rendered incorrectly -- I'm still unsure at this point as to whether the data being stored with these bad characters or is being retrieved using bad characters).

Still thinking :thinking: hmmmmmmmmm

mcguirepr89, 2022-01-17:
I misspoke earlier when I said that spectrogram labels are made from the `model/labels.txt` file -- they are not made from that file. They are made from the BirdNET-Lite .csv files produced from each analysis. My supposition is that we will find the correct characters represented therein (you can verify by looking through some of your "Processed" files at the original .csv results to see if the special characters are represented there properly). If the .csv files within your "Processed" directory have the correct characters, then `sox` may be to blame for the spectrogram labels. If the .csv files you find in "Processed" _also_ show malformed special characters, then we will need to focus the attention on the actual `model/labels.txt` file at that point.

nilspupils, 2022-01-17:
Ok, that worked. This is what i got: 
[image]

mcguirepr89, 2022-01-17:
Aha! So the entries are going into the database with the funky characters . . . which makes me think it may be due to the `labels.txt` file. The Python script that runs the analysis is also responsible for entering database entries, so it seems the analysis itself is grabbing the labels from `model/labels.txt` and in the process is mangling things.

Just to double-check something, could you change the language setting for the database maintenance page using the drop down in the top left just to verify that changing the language to German doesn't start rendering the characters properly? Thank you in advance for checking!

nilspupils, 2022-01-17:
When changing the language settings in the database page the characters did't change. 
Also i checked the .csv files and in these the characters are correctly displayed: 
[image]

mcguirepr89, 2022-01-17:
Hmmmm so NOW I'm leaning more towards this being about the system's locale settings . . . :thinking: <--- That is what I look like right now

Any chance you'd be able to forward the ports needed for me to get remote NoMachine access?? You're one of my only original testers whose system I've not accessed remotely to find problems, now that I think about it. 

I think I would need to mess about a bit in tweaking some system settings to really get an idea of which part of the system is controlling how these characters get displayed. There are several places within the software stack (forgive me if that phrase is mere jargon) that want to manage displaying information, but the bottom of the stack (the Operating System itself) is the place to start and to set that with any kind of authority that can be inherited.

Also, it may be worth running an update -- (they're pretty harmless these days -- it is no longer the whole-system-wiper it was when I was starting out implementing the update tool). 

Lastly, I wonder (to @Svardsten53's point) whether keyboard settings would affect any change?

I'm stumped, Nils!!!!! If I had a pi with me for testing, I would have more helpful info for you, but I won't be able to test for the next couple of days.

mcguirepr89, 2022-01-17:
I'm also curious if there are PHP settings that need to be adjusted to handle German properly? (Not knowing is a consequence of everything seeming to default to English -- which I have always loved and now am finding frustrating!!)

mcguirepr89, 2022-01-17:
I'm doing some research on the topic now and think there are some php/html adjustments that can be made to have things render properly -- knowing that the data is going into the .csv properly makes me thing this is 100% a rendering issue. I know that web browsers need to know which character set the document it renders what written to use in order to translate bytes to the characters, and I think this will mean more explicitly declaring the UTF-8 characterset within the php and html headers.

I may put together a few new php/html pages for you to try out that may be able to get things working for you. Once we find the declarations that work consistently, I can put that new knowledge to use in cleaning up the code and in my efforts to add localization support during installation and configuration.

nilspupils, 2022-01-17:
Patrick, you are more than welcome to visit my Pi! I did uninstall Nomachine before, so i would have to install it again. Also i have basically no clue how to forward the ports as i have never done it... I would be up for a challenge though!

mcguirepr89, 2022-01-17:
That's great!!! Allow me to help :)

First, begin by reinstalling nomachine, which should be something like this in the command line:
`sudo apt -y install nomachine`

If that doesn't work, for whatever reason, copy and paste the following into the terminal and the original NoMachine installation process from installation will re-run:
```
nomachine_url="https://download.nomachine.com/download/7.7/Arm/nomachine_7.7.4_1_arm64.deb"
install_nomachine() {
  if [ ! -d /usr/share/NX ];then
    echo "Installing NoMachine"
    curl -s -o ${HOME}/nomachine.deb -O "${nomachine_url}"
    apt install -y ${HOME}/nomachine.deb
    rm -f ${HOME}/nomachine.deb
  fi
}
sudo HOME=/home/pi/ install_nomachine
```

Hopefully, though, you should just be able to run the `sudo apt -y install nomachine` command.

After that, open the NoMachine server window by clicking the red NoMachine icon that appears in the desktop panel. Select the option to view the server status.

That window will have a list of addresses to which I would be able to connect remotely. Often, one or more of the addresses will be local-only (often these are 192.168.x.x or 10.80.x.x in range) -- there should (hopefully) also be an address option (typically the bottom-most of the address list) that is remote.

Go ahead and get to that point and email me at mcguirepr89@gmail.com with what you see and we will go from there in a private email.

My best regards!!!

mcguirepr89, 2022-01-17:
Also, check out the database view on this installation that is in German -- it looks like the special characters are displaying without any issue!! Very promising! https://birdnetgv.ddnss.de/

mcguirepr89, 2022-01-17:
And with that finding I bet I know what the issue is!!! You're using my pre-installed image that was build in English -- the installation above was built in German . . . coincidence or explanation? Hard to say

nilspupils, 2022-01-17:
Nomachine is installed, the red icon is missing though...

mcguirepr89, 2022-01-17:
reboot real quick -- the VNC icon likes to fight for a spot with the NoMachine icon

nilspupils, 2022-01-17:
Wrote you a mail, it is getting late though... Proceed tomorrow??

mcguirepr89, 2022-01-17:
That will be just fine! Email me tomorrow when you either know a good window of time or when you are ready and I will do what I can to also be ready. Also, I replied to your email and also suggested we perhaps continue tomorrow. Talk to you then!

nilspupils, 2022-01-17:
Lets talk tomorrow! Many thanks for all your efforts!
Best, Nils

nilspupils, 2022-01-19:
@mcguirepr89: Thanks for the great session yesterday! Will report after the new installation.

mcguirepr89, 2022-01-19:
My pleasure!!! Sorry for blowing up your issue thread, @reglund!!