mcguirepr89/BirdNET-Pi #106: Feature request: extractions of same length as recordings

DD4WH, 2022-01-18:
Hi Patrick, I am still very happy with BirdNET-Pi and it is a very valuable tool for me in monitoring Pygmy Owls, as you know. I discovered they are more active in Janauary than in March (where most of the monitoring is regularly done) . . . :-) and my remote BirdNET-pi has been running successfully since November 2021 now!

This is a request for a "new"/modified feature:

* if EXTRACTION_LENGTH is equal to RECORDING_LENGTH, I would like BirdNET-Pi to only "extract" the original recordings, not more
* my standard setting is to set both values to 30 seconds: at the moment. However, this results in several redundant ("overlapping") recordings. But I would like to have extracted recordings that do not have the same audio in common. So, the best and easiest would be to just take the original recording (of 30sec length in my case) and save this as the "extractions", even if there is only one call in one of the original recordings
* This behaviour could/should be available by setting a software switch in birdnet.conf

This would be awesome, if you could implement it! I tried to fiddle with the script file extract_new_birdsound.sh line 115ff., but my knowledge in shell script language is very poor, so your help would be very much appreciated !

Best regards, Frank DD4WH

mcguirepr89, 2022-01-18:
@DD4WH Hi, Frank! So great to hear from you and to get the update on your long-running setup! November! That's great! It is also really gratifying to hear that you have learned new behavioral patterns for the Pygmy Owls from using BirdNET-Pi -- what a wonderful outcome.

Thank you also for this thoughtful and, as always, very clear and manageable request. I _should_ be able to implement this for you. I am still away from my installation and forgot to arrange remote SSH access to it, so I cannot reach it for testing at this time. But here is what I'm thinking:

- One way that I think would achieve what you are going for may be to simply discontinue reporting the confidence score within the {NEW_FILE} name (line 67).
  ```
   NEWFILE="${COMMON_NAME// /_}-${CONFIDENCE_SCORE}-${OLDFILE}"
  ```
  could look like
  ```
   NEWFILE="${COMMON_NAME// /_}-${OLDFILE}"
   ```
I believe this would achieve what you are hoping for (in the short term without a `birdnet.conf` option). The thought behind this is that there is a check in the script that will _not_ make an extraction if any extraction with _the exact same name_ already exists. Most of the time, each detection within a recorded audio clip will receive different confidence scores, so `extract_new_birdsounds.sh` will see the difference in confidence score and will create a new extraction that reflects that difference. If the extracted file names do not report the confidence score, however, the script (should) only make 1 extraction per recording per species. That way, if 2 distinct species are detected within the same original audio clip, an extraction will be made for each species. However, if 2 detections of the same species are detected within the same original audio clip, only 1 extraction will be made for that species for that recording.

The method above may work (it may not, but I think it will), though I know it is not the request you put forward. Likewise, I would like to devise a method that retains the highest confidence score per species per recording so that you know what the _most confident_ detection score was for a given species within a recording.

I'm going to keep this open as I tinker with possibilities. In the meantime, if you are able to make the adjustment above to test, it would be really helpful. If you are unable to test that, please don't worry as I should be back with my installation later this evening or tomorrow.

My best regards as always!
Patrick

DD4WH, 2022-01-19:
Hi Patrick, thanks a lot for your very quick and very helpful answer.
I tried that and it worked ! I now get one extraction each for every recorded file.
However, the length of the files differ, they are shorter than the original files . . . What I would like is to extract the original file with its original length. 
Maybe you also have a quick fix for that ?

DD4WH, 2022-01-19:
Yes, that hard-coded 0 to 30s works well! 
Thank you, it already runs on my remote BirdNET-Pi :-)
Of course, you are right, would be good to be able to set this behaviour in birdnet.conf

mcguirepr89, 2022-02-01:
Quick small update:
This is built into `birdnet_analysis.sh` now, so anytime the `${RECORDING_LENGTH} == ${EXTRACTION_LENGTH}`, `${START}` will `=0` and `${END}` will `=${RECORDING_LENGTH}`

I am still working on having only 1 extraction per recording with the highest confidence score as the keeper.

mcguirepr89, 2022-02-23:
I think this was addressed -- please let me know if I am wrong about that, but it seems now that you should be able to set the extraction length to equal the recording length to achieve the desired results. I'll reopen if this is not the case.

closed 2022-02-23 by mcguirepr89