mcguirepr89/BirdNET-Pi #111: How to handle wifi-access "in the field"?

joknjokn, 2022-01-29:
Hi,

I want to try BirdNET-pi in sessions in the field, running from a powerbank.

I'm now trying to figure out how to handle wifi/access to the RPi. Any experiences with this?

I tried "sudo raspi-config" to connect the RPi to my smartphone hotspot hoping I could open birdnetpi.local from that, but it doesn't work. However, it does have a connection. I used an app to find the IP of the RPi on my phones' network, and opening that ip with :8080 opened the green BirdNET-Pi log screen. :80 just showed a white page. So, not really a solution yet. I installed BirdNET-Pi by first installing RaspiOS, then the "newinstaller".

I think the optimal solution would be:
1. The website (birdnetpi.local) should work when connected to a phone-hotspot as well (no clue if possible?)
2. If the RPi doesn't find a network, it creates its' own hotspot (how to handle password?) where birdnetpi.local works. I think I saw this mentioned as a feature in another thread, but unsure how to configure it, or if it's even available with my installation method?

Bonus info:
Just got my RPi 4 2gb mounted in an Argon Neo case without a fan. Running at 48-51 degrees. Very smooth! Hoping to run it with a good powerbank in the field.

joknjokn, 2022-01-29:
Also: When turning off the hotspot of my phone, the RPi doesn't automatically connect to my home wifi afterwards.

joknjokn, 2022-01-29:
A few other ideas for use in the field (I guess it's a suggestion to add something to the documentation, if anyone has a solution... I will probably try at some point):
- Adding an indicator LED that shows if BirdNET-PI is running (and working). The RPi takes some time to boot, and things can go wrong, so a simple indicator would be nice. I could probably connect an LED myself, but unsure how to turn it on when BirdNET-Pi is running.
- Shutdown-button to safely shut the RPi down instead of unplugging a powerbank. I guess I can find a bunch of tutorials for this.

nilspupils, 2022-01-30:
Hi @joknjokn, just a quick info that most of your issues will be resolved by using the pre-installed image, which offers a hotspot when no network is found. Please have a look here:
Btw. i like your idea with the LED as a simple status display. This could be used for the other functions as well.
About powering the Pi with a powerbank this will be difficult as the Pi is quite power hungry. Even with a large powerbank only a couple of hours will be possible at best.

mcguirepr89, 2022-01-30:
I am coming back to this with a nice long response, stay tuned!

mcguirepr89, 2022-01-30:
Hi, @joknjokn and thank you for the great question. Thanks, too, to @nilspupils for pointing out where we've discussed this possibility before.

@nilspupils is correct that the exact feature you're hoping for was already included in the pre-installed image for version 0.10. Version 0.11 (currently known as the testing `forms` branch) will be including an interface to enable this auto-hotspot feature manually. Once I've really finished development on v0.11, I will create a new pre-installed image for it that will include the auto-hotspot feature.

The basics are that when you first boot up the pre-installed image, it will present its own hotspot to which you would connect in the field. From that connection, you would set the system time (since RPis don't have RTCs built in) and then give it the LAT and LON it needs to get started. From that hotspot, you have full access to the installation. Likewise, you can use that hotspot connection to configure a WiFi connection to a WiFi network. The goal is that the pre-installed image would work as both a quick way to get the installation up and running, but also a way for fully remote field work to function as needed without the need for a host network other than the one the BirdNET-Pi can provide.

You can take a look at the software that I use for this feature [here](https://github.com/gitbls/autoAP). I customize it a bit before making it part of the base image, but it works well and can be used as is.

>Adding an indicator LED that shows if BirdNET-PI is running (and working). The RPi takes some time to boot, and things can go wrong, so a simple indicator would be nice. I could probably connect an LED myself, but unsure how to turn it on when BirdNET-Pi is running.

This would be very easy to implement using the GPIOs, but it is no longer (reasonably) possible to control the Pi's built in LEDs. If you put something together that uses this (probably using the python RPi.GPIO library), let me know and I can work with you on getting it to indicate whatever you need.

My best regards,
Patrick