mcguirepr89/BirdNET-Pi #112: Hostnames on the same network

davidmam, 2022-01-29:
The scripts force the hostname of every birdnetPi to be birdnetpi. This causes issues if you are trying to set up more than one on the same network.

It fails to resolve properly. I'm just getting a plain blank page.

I've done a wee hack to allow the setting of a custom hostname which then allows two BirdNET pi on the same network. Set the config variable BIRDNET_HOST to what you want, and that will be the name of the machine. This will be important if I am putting out a number on the same network and wanting unique hostnames.

davidmam, 2022-01-29:
I've created a pull request with the changes. It appears to work for me. I may have missed some somewhere - will have another scout through.

davidmam, 2022-01-29:
OK, there may also be an issue with the naming in the Caddyfile with extractionlog.local and so on. (templates/Caddyfile_service
) There are also the hardcoded links in the html files. I'll look at a shell script to rewrite those with the correct hostname.
Probably put them in a homepage-templates and then a sed to copy them over to the homepage directory?

nilspupils, 2022-01-30:
Hi davivmam, i am sure Patrick @mcguirepr89 will chime in on this, but until then you can have a look at our discussion section, where we have a specific topic about it. See here https://github.com/mcguirepr89/BirdNET-Pi/discussions/89
I hope this can point you in the right direction!

davidmam, 2022-01-30:
> CAUTION: This email originated from outside the University of Dundee. Do not click links or open attachments unless you recognise the sender's email address and know the content is safe.

I've put in a pull request with an updated set of scripts that seems to work. Next task is to testshutdown and start up on solar power, then on a cellular modem

..d

Dr David Martin
Senior Lecturer in Bioinformatics
College of Life Sciences
University of Dundee

The University of Dundee is a registered Scottish Charity, No: SC015096

mcguirepr89, 2022-01-30:
Howdy! Great idea!

I've looked through the pull request and really appreciate all of your work. I won't be accepting it, however, but this is because you and I worked on the same thing at the same time!! I added this feature yesterday morning, but have taken a different approach.

My approach utilizes the actual device's hostname and reconfigures the system to use the hostname as its reachable URL (and avahi-alias).

I'd love to share with you the few places I've altered to get this working so that we can compare notes -- I will only highlight the changes:
I've edited `install_services.sh` and `update_services.sh` with nearly identical changes, with one exception.

`install_services.sh` & `update_services.sh`
For `install_services.sh`, I changed the `set_hostname` function to only take place when the current hostname is still the default "raspberrypi". For `update_services.sh`, I've removed this function altogether.
```
...
set_hostname() {
  if [ "$(hostname)" == "raspberrypi" ];then
    echo "Setting hostname to 'birdnetpi'"
    hostnamectl set-hostname birdnetpi
    sed -i 's/raspberrypi/birdnetpi/g' /etc/hosts
  fi
}
...
```
Then, I changed the `install_avahi_aliases` function to utilize the results from `hostname` as its avahi-alias mDNS announcement.
```
...
install_avahi_aliases() {
...
systemctl enable avahi-alias@$(hostname).local.service
}
...
```
Then I replaced all instances of "`birdnetpi.local`" with "`$(hostname).local`" in the `create_necessary_dirs` function (this function is sloppy, so I apologize that it is so difficult to read).
```
...
create_necessary_dirs() {
  ...
  if [ -z ${BIRDNETPI_URL} ];then
    sudo -u${USER} sed -i "s/birdnetpi.local/$(hostname).local/g" $(dirname ${my_dir})/homepage/*.html
    sudo -u${USER} sed -i "s/birdnetpi.local/$(hostname).local/g" $(dirname ${my_dir})/scripts/*.html
    sudo -u${USER} sed -i "s/birdnetpi.local/$(hostname).local/g" $(dirname ${my_dir})/scripts/*.html
    sudo -u${USER} sed -i "s/birdnetpi.local/$(hostname).local/g" $(dirname ${my_dir})/scripts/*.php
    sudo -u${USER} sed -i "s/birdnetpi.local/$(hostname).local/g" $(dirname ${my_dir})/scripts/*/*.php
  else
...
...
```
Then I edited the portion that creates the Caddyfile so that it will respond to the `hostname`
```
...
install_Caddyfile() {
  ...
  cat << EOF > /etc/caddy/Caddyfile
http://localhost http://$(hostname).local ${BIRDNETPI_URL} {
  root * ${EXTRACTED}
  file_server browse
  basicauth /Processed* {
    birdnet ${HASHWORD}
  }
  basicauth /scripts* {
    birdnet ${HASHWORD}
  }
  basicauth /stream {
    birdnet ${HASHWORD}
  }
  basicauth /phpsysinfo* {
    birdnet ${HASHWORD}
  }
  reverse_proxy /stream localhost:8000
  php_fastcgi unix//run/php/php${php_version}-fpm.sock
}
EOF
  else
    cat << EOF > /etc/caddy/Caddyfile
http://localhost http://$(hostname).local ${BIRDNETPI_URL} {
  root * ${EXTRACTED}
  file_server browse
  reverse_proxy /stream localhost:8000
  php_fastcgi unix//run/php/php${php_version}-fpm.sock
}
EOF
  fi
...
```

Those changes allow you to change (or pre-set) the device's hostname and then running an update (or the initial installer) will ensure everything that usually defaults to "birdnetpi.local" will be using your preferred hostname instead.

I hope that is clear enough to follow and helpful in its use!
My best regards,
Patrick McGuire

mcguirepr89, 2022-01-30:
Closing this now that the feature is part of the `forms` branch

davidmam, 2022-01-30:
Absolutely no problem and great to see this in. You know your code better than I do so have a more elegant solution.
I look forward to trying that out later today - I'll start over with the forms branch.

..d

Dr David Martin
Senior Lecturer in Bioinformatics
College of Life Sciences
University of Dundee

mcguirepr89, 2022-01-30:
There is unfortunately a breaking issue upstream -- the tflite.whl is not installing. I've opened an issue with the wheel maintainer and hope to have a solution soon.

Sorry for the inconvenience!

closed 2022-01-30 by mcguirepr89

davidmam, 2022-01-30:
Looks like a pip issue. This has broken my current install effort.

..d

Dr David Martin
Senior Lecturer in Bioinformatics
College of Life Sciences
University of Dundee

mcguirepr89, 2022-01-30:
Yeah :/ I'm not sure what the fix is going to be for this -- that bin wheel is the working piece of the whole system! It seems that the wheel is looking for a version of numpy that is no longer available. numpy causes so many issues environment issues with TF. There are certainly alternatives, but I would like this to get back to working ASAP instead of having to switch gears completely to reworking how to get the TFLite binaries. Ugh

mcguirepr89, 2022-01-30:
https://github.com/pypa/pip/issues/10825 that seems to be the actual issue -- numpy is still a culprit, but the deprecation of html5lib is messing things up across `pip`

davidmam, 2022-01-30:
OK, I'll have to put this on hold for now till you have a working version.

No problem, I'll pick it up again once it is back in the land of the living.

..d

Dr David Martin
Senior Lecturer in Bioinformatics
College of Life Sciences
University of Dundee

pypa/pip#10825<https://github.com/pypa/pip/issues/10825> that seems to be the actual issue -- numpy is still a culprit, but the deprecation of html5lib is messing things up across pip

mcguirepr89, 2022-01-30:
Found a solution until pip fixes things -- I'm putting it in the code now

davidmam, 2022-01-30:
Oh, superb. I've just left my friend's house where we were setting up a basic birdnet setup. I'll get back to that one later in the week and hopefully it will all run nicely.

..d

Dr David Martin
Senior Lecturer in Bioinformatics
College of Life Sciences
University of Dundee

davidmam, 2022-01-30:
I think the generic term for this kind of event is "Nebraska'ed"

xkcd: Dependency<https://xkcd.com/2347/> if you want the derivation of the term.

..d
[https://imgs.xkcd.com/comics/dependency_2x.png]<https://xkcd.com/2347/>
xkcd: Dependency<https://xkcd.com/2347/>
This work is licensed under a Creative Commons Attribution-NonCommercial 2.5 License. This means you're free to copy and share these comics (but not to sell them). More details..
xkcd.com

Dr David Martin
Senior Lecturer in Bioinformatics
College of Life Sciences
University of Dundee