mcguirepr89/BirdNET-Pi #115: Feature request: "Urban mode"

DD4WH, 2022-01-30:
Many (if not most!) BirdNET-Pi users live in an urban environment and just hang the microphone out of their windows. This works very well, but also tends to capture high numbers of noise events commonly found in urban environments/cities. These can be police sirens, helicopters, planes, cars, and other technical noise but also child screaming and human talk.
Often, these noises are identified by BirdNET as bird species and this is somehow predictable, because these urban sound events have a striking similarity with certain bird calls/songs. One example is the Great Bittern, which emitts an extremely low sinusoid tone at 150Hz. This same pattern is produced by many starting cars, helicopters, aeroplanes etc., so the probability of having an ID of a bittern in the centre of a city is really high. But this bird species (specialized on specific wetlands) is extremely unlikely to emitt its breeding bird song in the centre of a city. So how do we handle these FALSE POSITIVES ? One solution could be the following:
URBAN MODE:
* with a switch in the Advanced settings one could switch on or off the "urban mode"
* "Urban mode" switches on a blacklist (ie. a list of species that would be set to Confidence == 0, whenever they will be identified, because these species are 1.) extremely unlikely to be encountered in a city, and 2.) their sounds are pretty similar to sound events in cities.

For Europe, these species (which should be included in such a list) would be:

Botaurus stellaris_Great Bittern
Cygnus cygnus_Whooper Swan
Cygnus olor_Mute Swan (little children are often identified as Whooper Swans)
Tetrao tetrix_Black Grouse
Tetrao urogallus_Western Capercaillie
Tetrastes bonasia_Hazel Grouse
Bubo bubo_Eurasian Eagle-Owl
Ciconia ciconia_White Stork
Ciconia nigra_Black Stork
Caprimulgus europaeus_Eurasian Nightjar
Aegolius funereus_Boreal Owl

For urban environments, this switch could eliminate a very large proportion of annoying false positive identifications associated with urban anthropogenic noises.

A big disadvantage would be that we would need separate blacklists/species additions for separate biogeographic regions in the world.

mcguirepr89, 2022-01-30:
Hi, Frank! That is a great idea -- I will work on integrating @CaiusX's work on [making an exclusion list](https://github.com/mcguirepr89/BirdNET-Pi/issues/45)! That Tiny File Manager is so handy in being able to setup `custom_species_lists.txt` files, but I think this is definitely the way to go for most users. 

I think the setting might be in the "Basic Settings" and in the `firstrun.html` that will be part of an eventual Captive Portal (like when you connect to WiFi at a hotel and get forced to go to a webpage to enter some information and agree to some terms) that will allow users to set things up from the "BirdNET-Pi Hotspot" to easily configure a pre-installed image from a web browser, whether a phone or computer.

Ideally, I'd like there to be 3 options:
1. Urban
2. Suburban
3. Field (Research)

Wherein the first is as you've mentioned, the second contains less of the first, but some that would never be around humans and their sounds (I don't even know if that's a thing, but would assume as much), and the third not excluding any species for research purposes.

>A big disadvantage would be that we would need separate blacklists/species additions for separate biogeographic regions in the world.

I don't think this is a big disadvantage at all -- I think it will just be one-continually growing, user-curated list of non-city-dwelling species _(until I learn how @timsterc does his post-processing :smile:).

I'll keep you posted here on how things are shaping up and how this idea will get included -- keep sending any additions or amendments to this idea as you think on it.

My best,
Patrick

mcguirepr89, 2022-01-30:
I finally brought @CaiusX's work to accept an `eclude_species_list.txt` and `include_species_list.txt` into `forms` for testing. So far so good. It looks for either file, and uses them if they're found (either one, or both). Will work on putting this into the webGUI for easier adminstration.

nilspupils, 2022-01-31:
I have some time on my hands (thanks Corona!) so i compiled a sample "exclude_species_list.txt" mostly with the above species.
[exclude_species_list.txt](https://github.com/mcguirepr89/BirdNET-Pi/files/7970692/exclude_species_list.txt)
Note: the list is in german....

mcguirepr89, 2022-01-31:
Thanks, @nilspupils! And for all the swift feedback you are providing to users -- it is _greatly_ appreciated!

mcguirepr89, 2022-01-31:
Update:
The species lists are now easily maintained in the web gui (I haven't messed with it on my phone, yet, so I don't expect it to look very good there).
[image]
[image]
[image]
[image]
[image]
[image]

nilspupils, 2022-02-01:
Looking good, will these changes install with an update? Also, since i already prepared these lists, where would i put them?
Thank you and best regards,
Nils

nilspupils, 2022-02-01:
Updated and seems to have worked. Ok, figured it out, i just copied the files in the BirdNET-Pi directory, where they were recognized. I was not able to add or remove from my lists, perhaps because of latin/german language and a mixup between the different versions?
Best, Nils

mcguirepr89, 2022-02-01:
>Somehow i was not able to add entries from my lists, removing worked though, perhaps because of latin/german language and a mixup between the different language versions? 

I suppose this is due to file permissions. Run 
```
chmod g+rw {~/BirdNET-Pi/include_species_list.txt,~/BirdNET-Pi/exclude_species_list.txt}
```
After that, you should be able to edit the file using the web gui tool.

>Also i noticed the labels.txt file still gets overwritten when updating, i remember it to be excluded?

It _is_ supposed to be excluded from updates! It probably has something to do with _how_ I have updates working. I've never made software before, so my update process is total crap (I would imagine).

Lots more comments to address! TTYS! Best regards!
Patrick

nilspupils, 2022-02-01:
Adding and removing works now! Also i just noticed that the new enties get appended at the end of the list, not in alphabeical order. I just did't look there before, and only now found the entries. So it might have worked from the beginning.... 
Greetings from isolation, 
Nils

mcguirepr89, 2022-02-01:
hahaha I hadn't considered adding a `sort` into the mix, we'll see how easy or difficult that is :)

mcguirepr89, 2022-02-01:
@Molokai 
This is in the newest version of the `forms` branch. The files can be created and managed using the "*Species List" tools on the "Tools" page.

The format is indeed simply copying and pasting _whatever_ is used in your `labels.txt` file. So just find the species of interest within the `labels.txt` file and paste that entire line in your `include_species_list.txt` or `exclude_species_list.txt` as needed. Though the web GUI is much easier.
The photo below shows the `labels.txt` on the left side and the `exclude_species_list.txt` file on the right.
[image]

Molokai, 2022-02-01:
I'm completely blind! Found it in the UI now!

mcguirepr89, 2022-04-15:
I've kept this issue open because it is a great feature request! Here is a small update:

I'm thinking that I will be using the new repo's ["`species.py`"](https://github.com/kahst/BirdNET-Analyzer/blob/main/species.py) tool to start tinkering with being able to put together regional lists that may be able to evolve into the "Urban Mode" we've discussed.

josefdlange, 2022-05-13:
A neat addition to this setup would be the ability, from a given detection spectrogram (where it lists the species detected), is the ability to begin excluding that species then and there. Just a li'l button that can exclude it.

It would also be nice to be able to tell the neural net "no, you were wrong" to improve future matches, but I imagine BirdNET operates on a pre-computed training set and doesn't accept new improvements?

aligator, 2022-07-16:
As the https://github.com/kahst/BirdNET-Lite is deprecated and quite old. Would it be possible to build a tflite model from https://github.com/kahst/BirdNET-Analyzer? 

I noticed that uploading false positive files to https://birdnet.cornell.edu/api/ results in much more valid results.

closed 2022-12-13 by mcguirepr89

andrew-wylie, 2023-02-08:
> As the https://github.com/kahst/BirdNET-Lite is deprecated and quite old. Would it be possible to build a tflite model from https://github.com/kahst/BirdNET-Analyzer?
> 
> I noticed that uploading false positive files to https://birdnet.cornell.edu/api/ results in much more valid results.

Same here. I uploaded a file which is being identified as a "Northern Saw whet Owl", but it was a truck reversing :) ... we do get these owls in Vancouver (BC) so I can't exclude it, and that api identified the sound as mostly human made origin.