mcguirepr89/BirdNET-Pi #121: Danish translation of species - Can I help?

joknjokn, 2022-01-31:
Hi,

Can I help with translating the species to Danish? How? I have the name list of birds observed in Denmark, used by the Danish Ornithological Society.

nilspupils, 2022-01-31:
Hi @joknjokn, 
thanks for the offer. There is a Danish list here (scroll completely down):
https://github.com/mcguirepr89/BirdNET-Pi 
Some species are not translated yet, and i think a completion of the list would be great. (It might be difficult though).
Best, Nils

joknjokn, 2022-01-31:
Oh, right - thanks for the reply. I did set the database language to Danish, but I had some species appear in english, i.e. "Great Tit", so I guess maybe that observation was just from before I changed the language then. I'll close this issue :)

closed 2022-01-31 by joknjokn

mcguirepr89, 2022-02-01:
@joknjokn I haven't yet had the opportunity to respond to your very generous offer. @nilspupils is right about the database language options, but I would _love_ and appreciate your help in translating the web pages! Do you feel comfortable editing web page source code? If not, I can send you over a list of just english phrases and words that need translating and anything you can provide would be amazing.

Let me know what you think, but no pressure!

My best regards,
Patrick

joknjokn, 2022-02-01:
@mcguirepr89 Yes, I can translate it. I would probably prefer doing it from a list :)

mcguirepr89, 2022-02-01:
Sounds great! I'll send you some stuff soon.
Thanks you!

mcguirepr89, 2022-02-01:
If you could do your best with this list of translations, it would help tremendously! No pressure and no worries if you decide not to (either way, let me know).
[forjoknjokn.txt](https://github.com/mcguirepr89/BirdNET-Pi/files/7981698/forjoknjokn.txt)
That file has (pretty much) all of the English words and phrases from BirdNET-Pi's webGUI.
I've left the file the line lives in so that I can keep track of where they go. If you could just put the danish to the right of the equals sign, it would be a huge help. Please feel free to skip any that are redundant -- I tried weeding those out, but I'm sure I left a few.

Thank you again so much for your kind offer, tremendous enthusiasm, and awesome contributions to this project.
My best!
Patrick

joknjokn, 2022-02-01:
> If you could do your best with this list of translations, it would help tremendously! No pressure and no worries if you decide not to (either way, let me know). [...]

Here we go. I will probably have some corrections when I've experienced it in live UI. Also a bit unsure about a few of the words, since the danish translation feels a bit off, and the alternative is not so close to the meaning of the english word. We'll see.
[forjoknjokn.txt](https://github.com/mcguirepr89/BirdNET-Pi/files/7982143/forjoknjokn.txt)

joknjokn, 2022-02-03:
@mcguirepr89 - Did you see above? :)

mcguirepr89, 2022-02-03:
@joknjokn Yes!! Sorry for the silence on my end, but it looks great!
Which were you unsure about? I may be able to clarify some of the small phrases or what is meant by them.

joknjokn, 2022-02-12:
@mcguirepr89 Btw (not entirely relevant to this ticket): 
I'm a little confused about which version to install if I want the newest stuff you add. There's both the main-version and the forms-version. What's the idea behind forms, and is it still relevant? 

Could it be an idea to only use 1 lane in terms of installation, and then maybe add an "Experimental" opt-in checkbox in the settings/tools? So that everyone installs the same thing, but we can opt-in to experimental/non-stable stuff through the website/UI.