mcguirepr89/BirdNET-Pi #130: Problem on PI3B+

gitalm, 2022-02-08:
Dear Patrick,
Sometimes the analyzing stops.

[image]

mcguirepr89, 2022-02-08:
Hi, @gitalm  -- could you please check the "System" > "Info" page and let me know what the available storage and memory are?

gitalm, 2022-02-08:
THanks for the hint, I deactivated BirdWeather and restarted the service. Let's see

mcguirepr89, 2022-02-08:
It looks like the issue was in _name resolution_, meaning your network connection was not able to come up with an IP address for app.birdweather.com. You might need to set a DNS on your cellular modem for it to resolve domains correctly.

gitalm, 2022-02-08:
Dear Patrick, the BirdNET Pi was just in  a normal WIFI.
Next step would be to use a cellular modem K5610.
Hope it works out.

Best wishes

mcguirepr89, 2022-02-08:
Let me know what you find and we'll see what else there might be to look into more closely.

gitalm, 2022-02-08:
Dear Patrick, it is now working without Problem. I just deleted the BirdWeather ID and restarted. 
The problem is perhaps the weak WIFI signal, therefore the box is a bit away from my house.
Thanks for helping. WIll start the test with cellular modems to be free ;-)

closed 2022-02-09 by mcguirepr89