mcguirepr89/BirdNET-Pi #131: json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

mcguirepr89, 2022-02-08:
Hi, @timsterc!

I can't believe I haven't tested this yet, but it seems like recording lengths at and above 45 seconds (still testing what the true threshold is here, but somewhere greater than 30 and at and above 45) are not posting to app.BirdWeather.com properly.

This error appears right after writing to the database when detections above the minimum confidence occur -- it appears the response from the POST from your `nginx` server is that the header size is too large.
```
Soundscape POST Response Status -  413
Exception in thread Thread-11:
Traceback (most recent call last):
  File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/requests/models.py", line 910, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/lib/python3.9/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.9/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.9/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/BirdNET-Pi/./scripts/server.py", line 382, in handle_client
    sdata = response.json()
  File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/requests/models.py", line 917, in json
    raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
requests.exceptions.JSONDecodeError: [Errno Expecting value] <html>
<head><title>413 Request Entity Too Large</title></head>
<body>
<center><h1>413 Request Entity Too Large</h1></center>
<hr><center>nginx/1.18.0 (Ubuntu)</center>
</body>
</html>
: 0
```

I will let you know what I find to be the true threshold for that, but I'm thinking there is a way to handle this elsewhere -- perhaps on the server side in `nginx`'s allowable header size limit, or in the code itself, chopping the audio and spectrograms into smaller POSTs??

Anywho, let me know what else I can provide for you, or if this needs to be a hard limit when having BirdWeather enabled.

My best,
Patrick!

timsterc, 2022-02-08:
In order to keep the audio storage reasonable, probably best to think about ways of breaking the audio into chunks. It would be possible to just use the 3sec chunks that it's already broken into in analyze.py ... I landed on using the entire recording_interval as a way to gain context for an identification that was in question.

mcguirepr89, 2022-02-08:
I figured -- it's not a very big deal for now, but I imagine it may be necessary to make these adjustments to let the RPi0W2 play along.

I'm not good enough at Python to make those types of changes, so I'm hoping you'll be taking a crack at it whenever you have some free time. Also, @CaiusX might be able to take a closer look and see if the chunks can be created to allow a certain amount of context (10-15 seconds, maybe more?).

Thanks for responding so quickly!

closed 2022-02-08 by mcguirepr89