mcguirepr89/BirdNET-Pi #135: Experimenting with RPi Zero 2W

DD4WH, 2022-02-09:
Patrick, thanks again for the accurate installation info for RPi Zero 2W. [HERE](https://github.com/mcguirepr89/BirdNET-Pi/discussions/122#discussioncomment-2134221)
I open this issue to have a place specifically for feedback on how BirdNET-Pi v0.10 is performing on this hardware. Feel free to close this issue or move it to another -more appropriate- place :-).

These are my experiences running BirdNET-Pi for half a day now:
* BirdNET-Pi works well and seems to keep up with analysing the files in the first hour or so
* when the Pi heats up, it reaches about 60 degrees Celsius and the load for the four processors goes up
* however, that does not influence the time for the TFlite analysis which seems to stay at about 12 seconds for a 30 second recording
* compared to BirdNET-Pi on RPi4B and RPi3B+, the web interface is (much) slower to respond and sometimes after updating or reboot, one has to wait for a minute or even longer until everything runs smoothly again

A serious issue is that after running for one hour or a bit longer, BirdNET-Pi seems to hang and no files are being processed any more. The log shows this:
`Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/bin/server.py", line 315, in handle_client
    writeResultsToFile(detections, min_conf, args.o)
  File "/usr/local/bin/server.py", line 204, in writeResultsToFile
    with open(path, 'w') as rfile:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/BirdSongs/February-2022/08-Tuesday/2022-02-08-birdnet-23:57:08.wav.csv'

I experienced this three times yesterday and this night. So it seems it is a reproducable issue.

After rebooting, log and analysis still seem to be stuck:
`    with open(path, 'w') as rfile:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/BirdSongs/February-2022/08-Tuesday/2022-02-08-birdnet-23:57:08.wav.csv'
birdnet_server.service: Main process exited, code=killed, status=9/KILL
birdnet_server.service: Failed with result 'signal'.
Stopped BirdNET Analysis Server.
birdnet_server.service: Consumed 18min 48.826s CPU time.
Started BirdNET Analysis Server.
INFO: Created TensorFlow Lite XNNPACK delegate for CPU.
INFO: Created TensorFlow Lite delegate for select TF ops.
INFO: TfLiteFlexDelegate delegate: 1 nodes delegated out of 182 nodes with 1 partitions.`

After RESTART ALL SERVICES it seems it has resumed analysis and shows the usual output, which confirms it really takes the real time recordings and processes without lag:
` ('Perdix perdix_Rebhuhn', 0.049771078)
 ('Perdix perdix_Rebhuhn', 0.0641239)
 ('Perdix perdix_Rebhuhn', 0.04529362)
 ('Perdix perdix_Rebhuhn', 0.051011525)
[NEW CONNECTION] ('127.0.0.1', 33360) connected.
[ACTIVE CONNECTIONS] 1
READING AUDIO DATA... DONE! READ 12 CHUNKS.
FULL FILENAME: -/home/pi/BirdSongs/February-2022/09-Wednesday/2022-02-09-birdnet-08:50:33.wav-
Date-time: 2022-02-09 08:50:33`

Hope this helps in finding the issue. Keep up the excellent work!

mcguirepr89, 2022-02-09:
>A serious issue is that after running for one hour or a bit longer, BirdNET-Pi seems to hang and no files are being processed any more. The log shows this:
```
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/bin/server.py", line 315, in handle_client
    writeResultsToFile(detections, min_conf, args.o)
  File "/usr/local/bin/server.py", line 204, in writeResultsToFile
    with open(path, 'w') as rfile:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/BirdSongs/February-2022/08-Tuesday/2022-02-08-birdnet-23:57:08.wav.csv
```
>I experienced this three times yesterday and this night. So it seems it is a reproducable issue.

The error in log excerpt above is indicating that there is something wrong with the `birdnet_recording.service`. The log excerpt ends the traceback with the `FileNotFoundError`, which means the `birdnet_recording.service` stopped for some reason. The next log excerpt supports this conclusion, as the "Restart ALL Services" script restarts the `birdnet_recording.service` and the `FileNotFound` error disappears.

>After rebooting, log and analysis still seem to be stuck (the date and time of the file still shows a very old file):
```
with open(path, 'w') as rfile:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/BirdSongs/February-2022/08-Tuesday/2022-02-08-birdnet-23:57:08.wav.csv
birdnet_server.service: Main process exited, code=killed, status=9/KILL
birdnet_server.service: Failed with result 'signal'.
Stopped BirdNET Analysis Server.
birdnet_server.service: Consumed 18min 48.826s CPU time.
Started BirdNET Analysis Server.
INFO: Created TensorFlow Lite XNNPACK delegate for CPU.
INFO: Created TensorFlow Lite delegate for select TF ops.
INFO: TfLiteFlexDelegate delegate: 1 nodes delegated out of 182 nodes with 1 partitions.
```

If you checked these logs right after the reboot, then the portion at the top of the log excerpt above is reporting on the last thing the service did as the system was shutting down. Basically, after any reboot, if you check logs very soon thereafter (for any service on the system), you'll see what it was reporting as it was shutting down for the reboot. I can't say for certain that that is what is indicated in the log excerpt above, but it would explain the old timestamp and the successful service behavior that followed.

>After RESTART ALL SERVICES it seems it has resumed analysis and shows the usual output, which confirms it really takes the real time recordings and processes without lag:
```
 ('Perdix perdix_Rebhuhn', 0.049771078)
 ('Perdix perdix_Rebhuhn', 0.0641239)
 ('Perdix perdix_Rebhuhn', 0.04529362)
 ('Perdix perdix_Rebhuhn', 0.051011525)
[NEW CONNECTION] ('127.0.0.1', 33360) connected.
[ACTIVE CONNECTIONS] 1
READING AUDIO DATA... DONE! READ 12 CHUNKS.
FULL FILENAME: -/home/pi/BirdSongs/February-2022/09-Wednesday/2022-02-09-birdnet-08:50:33.wav-
Date-time: 2022-02-09 08:50:33
```

Keeping with my original supposition, the "Restart ALL Services" script worked again in this scenario because the issues seemed to be in `birdnet_recording.service` getting stuck.

The next time you see the `FileNotFound` error, try restarting the recording service ("System" > "Tools" > "Manage Services" > "Restart Recording").

And if you have the ability to access the terminal when things are malfunctioning, could you please send the output of:
```
journalctl -n50 --no-pager --no-hostname -u birdnet_recording.service
```

Thank you for providing such great information on this issue (as I have come to expect from you :smile:) -- it is truly necessary when trying to troubleshoot that I have as much information as possible, and the more log out put I can get my hands on, the better, so thanks!

I've stopped testing the RPi0W2 for today so that I can work on other things (still only have one microphone!) so let me know what you observer as you observer it.

DD4WH, 2022-02-09:
Surprisingly, the RPi Zero 2W has had no more issues since the reboot this morning . . . 6 hours 50min
Will report as soon as I recognize something strange :-).

mcguirepr89, 2022-02-09:
I love it! Thanks for keeping me posted!

DD4WH, 2022-02-09:
I have ordered thermal glue (my old glue is hard as concrete by now :-) for a heatsink to be glued on the Zero CPU. It gets really hot:

[image]

mcguirepr89, 2022-02-09:
Yes I've noticed that right when you start things up, it performs very well, then the heat sets in and everything starts to get sluggish. The heatsink will be great. I wonder if there would be a way to thermally attach your heatsink to your outdoor metal enclosure? And if that enclosure were aluminum, perhaps even adhere the CPU itself to the enclosure with thermal paste? Just a tinkerer's thought.

mcguirepr89, 2022-02-09:
Nope `0x0` means everything is running perfectly well -- not throttling no voltage issues and no CPU limits

DD4WH, 2022-02-09:
Ah, yes, thanks! Pretty clear!
0x0 means that bit 0 is NOT set . . . [bit 0 seems to be the indicator for undervoltage as far as I remember]

mcguirepr89, 2022-02-09:
>Ah, yes, thanks! Pretty clear!
0x0 means that bit 0 is NOT set . . . [bit 0 seems to be the indicator for undervoltage as far as I remember]

A `0` bit may indeed indicate an undervoltage error, but these are not bits -- this is a Hex code. The bit translation for `0` (you only use the second half of the hex for the bit indication) is:
```
0000
```
Then you need to identify which bit represents which monitored state:

From the `vcgencmd` manual page:
```
       get_throttled
              Returns the throttled state of the system. This is a bit pattern - a bit being set indicates the following
              meanings:

              Bit   Meaning
              ────  ────────────────────────────────────
               0    Under-voltage detected
               1    Arm frequency capped
               2    Currently throttled
               3    Soft temperature limit active
              16    Under-voltage has occurred
              17    Arm frequency capping has occurred
              18    Throttling has occurred
              19    Soft temperature limit has occurred

              A value of zero indicates that none of the above conditions is true.

              To find if one of these bits has been set, convert the value returned to  binary,  then  number  each  bit
              along the top. You can then see which bits are set. For example:

              0x50000 = 0101 0000 0000 0000 0000

              Adding the bit numbers along the top we get:

              19   18   17   16   15   14   13   12   11   10   9   8   7   6   5   4   3   2   1
               0    1    0    1    0    0    0    0    0    0   0   0   0   0   0   0   0   0   0

              From this we can see that bits 18 and 16 are set, indicating that the Pi has previously been throttled due
              to under-voltage, but is not currently throttled for any reason.
```

Then that means:
```
00000000000000000000
||||            ||||_ Under-voltage detected
||||            |||_ Arm frequency capped
||||            ||_ Currently throttled
||||            |_ Soft temperature limit active
||||_ Under-voltage has occurred since last reboot
|||_ Arm frequency capped has occurred
||_ Throttling has occurred
|_ Soft temperature limit has occurred
```
Notice, none of the bits are "on", so your system is in tip top shape!

mcguirepr89, 2022-02-09:
For some reason, I'm seeing the `birdnet_recording.service` stop on my installation for no apparent reason :( this is no good!

mcguirepr89, 2022-02-09:
I FOUND THE BUG!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! in `birdnet_analysis.sh`
```
 19 ...
 20 make_thisrun &> /dev/null
 21 if ! diff ${LAST_RUN} ${THIS_RUN};then
 22   echo "The birdnet.conf file has changed"
 23   echo "Reloading services"
 24   cat ${THIS_RUN} > ${LAST_RUN}
 25   sudo systemctl stop birdnet_recording.service
 26   sudo rm -rf ${RECS_DIR}/$(date +%B-%Y/%d-%A)/*
 27 fi
 28 ...
```
I forgot to restart the `birdnet_recording.service`!!!!!!!!!!!!

DD4WH, 2022-02-09:
Sounds really good
Will do an update and report.

mcguirepr89, 2022-02-09:
heads up, the update takes a long time on the RPi0W2 -- if I were you, I would run the update via PuTTY SSH `update_birdnet.sh` to watch and see the update (for testing purposes) in case anything weird happens. It isn't necessary that you run the update that way, but if you do use the Web GUI for the update, allow it a few more minutes than the default 60 seconds the web page suggests.

DD4WH, 2022-02-09:
Thanks! Thats a good hint, probably one of the reasons why I experienced some strange effects last evening ;-).

DD4WH, 2022-02-09:
BTW, by now even the Raspberry Pi Zero 2W is unobtainable here in Germany . . .

DD4WH, 2022-02-09:
Hmm, it really takes a long time to update.
The Pi is now working and trying to catch up with the recordings that have been made, but not yet analyzed.
The time for the analysis of a 30seconds recording with 0.5 seconds overlap is on average about 12-14 seconds.
However, sometimes it needs MUCH longer, I observed up to 58 seconds for the same recording length . . . in this case (where I captured a screenshot), it needed 39 seconds
[image]

DD4WH, 2022-02-09:
Another anecdotal observation:
* it does not seem to make any difference whether the number of threads in the call in server.py is 1, 2 or even 4
* The time for analysis seems exactly the same
`    myinterpreter = tflite.Interpreter(model_path='/home/pi/BirdNET-Pi/model/BirdNET_6K_GLOBAL_MODEL.tflite',num_threads=2)`

mcguirepr89, 2022-02-09:
I haven't had great success yet with getting the RPi0W2 to truly keep up. I haven't tried any recording length over 60 seconds, but even at 60 seconds, things really get backed up once there are detections. It runs really smoothly, until it needs to do its job!

The other consideration is that there are extractions going on for detections which are not doing anything until the analysis service gives it detections. Thus you end up getting backed up when it starts working :/

I'm going to experiment with cutting down on other resources, adding cooling, and maybe overclocking a tiny bit to see if I can get something that works well when the birds get talkative.

DD4WH, 2022-02-09:
I think the reason why the Zero does not manage to keep up is that some analysis events last a huge lot longer than others. Spotted another of those events with 75 seconds instead of the usual 12 seconds !

[image]

mcguirepr89, 2022-02-09:
FINGERS ARE CROSSED! :crossed_fingers:

DD4WH, 2022-02-09:
just one observation:
UPDATE in Zero 2W over SSH
* it takes longer than 15 minutes
* it seems to hang at this command `+ apt install -qqy php php-fpm php-mysql php-xml php-zip
`
* or does the process end with this command? Hmm...

mcguirepr89, 2022-02-09:
Yikes! Do this next time, before even accessing the web browser:

SSH into the BirdNET-Pi and issue `stop_core_services.sh`
THEN to `update_birdnet.sh` and see if that helps.

DD4WH, 2022-02-09:
look at this --> 7.2 seconds for an analysis of 30 seconds of audio !!!

[image]

[image]

whats happening there? :-)

DD4WH, 2022-02-09:
Is this really a function of the CPU temperature ? 
No, it does not seem to be the case . . .

[image]

mcguirepr89, 2022-02-09:
It is very curious -- _with_ adequate cooling, we would know if that were the issue, without it, it will be a looming factor. It is possible that `vcgen_cmd` doesn't report accurately for this new RPi0W2 chipset? It was the case that above 60degrees C throttling led to this exact issue for the RPi3B _and_ placed a ARM frequency limit as a result.

I want to rule it out before I rule it out

DD4WH, 2022-02-09:
hmm, its slow again every time I return from one of the other pages of BirdNET-Pi (and when I stay in the log window a little longer, the time goes down again):

[image]

mcguirepr89, 2022-02-09:
Check the temperature now and see if it has neared 60C

DD4WH, 2022-02-09:
no, it is at 54 degrees right after return from the log where it said 75 seconds for one recording . . .

[image]

DD4WH, 2022-02-09:
the strange thing is: I never had durations as low as 8 seconds. Before the update and overheating I reported, the minimum was about 11-12 seconds.

DD4WH, 2022-02-09:
Next week I will have thermal glue and some heatsinks at my disposal and I will do exactly the same test WITH heatsink, we will see whether the behaviour changes.

DD4WH, 2022-02-09:
My system seems a little awkward now. I cant write any more the any file ("no permission")
will reboot and then stop for today . . . heat issue in my head ;-)

DD4WH, 2022-02-09:
Reboot did not fix it.
Stopped services and did an update --> only about two minutes!
After that I had to Restart all services, because nothing was running.
And then normal function can be observed.

DD4WH, 2022-02-10:
Unfortunately, in the night, the RPi Zero 2W stopped the analysis service again. It is reachable via the usual birdnetpi.local and everything seems normal, except that the log shows the same error message again, that I observed the night before.

[image]

Will not have much time the next days, so I will stop testing at this point.

mcguirepr89, 2022-02-12:
Two things worth noting:

1. I was able to run an installation from scratch directly on the RPi0W2 without changing any settings (e.g., swapping, clock speed, service settings)
2. Disabling the `livestream.service` drastically reduces the CPU load outside of analysis.

Recent tweaks that have helped a lot:
1. I have configured the `birdnet_recording.service` to require the `server.py` socket be open and ready before collecting data (to mitigate data getting backed up).
2. Likewise, I've amalgamated the `birdnet_server.service` and `birdnet_analysis.service` logs so that the logs are more verbose and show their interaction (as well as the `python` call that `birdnet_analysis.service` uses to send the data to the server socket).

I hope to get my hands on another microphone soon, so hopefully I will be able to keep testing this alongside my main installation to keep this developing.

My best,
Patrick

DD4WH, 2022-02-14:
After another 30 minutes of playing with the system, it gets really laggy and inresponsive and then does not respond any more at all.

* Test finished . . .

mcguirepr89, 2022-02-14:
@DD4WH I really appreciate your efforts.
I'm going to test this intermittently as shifts towards efficiency and reduced resource usage come to mind. It seems SO close to being able to work, but invariably it gets bogged down from normal use. It may be that _a_ solution could be to bypass the database and PHP altogether and have a system more like BirdNET-system wherein there is simply recordings and extractions with a file server. That's a much simpler "stack" and can still have `BirdDB.txt` for exporting the data and importing it into another system's database (maybe to have RPi0W2 collecting and sorting data, then a RPi4B houses the database and all its PHP visualizations etc.:thinking:

mcguirepr89, 2022-02-17:
Hello!

Update:
I am working on a new branch that will switch BirdNET-Pi over to using a single-file SQLite database to do away with the MariaDB server altogether. I have started testing with this and have already noticed that the RPi4B handles the database entries with MUCH more easy using the SQLite database, so I'm very hopeful that this is a step in the right direction in making things work on the RPi0W2.

DD4WH, 2022-02-17:
Sounds very good! Let me know as soon as you want me to test the new branch!

mcguirepr89, 2022-02-17:
I will! But it won't be for a little while -- I have a lot of testing I'd like to do before I ask any others to join in. In typical crappy-development-praxis-Patrick fashion, I will be working on the SQLite at the same time as a new, simplified installation (the user won't really notice a difference).

:crossed_fingers:

DD4WH, 2022-02-22:
Testing the Zero 2W with the sqlite version of today:

After an hour of playing the Pi gets really slow and unresponsive . . .

It seems it is under 100% load:

[image]

DD4WH, 2022-02-22:
OK, now the Pi is so slow and unresponsive, that it does not make sense to continue further testing.

I think I will now try to glue on some heatsinks onto the CPU and then retry with a cold start.

DD4WH, 2022-02-22:
many problems after restart:
* no audio any more, spectrogram is blank
* issueing `pulseaudio -k` in the web terminal fails: 
[image]

The Pi is still sluggy, although the temperature is much lower now with the heatsink, has not exceeded 45 degrees

DD4WH, 2022-02-22:
Hmm, I have to stop now with testing, the Pi does not respond any more.

mcguirepr89, 2022-02-22:
Bummer!!!!!!!!!!!!!!!!!
Thank you so much for your time! Function on the RPi0W2 will just have to be a wild wish for now -- we don't know what the future holds, so I'm not ruling out forever, but it can be put to rest for now.

Thank you again for you help with this--
My very best regards,
Patrick

mcguirepr89, 2022-04-04:
With the recent changes to `main`, BirdNET-Pi does install and work on an RPi0W2 now-- its official!

The caveats include:
- "Species Stats" is too unresponsive to be usable on an RPi0W2 (it works, it is just super duper slow)
- The livestream and icecast2 need to be disabled
- The default swap needs to be raised to around 2G in `/etc/dphys-swapfile` (not the zRAM kernel module)

I will be putting together some installation notes for folks who want to start testing on this hardware.
Take a look at my RPi0W2 test installation [here](https://pi0w2.birdnetpi.com:8443)

DD4WH, 2022-04-04:
Patrick,

that is very good news!!! Really looking forward to testing this !

mcguirepr89, 2022-04-04:
>is it sufficient to set CONF_SWAPSIZE = 2048, or do I need to set other limits too ?

This is my `/etc/dphys-swapfile`:
```
# /etc/dphys-swapfile - user settings for dphys-swapfile package
# author Neil Franklin, last modification 2010.05.05
# copyright ETH Zuerich Physics Departement
#   use under either modified/non-advertising BSD or GPL license

# this file is sourced with . so full normal sh syntax applies

# the default settings are added as commented out CONF_*=* lines

# where we want the swapfile to be, this is the default
#CONF_SWAPFILE=/var/swap

# set size to absolute value, leaving empty (default) then uses computed value
#   you most likely don't want this, unless you have an special disk situation
CONF_SWAPSIZE=2048

# set size to computed value, this times RAM size, dynamically adapts,
#   guarantees that there is enough swap without wasting disk space on excess
#CONF_SWAPFACTOR=2

# restrict size (computed and absolute!) to maximally this limit
#   can be set to empty for no limit, but beware of filled partitions!
#   this is/was a (outdated?) 32bit kernel limit (in MBytes), do not overrun it
#   but is also sensible on 64bit to prevent filling /var or even / partition
CONF_MAXSWAP=4096
```

I don't know if `CONF_MAXSWAP=4096` is necessary, but it is working.

I realized when testing this out myself that disabling the `livestream.service` doesn't disable the `icecast2` init (`systemd-sysv-install`) backend process, so disabling the `livestream.service` via "Tools" > "Services" is not sufficient. I'll need to fix that, then you can test. Alternatively, if you'd prefer altering an existing installation's service settings and then popping it into an RPi0W2 to test quickly, you can:
```
sudo systemctl disable --now livestream.service
sudo systemctl disable icecast2
sudo /etc/init.d/icecast2 stop
```

Note that changing the `/etc/dyphswap` seems to need a reboot (though in principle I don't see why) for the swap space to be allocated properly.

Also note that you may want to call `stop_core_services.sh` before beginning all this, since it will get locked up trying to process stuff really shortly after getting started. Then you can restart all services in "Tools" > "Services"

DD4WH, 2022-04-04:
Wow! Thanks for the hints! Have it running now! Its late here, so I will leave it running tonight and we´ll see what tomorrow brings . . .

mcguirepr89, 2022-04-04:
Just wanted to make an update here that I have made the necessary changes to be able to enable/disable/start/stop/restart `icecast2` alongside the `livestream.service` using "Tools" > "Services" now, so that is one less thing to do in the command line.

The last piece of the puzzle (for this) is a little GUI slider to adjust `CONF_SWAPSIZE` in "Advanced Settings". Or, an auto-detected portion in the installer that will check what `awk '/Mem/ {print $2}` is for the hardware and make the adjustments during installation. :thinking: thoughts?

DD4WH, 2022-04-05:
> Or, an auto-detected portion in the installer that will check what `awk '/Mem/ {print $2}` is for the hardware and make the adjustments during installation. 🤔 thoughts?

Yes: I would vote for autodetecting the RPi Zero hardware and 
* adjusting CONF_SWAPSIZE during the installation process
* deleting menu item "Species Stats"

Just now I am testing the Live Stream:
* it works (I restarted icecast2 and live stream service)
* however I sporadically experience very short hickups (they sound like a 100ms of "rrrrr") every two to 5 minutes or so --> not annoying, but it may make people think that something is not correct

Also, it would be good to inform those who dare to use the Rpi Zero that the GUI is really slow --> some menu items need several seconds to display. Not a severe problem at all, but it may irritate some users if they have not been "warned" ;-).

One more thought:
* it might be good to notice that the installation can take very long (or may even hang), if done on the Zero itself and should better be done on a RPi 4B
* maybe it would be possible to offer to donwload a specific image for the Zero, which already contains all the service adjustments, swap adjustments etc. ? 

Very nice, that now a very low cost version of BirdNET-Pi can be assembled!!!
* RPi Zero W2 --> about 20,- EURO [however, unavailable in this part of the world at the moment, like all other RPi models . . .]
* USB MONO audio sound card --> about 10,- EURO
* microphone --> about 10,- EURO (selfbuilt with AOM5024) or 30,- EURO (commercial Lavalier mic)

DD4WH, 2022-04-05:
Ooops, watch out for the temperature of the CPU !

[image]

And this temperature is reached WITH two heatsinks attached with thermal adhesive glue:

[image]

DD4WH, 2022-04-05:
This happens from time to time:

[image]

and coincidentally the live audio stream goes silent.

After a minute I can start the stream again and the server.py goes GREEN again and everything working as usual (hiwever, load is 89% on average . . .

mcguirepr89, 2022-04-05:
@DD4WH Thanks for the great notes! I agree that the user needs to know what they're getting themselves into when deciding to run things on the RPi0W2.

I am really hoping to be able to get started on putting together some pre-installed images for which I can specify the hardware they are meant to support. (Or I'll need to decide whether to have the RPi0W2 settings detected during installation, OR recommend to folks to perform the installation on an RPi4B, but won't be able to suggest both for obvious reasons.)

Re: temp --
Thanks for the heads up. For what it is worth, throttling won't occur until you hit 80 degrees Celsius, and then it will be fully throttled at 85 degrees. There is a `soft_limit` that can be hit at 60 degrees Celsius, but it is unclear whether the `soft_limit` is enabled on the RPi0W2, as it is an RPi3B+ feature (RPi4B does not have a `soft_limit` for the temperature threshold).

(Is that picture showing _red hot_ heatsinks????)

Regarding the "Process Status" indicator:
The way that indicator functions is by `grep`ping the output of the `ps` command every 30 seconds. If it is indicating red, that _could_ meant that there is an issue with the `birdnet_server.service` (server.py), _or_ that "Process Status" didn't get back the response it was looking for when it checked.

Since I haven't observed this on the RPi4B, and I haven't noticed any _actual_ interruptions with the `birdnet_server.service` (no errors in the log), I am inclined to assume that the "Process Status" check failed because the CPU load is so high that it was not able to fetch the full process list (output of `ps`) and grep through for `server.py` in time to display accurate results. With the silence in the live stream coinciding with this observation, a CPU load spike still makes sense (as it would again hinder the results of `ps` being parsed in time to display on the phpSysInfo page).

To test this theory, the next time you observe any red "Process Status" indications, click the green circular arrows at the top right of the "Process Status" window to rerun the check and see if the indicators change from red to green.

My system is still running strong, starting its third day now without any need for service restarts, so let's see how your system looks given a little more time to operate.

Thank you so much for your super quick testing on this hardware! Exciting prospect!

DD4WH, 2022-04-05:
> (Is that picture showing _red hot_ heatsinks????)

hihi, no, they are just made out of copper ;-).

DD4WH, 2022-04-07:
**Experience after three days:**
* system is running reliably, making detections and extracting bird recordings
* system is slow in responding, opening pages may take several seconds
* rare: hickupps when live audio is streaming: "rrr" sounds and digital artifacts
* very rare: hickupps with "rrr" sounds in the extracted recordings --> in one occasion this caused a wrong detection: nightjar with its rattling sound :-)
* CPU gets hot: 55 to 65 degress WITH heatsink: no throttling observed or indicated
* very rare: ffmpeg or server.py indicator in System Info is red --> in all occasions this "self-healed" in a few minutes time and every service was green again

Maybe live audio stream (icecast2 etc.) should be switched off, when RPi Zero hardware is detected ? This could potentially prevent the rare hickupps in the recordings!?

mcguirepr89, 2022-04-07:
I fixed the "Tools" > "Services" so that disabling the livestream will also handle disabling icecast2, so I'd like to keep the option open to the user as to whether or not they'd like to keep the live audio stream on their RPi0W2 installations.

Thank you so much for your feedback! I'm going to put together an RPi0W2 installation guide today to go over the little changes that are needed to get it working.

mcguirepr89, 2022-04-15:
Closing this now that documentation exists to get folks up and running. Thanks again to everyone for your help with testing!

closed 2022-04-15 by mcguirepr89