mcguirepr89/BirdNET-Pi #137: Cosmetic issue - supress scientific notation

DD4WH, 2022-02-09:
It seems the suppression of scientific notation in daily_plot.py has been forgotten.
https://github.com/mcguirepr89/BirdNET-Pi/commit/2171c84b5c1c43bb9ae68fc5c7f12076580ad4b6
Please add again :-)

mcguirepr89, 2022-02-12:
I made the change in the `forms` branch, but haven't the data to test its efficacy -- please let me know when you are able to confirm that the change fixed the scientific notation.

Also, HUGE thanks for linking the original fix -- super helpful (and why I am able to make the change without testing the outcome)

I'll keep this open until the fix is confirmed, at which point I will also pull the fix into `main` @Molokai -- you mentioned this, so I wanted to tag you here in case you are able to test the fix and can help confirm if that discontinues scientific notation.

Thank you both!

Molokai, 2022-02-13:
Oh I was referring to at what number did the scientific notation kick in? 100? Higher? 

Norway, we are in Norway :)

mcguirepr89, 2022-02-13:
>Oh I was referring to at what number did the scientific notation kick in? 100? Higher?

Ahh!! It kicks in at 3 digits, so anything 100 or greater.

>Norway, we are in Norway :)
Awesome! Thank you for reminding me :)

Molokai, 2022-02-13:
Then we are all good again

[image]

mcguirepr89, 2022-02-13:
Awesome!! I will close this issue. Thank you very much for checking that out for us!

closed 2022-02-13 by mcguirepr89