mcguirepr89/BirdNET-Pi #143: Wikipedia & AllAboutBirds links should use Scientific name

DD4WH, 2022-02-13:
Patrick, thanks so much for the new version!!! The links are really awesome and very useful!
Please use Scientific names, because non-English Common name links lead to dead links if directed to Wikipedia.org.

I changed this line in stats.php to:

`  <td><a href=\"https://de.wikipedia.org/wiki/$dbsciname\" target=\"top\"/>Wikipedia</a>, <a href=\"https://allaboutbirds.org/guide/$dbsciname\" target=\"top\"/>All About Birds</a></td>
`

In the longterm it would be good to use dynamic Wikipedia links with the language indicated, see above for hardcoded German version . . .

mcguirepr89, 2022-02-14:
Thanks for letting me know -- I will add that to my next changes -- the next change will also need to include the new `get_images.sh` script and that script will need to add some sort of image attribution, because as of now, it just grabs the images (from wiki-commons) without saying where the image came from.

mcguirepr89, 2022-02-15:
I've changed this now

closed 2022-02-15 by mcguirepr89