mcguirepr89/BirdNET-Pi #146: Birdnet log and web terminal link

oliverk71, 2022-02-14:
The links I provided for birdnet log and web terminal in the settings seem not to be taken by birdnetpi. I ran the following commands via SSH, as you recommended: `stop_core_services.sh && update_birdnet.sh && update_birdnet.sh && restart_services.sh`
After this I checked the menu.html and it still uses the old link, which is in my case https://birdnetgv.ddnss.de:8080

oliverk71, 2022-02-14:
Would the output be of interest? I can paste it here, if you want. 

However, none of the links works right now: 

https://birdnetgv.ddnss.de:8080/
https://birdnetgv.ddnss.de:8000/
https://birdnetlog.dnshome.de/
https://webterminal.dnshome.de/

I tested the last two last night (manually pasted in the browser, still on 4B) and I swear they worked. However, this is my Caddyfile: 

```
http://localhost http://birdnetpi.local https://birdnetgv.ddnss.de {
  root * /home/pi/BirdSongs/Extracted
  file_server browse
  reverse_proxy /stream localhost:8000
  php_fastcgi unix//run/php/php7.4-fpm.sock
}
https://webterminal.dnshome.de/ {
  reverse_proxy localhost:8888
}

https://birdnetlog.dnshome.de/ {
  reverse_proxy localhost:8080
}
```

oliverk71, 2022-02-14:
In birdnet.conf , the links are correct: 
```
BIRDNETPI_URL=https://birdnetgv.ddnss.de
BIRDNETLOG_URL=https://birdnetlog.dnshome.de/
WEBTERMINAL_URL=https://webterminal.dnshome.de/
```
but in menu.html it is:
```
    <h5>System</h5>
    - <a href="/scripts/" target="content">Tools</a><br>
    - <a href="/phpsysinfo/" target="content">Info</a><br>
    - <a href="https://birdnetgv.ddnss.de:8080" target="content">Log</a><br>
```

mcguirepr89, 2022-02-14:
Thank you for providing that information. 

Please try the following:
1. Remove all three Custom URLs from "Advanced Settings" and press "Update Settings".
2. Wait 30 seconds
3. Go to http://birdnetpi.local "System" > "Tools" > "Settings" > "Advanced Settings"
4. Re-enter all three custom URLs.
5. Wait 30-45 seconds
6. Go to https://birdnetgv.ddnss.de and you should hopefully have https://webterminal.dnshome.de as your Log link

Let me know how that works and we will go from there

mcguirepr89, 2022-02-14:
I can see that did not work -- I am taking a look through your system now -- give me 5 minutes.

oliverk71, 2022-02-14:
What I recognized was the last line in the output from `update_birdnet.sh` was: `sed: no input files`

mcguirepr89, 2022-02-14:
>What I recognized was the last line in the output from update_birdnet.sh was: sed: no input files

I have noticed that, but it is harmless -- an error when there is nothing to update, so to speak.

But what I noticed is that you have trailing slashes `/` on your URL, which _maybe_ is the issue. Try removing them and running a full update so your browser doesn't send a cached version with the slash.

oliverk71, 2022-02-14:
Okay, I will try. Takes a few minutes.

oliverk71, 2022-02-14:
Mh, I just noticed no trailing slashes shown in the advanced settings links. But however, I can try.

oliverk71, 2022-02-14:
Ah, okay, I see the trailing slashes in birdnet.local page, which I left open in another tab. We will see ...

oliverk71, 2022-02-14:
Now it works! That really seemed to be the reason. Thanks a lot!

mcguirepr89, 2022-02-14:
You are very welcome! I'll look into parsing those out, but I will close this issue now. Happy birding!

closed 2022-02-14 by mcguirepr89

mcguirepr89, 2022-02-14:
PS
You're going to want to set a webpage password as soon as possible