mcguirepr89/BirdNET-Pi #147: Species stats does not show

DD4WH, 2022-02-15:
On the new page "Species stats":
* this page is not shown in the "mobile" version of the BirdNET-Pi page
* on the standard page version (on my PC): if I choose a species from the dropw-down menu, and press `Show species statistics`, nothing is happening
* maybe language issue? 
* I use locale settings for Gerrmany
* I use German Common names

oliverk71, 2022-02-15:
Until now, it is very slow. May have something to do with large images, but even for that it is very slow. 
The way how to figure out which image to load, can be optimized, as this screenshot shows: 
[image]
It shows a postcard of "Bad Elster". "Elster" is German for "magpie". ;)

DD4WH, 2022-02-15:
:-)
Thanks for that hint!
Yes, I can confirm, after ten minutes loading time, my species stats is there !
[image]

The looking up of webpages and photographs etc. should ALWAYS be based on the SCIENTIFIC NAME.

closed 2022-02-15 by mcguirepr89