mcguirepr89/BirdNET-Pi #150: Update from forms breaks with file not found and connection refused

davidmam, 2022-02-15:
Here is a log excerpt. Unfortunately Edge on Windows 11 doesn't support copy/paste from Webterminal.

[image]

davidmam, 2022-02-15:
The folder ~/BirdSongs contains three folders - Extracted, February-2022 and Processed
Processed has wav and spectrograms
The February-2022 has 15-February which contains lots of wav files, I presume the backlog of unprocessed ones. Nothing matches that wildcard.

mcguirepr89, 2022-02-15:
Hi, Dr. Martin -- 
You log is implying that you've cleared all data and have not yet processed any files (they are collecting, probably, but not analyzing).
If you are able to issues commands via SSH, please issue the following:
```
journalctl --no-hostname -ubirdnet_analysis -ubirdnet_server -ucron -ubirdnet_recording -uextraction --since "19:00:00" > /home/pi/BirdNET-Pi/scripts/davidslog.log
```
Then upload the file from the `scripts` directory here for me to look through.

If not, try restarting all services through the web gui and let me know how the log looks at that point.

davidmam, 2022-02-15:
[davidslog.log](https://github.com/mcguirepr89/BirdNET-Pi/files/8074467/davidslog.log)
It is pretty big

davidmam, 2022-02-15:
pi@woodland:~/BirdNET-Pi$ git log
commit 4863a2eef707aa127452f20ca81cc0024523003c (HEAD -> main, origin/main, origin/HEAD)
Author: mcguirepr89 <mcguirepr89@gmail.com>
Date:   Tue Feb 15 13:54:22 2022 -0500

    removing old files

mcguirepr89, 2022-02-15:
Thank you -- 

This error is most definitely related to trying to update from `forms` to main. I will need a lot more information to get this working, but to start, could you please run this update via SSH from PuTTY or another terminal emulator (not the webterminal) and send me the output:
```
update_birdnet.sh
```

davidmam, 2022-02-15:
It will be tomorrow or Thursday.

davidmam, 2022-02-17:
this is about the n'th time i've run the update script trying to capture the outpu
[update_log.txt](https://github.com/mcguirepr89/BirdNET-Pi/files/8091960/update_log.txt)
t

davidmam, 2022-02-17:
Yes, problem hasn't gone

davidmam, 2022-02-17:
Still getting the connection refused error

davidmam, 2022-02-17:
_LENGTH set to 15
20:30:13 birdnet_analysis.sh[38241]: python3 analyze.py --i /home/pi/BirdSongs/February-2022/17-Thursday/2022-02-17-birdnet-20:11:04.wav --o /home/pi/BirdSongs/February-2022/17-Thursday/2022-02-17-birdnet-20:11:04.wav.csv --lat 56.369 --lon -3.135 --week 7 --overlap 0.5 --sensitivity 1.25 --min_conf 0.7 --birdweather_id IN_USE
20:30:13 birdnet_analysis.sh[39063]: Traceback (most recent call last):
20:30:13 birdnet_analysis.sh[39063]:   File "/home/pi/BirdNET-Pi/scripts/analyze.py", line 12, in <module>
20:30:13 birdnet_analysis.sh[39063]:     client.connect(ADDR)
20:30:13 birdnet_analysis.sh[39063]: ConnectionRefusedError: [Errno 111] Connection refused
20:30:13 birdnet_analysis.sh[38241]: RECORDING_LENGTH set to 15
20:30:13 birdnet_analysis.sh[38241]: python3 analyze.py --i /home/pi/BirdSongs/February-2022/17-Thursday/2022-02-17-birdnet-20:11:19.wav --o /home/pi/BirdSongs/February-2022/17-Thursday/2022-02-17-birdnet-20:11:19.wav.csv --lat 56.369 --lon -3.135 --week 7 --overlap 0.5 --sensitivity 1.25 --min_conf 0.7 --birdweather_id IN_USE
20:30:14 birdnet_analysis.sh[39075]: Traceback (most recent call last):
20:30:14 birdnet_analysis.sh[39075]:   File "/home/pi/BirdNET-Pi/scripts/analyze.py", line 12, in <module>
20:30:14 birdnet_analysis.sh[39075]:     client.connect(ADDR)
20:30:14 birdnet_analysis.sh[39075]: ConnectionRefusedError: [Errno 111] Connection refused
20:30:14 birdnet_analysis.sh[38241]: RECORDING_LENGTH set to 15
20:30:14 birdnet_analysis.sh[38241]: python3 analyze.py --i /home/pi/BirdSongs/February-2022/17-Thursday/2022-02-17-birdnet-20:11:34.wav --o /home/pi/BirdSongs/February-2022/17-Thursday/2022-02-17-birdnet-20:11:34.wav.csv --lat 56.369 --lon -3.135 --week 7 --overlap 0.5 --sensitivity 1.25 --min_conf 0.7 --birdweather_id IN_USE
20:30:14 birdnet_analysis.sh[39088]: Traceback (most recent call last):
20:30:14 birdnet_analysis.sh[39088]:   File "/home/pi/BirdNET-Pi/scripts/analyze.py", line 12, in <module>
20:30:14 birdnet_analysis.sh[39088]:     client.connect(ADDR)
20:30:14 birdnet_analysis.sh[39088]: ConnectionRefusedError: [Errno 111] Connection refused
20:30:14 birdnet_analysis.sh[38241]: RECORDING_LENGTH set to 15
20:30:15 birdnet_analysis.sh[38241]: python3 analyze.py --i /home/pi/BirdSongs/February-2022/17-Thursday/2022-02-17-birdnet-20:11:49.wav --o /home/pi/BirdSongs/February-2022/17-Thursday/2022-02-17-birdnet-20:11:49.wav.csv --lat 56.369 --lon -3.135 --week 7 --overlap 0.5 --sensitivity 1.25 --min_conf 0.7 --birdweather_id IN_USE
20:30:15 birdnet_analysis.sh[39109]: Traceback (most recent call last):
20:30:15 birdnet_analysis.sh[39109]:   File "/home/pi/BirdNET-Pi/scripts/analyze.py", line 12, in <module>
20:30:15 birdnet_analysis.sh[39109]:     client.connect(ADDR)
20:30:15 birdnet_analysis.sh[39109]: ConnectionRefusedError: [Errno 111] Connection refused
20:30:15 birdnet_analysis.sh[38241]: RECORDING_LENGTH set to 15
20:30:15 birdnet_analysis.sh[38241]: python3 analyze.py --i /home/pi/BirdSongs/February-2022/17-Thursday/2022-02-17-birdnet-20:12:04.wav --o /home/pi/BirdSongs/February-2022/17-Thursday/2022-02-17-birdnet-20:12:04.wav.csv --lat 56.369 --lon -3.135 --week 7 --overlap 0.5 --sensitivity 1.25 --min_conf 0.7 --birdweather_id IN_USE
20:30:15 birdnet_analysis.sh[39122]: Traceback (most recent call last):
20:30:15 birdnet_analysis.sh[39122]:   File "/home/pi/BirdNET-Pi/scripts/analyz

davidmam, 2022-02-17:
I ran the script you asked for previously (upthread) - this one is a new request and a different command. I'll run it now.
That appeared to run OK, giving an analysis output on the command line.

How long does it normally take to release the connection? Still showing connection refused in the syslog.

mcguirepr89, 2022-02-17:
I'm sorry, but not reading my comments AND not sending the requested output will not allow me to help you. Please just send me the output so that I can read your logs -- I cannot help without the information I need, and I have requested that information, not simply that you run those commands just to have you run them.

davidmam, 2022-02-17:
I'm looking back through the comments. You had not asked me to run that last command before. I don't know if it is a browser thing (Edge on Win 11) but all I saw was 
journalctl --no-hostname -ubirdnet_analysis -ubirdnet_server -ucron -ubirdnet_recording -uextraction --since "19:00:00" > /home/pi/BirdNET-Pi/scripts/davidslog.log
 and I sent you that file.

You then asked me to run  
update_birdnet.sh

which I did.

It was a bit of a pain to capture the output from that, but I did and uploaded that. 

So I am not sure which bits I missed of your requests until your comment about not running

set -x; restart_services.sh && journalctl -n100 -ubirdnet_server --no-hostname --no-pager

which I then ran, but capturing the output was not requested. So either there is something in the web interface with github that is hiding bits of your comments, or they were not there.

I have read your comments.

davidmam, 2022-02-17:
This is the ifconfig for the loopback
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 194106  bytes 145956193 (139.1 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 194106  bytes 145956193 (139.1 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

which probably isn't helpful.

mcguirepr89, 2022-02-17:
The loopback interface is irrelevant here. My only other recommendation is a clean installation.

davidmam, 2022-02-17:
That much I feared. OK, will proceed with that. I'll also take it  indoors as sitting in the garden at 3C is not fun.

closed 2022-02-17 by mcguirepr89