mcguirepr89/BirdNET-Pi #151: systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 24.

oliverk71, 2022-02-16:
I noticed these strange log and my swap file being 100% in use on my pi 3B. Might have something to do with that I just put the Pi 4B SD card into the pi 3B and forgot about the desktop environment, which is installed on SD. But I configured the pi not to boot into desktop. I restarted the pi, but the same log entries occur, but at least the swap file is only being used 3%, then. Restarted services, but again those log entries.

```
-- Journal begins at Mon 2022-02-14 01:24:09 CET. --
18:06:47 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 24.
18:06:47 systemd[1]: Stopped BirdNET Analysis.
18:06:47 systemd[1]: Started BirdNET Analysis.
18:06:48 birdnet_analysis.sh[2770]: Wed 16 Feb 18:06:48 CET 2022
18:06:48 systemd[1]: birdnet_analysis.service: Succeeded.
18:06:51 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 25.
18:06:51 systemd[1]: Stopped BirdNET Analysis.
18:06:51 systemd[1]: Started BirdNET Analysis.
18:06:52 birdnet_analysis.sh[2833]: Wed 16 Feb 18:06:52 CET 2022
18:06:52 systemd[1]: birdnet_analysis.service: Succeeded.
18:06:54 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 26.
18:06:54 systemd[1]: Stopped BirdNET Analysis.
18:06:54 systemd[1]: Started BirdNET Analysis.
18:06:55 birdnet_analysis.sh[2872]: Wed 16 Feb 18:06:55 CET 2022
18:06:55 systemd[1]: birdnet_analysis.service: Succeeded.
18:06:57 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 27.
18:06:57 systemd[1]: Stopped BirdNET Analysis.
18:06:57 systemd[1]: Started BirdNET Analysis.
18:06:59 birdnet_analysis.sh[2919]: Wed 16 Feb 18:06:59 CET 2022
18:06:59 systemd[1]: birdnet_analysis.service: Succeeded.
18:07:01 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 28.
18:07:01 systemd[1]: Stopped BirdNET Analysis.
18:07:01 systemd[1]: Started BirdNET Analysis.
18:07:02 birdnet_analysis.sh[2971]: Wed 16 Feb 18:07:02 CET 2022
18:07:02 systemd[1]: birdnet_analysis.service: Succeeded.
18:07:04 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 29.
18:07:04 systemd[1]: Stopped BirdNET Analysis.
18:07:04 systemd[1]: Started BirdNET Analysis.
18:07:05 birdnet_analysis.sh[3021]: Wed 16 Feb 18:07:05 CET 2022
18:07:05 systemd[1]: birdnet_analysis.service: Succeeded.
18:07:07 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 30.
18:07:07 systemd[1]: Stopped BirdNET Analysis.
18:07:07 systemd[1]: Started BirdNET Analysis.
18:07:08 birdnet_analysis.sh[3066]: Wed 16 Feb 18:07:08 
```

oliverk71, 2022-02-16:
The log looks normal to me. But swap is 90% in use.

mcguirepr89, 2022-02-16:
>The log looks better but still the same issue as it seems. Swap is 90% in use.

Please issue the command below to see the top ten processes by memory usage:
```
ps -o pid,user,%mem,command ax | sort -b -k3 -r | head -n11
```
Also, let's see what the whole memory situation is:
```
free -h
```
I do not see anything wrong with the log excerpt. Are you seeing something wrong with the system? Using 90% of the swap is not an indicator of malfunction at this point.

oliverk71, 2022-02-16:
Output of the first command: 

```
    PID USER     %MEM COMMAND
    146 root      6.4 /lib/systemd/systemd-journald
  38489 pi        5.2 journalctl --no-hostname -q -o short -fu birdnet_analysis -ubirdnet_server -uextraction
  38484 pi        4.7 journalctl --no-hostname -q -o short -fu birdnet_analysis -ubirdnet_server -uextraction
  20623 pi        4.5 ffmpeg -loglevel 52 -ac 2 -f alsa -i default -acodec libmp3lame -b:a 320k -ac 2 -content_type audio/mpeg -f mp3 icecast://source:birdnetpi@localhost:8000/stream -re
  21191 pi       40.0 /home/pi/BirdNET-Pi/birdnet/bin/python3 /usr/local/bin/server.py
    993 caddy     2.1 /usr/bin/caddy run --environ --config /etc/caddy/Caddyfile
  12900 pi        2.0 /usr/bin/pulseaudio --daemonize=no --log-target=journal
  38525 caddy     1.3 php-fpm: pool www
  39879 caddy     1.2 php-fpm: pool www
  39270 caddy     1.2 php-fpm: pool www
```
and for the second command: 

```
Mem:           909Mi       305Mi        99Mi       3.0Mi       505Mi       540Mi
Swap:           99Mi        90Mi       9.0Mi
```

> I do not see anything wrong with the log excerpt. Are you seeing something wrong with the system? Using 90% of the swap is not an indicator of malfunction at this point

This line: 

`18:40:57 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 21.`

I did not recognize it earlier.

oliverk71, 2022-02-16:
I looked into the log because I did not had any recognitions, today. But this can be due to the weather here (stormy).

mcguirepr89, 2022-02-16:
First command looks good

Though the second command suggests you may want to increase the size of your swap partition, or disable it in lieu zRAM swapping.
To add to the traditional swap:
```
sudo fallocate -l 2G /swap
sudo mkswap /swap
sudo chmod 0600 /swap
sudo swapon /swap
```
Or to enable the zRAM kernel module and disable traditional swapping:
```
install_zram_service.sh &&
sudo dphys-swapfile swapoff
sudo dphys-swapfile uninstall
sudo update-rc.d dphys-swapfile remove
sudo apt purge dphys-swapfile
```

>This line:
`18:40:57 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 21.`
I did not recognize it earlier.

It has always been a part of the logging -- `birdnet_analysis.service` restarts when it finished analyzing everything that is available to it, so it will grow to be a large number the longer the system runs.
From my log: `13:17:39 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 139.`

>I looked into the log because I did not had any recognitions, today. But this can be due to the weather here (stormy).

I do think that there was a bug in the socket restarting process from the version running overnight for you -- the update and service restart got you back to a good, known-working state.

Sorry for the bugginess, but thank you for opening this issue.

oliverk71, 2022-02-16:
I will give zRAM a try.  Thank you very much, again, Patrick!

closed 2022-02-16 by mcguirepr89