mcguirepr89/BirdNET-Pi #152: Image download in species stats

oliverk71, 2022-02-16:
Just discovered the following: 

[image]

Two things to notice: The lookup should return a picture of _Pica pica_ (Eurasian Magpie) not _Pica mauritanica_ (Maghreb magpie). But the screenshot actually shows _Cittarium pica_ (a shell species).

mcguirepr89, 2022-02-16:
Since I don't have access to an actual photo database, this link parsing is the best I can do for now. The link for that result is simply the first that comes up when you search for "Pica Pica" on WikiMedia Commons:
[image]
https://commons.wikimedia.org/w/index.php?search=pica+pica&title=Special:MediaSearch&go=Go&type=image

Sorry I won't be able to do anything about that right now, and I know that my method of image retrieval is very unsophisticated, but until I have API access to a database of images, this will have to do :/

closed 2022-02-16 by mcguirepr89

oliverk71, 2022-02-16:
If you add exclamation marks ("Pica pica") you actually get better search results.

oliverk71, 2022-02-16:
Maybe you can restrict the search to a category (Aves) as well? 

https://commons.wikimedia.org/wiki/Category:Aves

mcguirepr89, 2022-02-16:
The image search URL is:
```
https://commons.wikimedia.org/w/index.php?search=${1//_/+}&Category=Aves&title=Special:MediaSearch&go=Go&type=image
```
where `${1//_+}` is the species scientific name, replacing the space with a `+` to format the URL properly. My results are mixed when adding quotation marks -- some images are better, some are no longer visible.

I haven't noticed any difference in results from the image URL when adding `&Category=Birds` nor `&Category=Aves`, so I'm not sure those parameters function for this path.

oliverk71, 2022-02-16:
Most probably the search will do its job. But I stumbled over this: https://www.mediawiki.org/wiki/Wikimedia_REST_API#Other_APIs
Don't know if it is helpful or if you already know about.

mcguirepr89, 2022-02-16:
I looks promising, but I only see 1 endpoint and it is for Recommendation data :/
https://commons.wikimedia.org/api/rest_v1/#/