mcguirepr89/BirdNET-Pi #156: Second LAN IP and some errors

oliverk71, 2022-02-18:
Good morning! My second Pi 3B now shows some errors. It seems that the cooling is bad, but however, I wonder why there is this second LAN IP. It should only use 192.168.1.6 and not 192.168.1.150. Both of my Pi 3B should run the same version of birdnetpi updated yesterday. Don't know if the other errors are a real issue or if it's just because of bad cooling. I am going to change the cooling pad and look what happens. 

```
/usr/local/bin/extra_info.sh - /usr/local/bin/extra_info.sh: line 33: [: -eq: unary operator expected
/usr/local/bin/extra_info.sh: line 36: [: -eq: unary operator expected
Return value: 0

./xml.php on line 39
./includes/output/class.WebpageXML.inc.php on line 203 in function getJsonString()
./includes/output/class.WebpageXML.inc.php on line 190 in function getXMLString()
./includes/xml/class.XML.inc.php on line 884 in function getXml()
./includes/xml/class.XML.inc.php on line 873 in function _buildXml()
./includes/xml/class.XML.inc.php on line 909 in function _buildPlugins()
./plugins/viewer/class.viewer.inc.php on line 63 in function execute()
./includes/class.CommonFunctions.inc.php on line 293 in function executeProgram( "extra_info.sh", "", "........................................IPs.....................................\nLAN IP: 192.168.1.150\nPublic IP: 46.89.145.102\n..................................`vcgencmd stats`..............................\nthrottled=0x20002\nBinary: 100000000000000010\n100000000000000010\nISSUES DETECTED\nArm frequency capped\nArm frequency capping has occurred\n....................................Clock Speeds................................\narm:\tfrequency(48)=1087000000\ncore:\tfrequency(1)=400000000\nh264:\tfrequency(28)=0\nisp:\tfrequency(45)=0\nv3d:\tfrequency(46)=300000000\nuart:\tfrequency(22)=47999000\npwm:\tfrequency(25)=100000000\nemmc:\tfrequency(50)=200000000\npixel:\tfrequency(29)=337000\nvec:\tfrequency(10)=108000000\nhdmi:\tfrequency(0)=0\ndpi:\tfrequency(4)=0\n........................................Volts...................................\ncore:\tvolt=1.3188V\nsdram_c:\tvolt=1.2000V\nsdram_i:\tvolt=1.2000V\nsdram_p:\tvolt=1.2250V\n.....................................Caddyfile..................................\nhttp://localhost http://birdnetpi.local http://192.168.1.6 {\n root * /home/pi/BirdSongs/Extracted\n file_server browse\n basicauth /Processed* {\n birdnet JDJhJDE0JDFhNDU3NUtLL1d2Q20zZFlkWG9Eci5ZLnY0dEltN3hFWFBwb3poMUNaUWVVeE5BMXFFVi9L\n }\n basicauth /scripts* {\n birdnet JDJhJDE0JDFhNDU3NUtLL1d2Q20zZFlkWG9Eci5ZLnY0dEltN3hFWFBwb3poMUNaUWVVeE5BMXFFVi9L\n }\n basicauth /stream {\n birdnet JDJhJDE0JDFhNDU3NUtLL1d2Q20zZFlkWG9Eci5ZLnY0dEltN3hFWFBwb3poMUNaUWVVeE5BMXFFVi9L\n }\n basicauth /phpsysinfo* {\n birdnet JDJhJDE0JDFhNDU3NUtLL1d2Q20zZFlkWG9Eci5ZLnY0dEltN3hFWFBwb3poMUNaUWVVeE5BMXFFVi9L\n }\n reverse_proxy /stream localhost:8000\n php_fastcgi unix//run/php/php7.4-fpm.sock\n}\n.................................... Crontab....................................\n\nSHELL=/bin/sh\nPATH=/usr/local/sbin:/usr/local/bin:/sbin:/bin:/usr/sbin:/usr/bin\n\n17 *\t* * *\troot cd / && run-parts --report /etc/cron.hourly\n25 6\t* * *\troot\ttest -x /usr/sbin/anacron || ( cd / && run-parts --report /etc/cron.daily )\n47 6\t* * 7\troot\ttest -x /usr/sbin/anacron || ( cd / && run-parts --report /etc/cron.weekly )\n52 6\t1 * *\troot\ttest -x /usr/sbin/anacron || ( cd / && run-parts --report /etc/cron.monthly )\n*/5 * * * * pi /usr/local/bin/disk_check.sh &> /dev/null\n*/3 * * * * pi /usr/local/bin/cleanup.sh &> /dev/null\n@reboot pi /usr/local/bin/cleanup.sh &> /dev/null" )
```

mcguirepr89, 2022-02-18:
Hi, Oliver -- the errors form the log excerpt you've sent point to small benign failures in the bit checking the `extra_info.sh` script uses. I've silenced those errors now (so if you update they will go away).

The script _does_ successfully indicate that the "Arm frequency capped" and "Arm frequency capping has occurred".

Regarding the secondary LAN IP, that is not anything I can explain. Your router (in 99.9% of instances) will also be the DHCP server for the network. It is responsible for issuing IP addresses to hosts on the network as well as configuring those connections to use the proper subnet mask, gateway, and host and network IP ranges.

The Raspberry Pi has two network cards: 1 for the Ethernet connection and 1 for WiFi connections.

Any network card in general can support more than one IP address, using one as primary and others as secondary, tertiary, quaternary, etc.

I may be able to get an idea of what is going on if you send me the output of the following commands:
```
ip a
```
--
```
sudo iwconfig
```
--
```
cat /etc/resolve.conf
```
--
```
cat /etc/dhcpcd.conf
```

Typically, a Raspberry Pi with two LAN IP addresses occurs for one of the following reasons:
1. The Pi is connected to the network via an Ethernet cable AND is connected via WiFi
2. The Pi has been configured to use a static IP address, but the static IP has not been reserved on the DHCP server, so the Pi gives itself the static IP in its configuration and the DHCP server gives it an IP in the DHCP range.
3. The network connection (LAN or WiFi) has a secondary IP address configured for the interface in `/etc/dhcpcd`

Let me know if any of that sounds possible

oliverk71, 2022-02-18:
Maybe it is because I use a wifi dongle on this pi. Yes, look like. I did not come into my mind that the pi uses both. But now as I think about, it's totally clear. Anyway:

```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether b8:27:eb:85:b0:fa brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 80:1f:02:a1:9a:74 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.6/24 brd 192.168.1.255 scope global dynamic noprefixroute wlan0
       valid_lft 863558sec preferred_lft 755558sec
    inet6 fd00::a97:48fc:3ba5:c9fa/64 scope global dynamic mngtmpaddr noprefixroute 
       valid_lft 7027sec preferred_lft 3427sec
    inet6 2003:d3:a729:f400:682a:32de:3b2d:8d68/64 scope global dynamic mngtmpaddr noprefixroute 
       valid_lft 7027sec preferred_lft 1627sec
    inet6 fe80::ea2b:c1c5:fdaa:e2ec/64 scope link 
       valid_lft forever preferred_lft forever
4: wlan1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:d0:e5:af brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.150/24 brd 192.168.1.255 scope global dynamic noprefixroute wlan1
       valid_lft 863564sec preferred_lft 755564sec
    inet6 fd00::b4f9:f606:9c6d:559a/64 scope global dynamic mngtmpaddr noprefixroute 
       valid_lft 6765sec preferred_lft 3165sec
    inet6 2003:d3:a729:f400:a44d:3a53:7de1:e78/64 scope global dynamic mngtmpaddr noprefixroute 
       valid_lft 6765sec preferred_lft 728sec
    inet6 fe80::8e86:8aa0:a2f:ccf/64 scope link 
       valid_lft forever preferred_lft forever
```
```
lo        no wireless extensions.

eth0      no wireless extensions.

wlan0     IEEE 802.11  ESSID:"Konfusius"  
          Mode:Managed  Frequency:2.437 GHz  Access Point: 34:81:C4:C7:C7:F8   
          Bit Rate=72.2 Mb/s   Tx-Power=20 dBm   
          Retry short limit:7   RTS thr=2347 B   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=48/70  Signal level=-62 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:2   Missed beacon:0

wlan1     IEEE 802.11  ESSID:"Konfusius"  
          Mode:Managed  Frequency:2.437 GHz  Access Point: 34:81:C4:C7:C7:F8   
          Bit Rate=12 Mb/s   Tx-Power=31 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:on
          Link Quality=37/70  Signal level=-73 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:37  Invalid misc:0   Missed beacon:0
```
```
# Configuration for resolvconf(8)
# See resolvconf.conf(5) for details

resolv_conf=/etc/resolv.conf
# If you run a local name server, you should uncomment the below line and
# configure your subscribers configuration files below.
#name_servers=127.0.0.1

# Mirror the Debian package defaults for the below resolvers
# so that resolvconf integrates seemlessly.
dnsmasq_resolv=/var/run/dnsmasq/resolv.conf
pdnsd_conf=/etc/pdnsd.conf
unbound_conf=/etc/unbound/unbound.conf.d/resolvconf_resolvers.conf
```

```
# A sample configuration for dhcpcd.
# See dhcpcd.conf(5) for details.

# Allow users of this group to interact with dhcpcd via the control socket.
#controlgroup wheel

# Inform the DHCP server of our hostname for DDNS.
hostname

# Use the hardware address of the interface for the Client ID.
clientid
# or
# Use the same DUID + IAID as set in DHCPv6 for DHCPv4 ClientID as per RFC4361.
# Some non-RFC compliant DHCP servers do not reply with this set.
# In this case, comment out duid and enable clientid above.
#duid

# Persist interface configuration when dhcpcd exits.
persistent

# Rapid commit support.
# Safe to enable by default because it requires the equivalent option set
# on the server to actually work.
option rapid_commit

# A list of options to request from the DHCP server.
option domain_name_servers, domain_name, domain_search, host_name
option classless_static_routes
# Respect the network MTU. This is applied to DHCP routes.
option interface_mtu

# Most distributions have NTP support.
#option ntp_servers

# A ServerID is required by RFC2131.
require dhcp_server_identifier

# Generate SLAAC address using the Hardware Address of the interface
#slaac hwaddr
# OR generate Stable Private IPv6 Addresses based from the DUID
slaac private

# Example static IP configuration:
#interface eth0
#static ip_address=192.168.0.10/24
#static ip6_address=fd51:42f8:caae:d92e::ff/64
#static routers=192.168.0.1
#static domain_name_servers=192.168.0.1 8.8.8.8 fd51:42f8:caae:d92e::1

# It is possible to fall back to a static IP if DHCP fails:
# define static profile
#profile static_eth0
#static ip_address=192.168.1.23/24
#static routers=192.168.1.1
#static domain_name_servers=192.168.1.1

# fallback to static profile on eth0
#interface eth0
#fallback static_eth0

```

mcguirepr89, 2022-02-18:
Yes, the WiFi dongle is why you are seeing an extra IP address -- if you'd like to totally disable the Raspberry Pi's WiFi network card (to _only_ use the dongle), do the following:
1. Open `/boot/config.txt` using the text editor of your choice (`nano` or `vim`)
   ```
   sudo nano /boot/config.txt
   ```
1. Scroll down to the "`[all]`" section and add the following line just below it:
   ```
   dtoverlay=disable-wifi
   ```
1. Reboot `sudo reboot`

That will tell the kernel not to load the onboard wifi module, but won't touch your WiFi dongle.

I'm going to close this issue unless there is anything else you wanted me to take a look at?

oliverk71, 2022-02-19:
Sorry for being late. You can close this issue. I currently have some problems with my other pi 3B, but it is completely different, so I am going to open another issue for that.

closed 2022-02-19 by oliverk71