mcguirepr89/BirdNET-Pi #157: Pi 3B problems

oliverk71, 2022-02-19:
As you know, I have the pi 3B where birdnet-pi is installed, both have the newest updates installed. On one pi zRAM is activated to avoid swap. Unfortunately this pi is not working reliable. It stopped recording a few times the last days and I had to restart services or reboot to get it running again. When this occured, I saw an empty spectrogram. But now, I recognized that the pi looks like is several hours behind, but the spectrogram looked okay. The log was like this: 

```
13:15:10 extract_new_birdsounds.sh[511232]: Moving processed files to /home/pi/BirdSongs/Processed
13:15:10 extract_new_birdsounds.sh[511232]: Moving processed files to /home/pi/BirdSongs/Processed
13:15:10 extract_new_birdsounds.sh[511232]: Moving processed files to /home/pi/BirdSongs/Processed
13:15:10 extract_new_birdsounds.sh[511232]: Linking Processed files to /home/pi/BirdSongs/Extracted/Processed web directory
13:15:10 systemd[1]: extraction.service: Succeeded.
13:15:20 systemd[1]: Started BirdNET BirdSound Extraction.
13:15:20 extract_new_birdsounds.sh[511282]: Moving processed files to /home/pi/BirdSongs/Processed
13:15:20 extract_new_birdsounds.sh[511282]: Moving processed files to /home/pi/BirdSongs/Processed
13:15:20 extract_new_birdsounds.sh[511282]: Moving processed files to /home/pi/BirdSongs/Processed
13:15:20 extract_new_birdsounds.sh[511282]: Linking Processed files to /home/pi/BirdSongs/Extracted/Processed web directory
13:15:20 systemd[1]: extraction.service: Succeeded.
13:15:30 systemd[1]: Started BirdNET BirdSound Extraction.
13:15:30 extract_new_birdsounds.sh[511331]: Moving processed files to /home/pi/BirdSongs/Processed
13:15:30 extract_new_birdsounds.sh[511331]: Moving processed files to /home/pi/BirdSongs/Processed
13:15:30 extract_new_birdsounds.sh[511331]: Moving processed files to /home/pi/BirdSongs/Processed
13:15:30 extract_new_birdsounds.sh[511331]: Linking Processed files to /home/pi/BirdSongs/Extracted/Processed web directory
13:15:30 systemd[1]: extraction.service: Succeeded.
13:15:40 systemd[1]: Started BirdNET BirdSound Extraction.
13:15:40 extract_new_birdsounds.sh[511380]: Moving processed files to /home/pi/BirdSongs/Processed
13:15:40 extract_new_birdsounds.sh[511380]: Moving processed files to /home/pi/BirdSongs/Processed
13:15:40 extract_new_birdsounds.sh[511380]: Moving processed files to /home/pi/BirdSongs/Processed
13:15:40 extract_new_birdsounds.sh[511380]: Linking Processed files to /home/pi/BirdSongs/Extracted/Processed web directory
13:15:40 systemd[1]: extraction.service: Succeeded.
13:15:50 systemd[1]: Started BirdNET BirdSound Extraction.
13:15:50 extract_new_birdsounds.sh[511429]: Moving processed files to /home/pi/BirdSongs/Processed
13:15:50 extract_new_birdsounds.sh[511429]: Moving processed files to /home/pi/BirdSongs/Processed
13:15:50 extract_new_birdsounds.sh[511429]: Moving processed files to /home/pi/BirdSongs/Processed
13:15:50 extract_new_birdsounds.sh[511429]: Linking Processed files to /home/pi/BirdSongs/Extracted/Processed web directory
13:15:50 systemd[1]: extraction.service: Succeeded.
13:16:00 systemd[1]: Started BirdNET BirdSound Extraction.
13:16:00 extract_new_birdsounds.sh[511478]: Moving processed files to /home/pi/BirdSongs/Processed
13:16:00 extract_new_birdsounds.sh[511478]: Moving processed files to /home/pi/BirdSongs/Processed
13:16:00 extract_new_birdsounds.sh[511478]: Moving processed files to /home/pi/BirdSongs/Processed
13:16:00 extract_new_birdsounds.sh[511478]: Linking Processed files to /home/pi/BirdSongs/Extracted/Processed web directory
13:16:00 systemd[1]: extraction.service: Succeeded.
13:16:10 systemd[1]: Started BirdNET BirdSound Extraction.
13:16:10 extract_new_birdsounds.sh[511525]: Moving processed files to /home/pi/BirdSongs/Processed
13:16:10 extract_new_birdsounds.sh[511525]: Moving processed files to /home/pi/BirdSongs/Processed
13:16:10 extract_new_birdsounds.sh[511525]: Moving processed files to /home/pi/BirdSongs/Processed
13:16:10 extract_new_birdsounds.sh[511525]: Linking Processed files to /home/pi/BirdSongs/Extracted/Processed web directory
13:16:10 systemd[1]: extraction.service: Succeeded.
```

So I restarted birdnet analysis service and now it is working again and catching up. But it is kind of annoying that my pi 3B is by far not working as reliable as the 4B was. This 3B uses the same SD card as the 4B before, I only adjusted recording time to 30 s and activated zRAM and installed the latest updates.

I do not know much about experiences of others regarding the 3B. But since I seem to encounter more problems with 3B I am thinking about to switch to 4B, again. Unfortunately it is sold out everywhere and I only have one 4B with 8 GB (which I wanted to use for something else). So, I would like to know, what you think about the pi 3B and why it is not as reliable as the 4B. Or am I maybe the only one who encounters reliability problems with the 3B? Can this maybe be related to use zRAM instead of swap?

Also, I would like to ask if it would be possible to run an email service (once a day or better night) to inform the user if services stop or errors occur. Could be used to give an daily or weekly overview of recognitions, too. 

This is the pi you can log into, so feel free to examine the logs and system, if you want.

mcguirepr89, 2022-02-19:
Hi, Oliver -- Thank you for offering remote access -- I will log in and see what I can find. Though, just guessing here before investigation, it sounds like this installation still uses RaspiOS-Full, which is probably why your resources and performance are an issue. I'll let you know what I find, though.

mcguirepr89, 2022-02-19:
Actually, I was not able to connect via NoMachine (which is a good sign indicating no Desktop environment). It is the installation at  birdnetgv.ddnss.de?

oliverk71, 2022-02-19:
Yes, it's true, since it was prior running on the 4B, it was the raspian with desktop, which I installed. I just swapped the SD card from 4B to the 3B. But I configured the pi via raspi-config to not boot into desktop. And yes, it is the birdnetgv.ddnss.de installation.

oliverk71, 2022-02-19:
In general, it works and is not too slow and it catches up slowly as soon all the services are running. The only thing is the reliability so that the services keep running.

oliverk71, 2022-02-19:
If necessary, I can forward SSH port to public internet temporarily.

mcguirepr89, 2022-02-19:
Yes forward SSH and email me credentials and I will take a closer look, but I'm fairly certain the issue is using RaspiOS full for the 3B.

oliverk71, 2022-02-19:
I just forwarded port 22. Isn't it the same password as for Nomachine?

mcguirepr89, 2022-02-19:
Yes, let me see if I can find those credentials (NoMachine saved them but I did not)

mcguirepr89, 2022-02-19:
Looks like there is a firewall enabled for port 22 on your network:
```
patrick @ mcchrome ➜  ~  sudo nmap -p22 birdnetgv.ddnss.de
Starting Nmap 7.80 ( https://nmap.org ) at 2022-02-19 09:39 EST
Nmap scan report for birdnetgv.ddnss.de (46.89.145.102)
Host is up (0.18s latency).
rDNS record for 46.89.145.102: p2e599166.dip0.t-ipconnect.de

PORT   STATE    SERVICE
22/tcp filtered ssh
```

oliverk71, 2022-02-19:
Oh, sorry, I forgot to hit Return, lol.

mcguirepr89, 2022-02-19:
Ok I'm in and will let you know what I find and whether I make any changes

mcguirepr89, 2022-02-19:
I do think it was the zRAM kernel module -- I disabled it and re-enabled the regular swap (at 2G instead of the default 100M) and the system looks like it is in good shape. I lowered the recording length to 15 seconds, since that is what I use for my system and it looks like it is operating well.

I apologize that the zRAM recommendation turned out to be an issue -- I monitored the CPU for a while (`top -o %CPU` then press `c`) and watched the kernel module spike the CPU often. After disabling it, no more spikes and the regular swap has taken back over.

Keep an eye on things and let me know if it gets backed up again and I'll take another look.

My best regards,
Patrick

oliverk71, 2022-02-19:
There is absolutely no need to apologize. Thanks a lot, again, Patrick! I will keep an eye on it 

It does not seem to be behind, now, but I am not sure if it caught up or you maybe skipped them. I was wondering if the recordings between 14:00 and 16:00 have already been analyzed, that's why I am asking.

mcguirepr89, 2022-02-19:
I restarted all services after changing the swap policy which removed un-analyzed, queued recordings. It was necessary to do that to allow the system a chance to get going without having a queue to take care of first. You _shouldn't_ see things queue up more than a few minutes (when it is creating database entries when the birds are quite talkative), but should catch back up if it does get a little behind from database entries and creating extractions.

oliverk71, 2022-02-19:
Okay. What about the history of recognized birds? Because I do not get any data from the past.

mcguirepr89, 2022-02-19:
>What about the history of recognized birds? Because I do not get any data from the past.

I don't think I understand. 
Your History page is loading fine with your data:
[image]

To what are you referring?

oliverk71, 2022-02-19:
I swear it does not show anything a few minutes ago. And it doesn't for most other dates. Can it be that you looked up the last two days? Because I get data for the last two days, suddenly, but not for any other date I tried. 

[image]

oliverk71, 2022-02-19:
May have something to do with date format?

oliverk71, 2022-02-19:
I also wanted to check a recording, because a gnome owl (Sperlingskauz in German) was detected, which could be a false recognition. However, I can see the recognition in the chart, but cannot find the recording. 

[image]
[image]

mcguirepr89, 2022-02-19:
This has to do with the way by which the history page looks up the Charts. I had to change the chart naming convention so that the HTML forms can address them. The charts that were created before that update use a different naming convention. To get those older charts to show up, you'll need to change the file names from the way they are currently:
[image]
To using "Combo[2]-YEAR-MONTH-DAY.png", so the first file in the screen shot above would need to be renamed "Combo-2022-02-01.png" so that the HTML can find it. I know this is inconvenient, but I'm working through the various places in the system that fetch dates to ensure that it is standardized for code (and **not** for locale)

mcguirepr89, 2022-02-19:
Regarding the missing files, I will see what I can find for that as well, but it is likely that those extractions were not made because of the previous resource issues. I'll try to find log evidence of that supposition, though.

mcguirepr89, 2022-02-19:
Your system's server.py service went down at 8:28am and didn't come back up until 1:18pm. 
```
Feb 19 08:28:01 birdnetpi server.py[117147]: ANALYZING AUDIO... Exception in thread Thread-1139:
Feb 19 08:28:01 birdnetpi server.py[117147]: Traceback (most recent call last):
Feb 19 08:28:01 birdnetpi server.py[117147]:   File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/urllib3/connection.py", line 174, in _new_conn
Feb 19 08:28:01 birdnetpi server.py[117147]:     conn = connection.create_connection(
Feb 19 08:28:01 birdnetpi server.py[117147]:   File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/urllib3/util/connection.py", line 72, in create_connection
Feb 19 08:28:01 birdnetpi server.py[117147]:     for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):
Feb 19 08:28:01 birdnetpi server.py[117147]:   File "/usr/lib/python3.9/socket.py", line 953, in getaddrinfo
Feb 19 08:28:01 birdnetpi server.py[117147]:     for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
Feb 19 08:28:01 birdnetpi server.py[117147]: socket.gaierror: [Errno -5] No address associated with hostname
Feb 19 08:28:01 birdnetpi server.py[117147]: During handling of the above exception, another exception occurred:
Feb 19 08:28:01 birdnetpi server.py[117147]: Traceback (most recent call last):
Feb 19 08:28:01 birdnetpi server.py[117147]:   File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/urllib3/connectionpool.py", line 703, in urlopen
Feb 19 08:28:01 birdnetpi server.py[117147]:     httplib_response = self._make_request(
Feb 19 08:28:01 birdnetpi server.py[117147]:   File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/urllib3/connectionpool.py", line 386, in _make_request
Feb 19 08:28:01 birdnetpi server.py[117147]:     self._validate_conn(conn)
Feb 19 08:28:01 birdnetpi server.py[117147]:   File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/urllib3/connectionpool.py", line 1040, in _validate_conn
Feb 19 08:28:01 birdnetpi server.py[117147]:     conn.connect()
Feb 19 08:28:01 birdnetpi server.py[117147]:   File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/urllib3/connection.py", line 358, in connect
Feb 19 08:28:01 birdnetpi server.py[117147]:     conn = self._new_conn()
Feb 19 08:28:01 birdnetpi server.py[117147]:   File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/urllib3/connection.py", line 186, in _new_conn
Feb 19 08:28:01 birdnetpi server.py[117147]:     raise NewConnectionError(
Feb 19 08:28:01 birdnetpi server.py[117147]: urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPSConnection object at 0x7f91eaa5b0>: Failed to establish a new connection: [Errno -5] No address associated with hostname
Feb 19 08:28:01 birdnetpi server.py[117147]: During handling of the above exception, another exception occurred:
Feb 19 08:28:01 birdnetpi server.py[117147]: Traceback (most recent call last):
Feb 19 08:28:01 birdnetpi server.py[117147]:   File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/requests/adapters.py", line 440, in send
Feb 19 08:28:01 birdnetpi server.py[117147]:     resp = conn.urlopen(
Feb 19 08:28:01 birdnetpi server.py[117147]:   File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/urllib3/connectionpool.py", line 785, in urlopen
Feb 19 08:28:01 birdnetpi server.py[117147]:     retries = retries.increment(
Feb 19 08:28:01 birdnetpi server.py[117147]:   File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/urllib3/util/retry.py", line 592, in increment
Feb 19 08:28:01 birdnetpi server.py[117147]:     raise MaxRetryError(_pool, url, error or ResponseError(cause))
Feb 19 08:28:01 birdnetpi server.py[117147]: urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='app.birdweather.com', port=443): Max retries exceeded with url: /api/v1/stations/RkibKnswM3EV1WwxuQcuQ8xs/soundscapes?timestamp=2022-02-19T08:27:15+01:00 (Caused by>
Feb 19 08:28:01 birdnetpi server.py[117147]: During handling of the above exception, another exception occurred:
Feb 19 08:28:01 birdnetpi server.py[117147]: Traceback (most recent call last):
Feb 19 08:28:01 birdnetpi server.py[117147]:   File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
Feb 19 08:28:01 birdnetpi server.py[117147]:     self.run()
Feb 19 08:28:01 birdnetpi server.py[117147]:   File "/usr/lib/python3.9/threading.py", line 892, in run
Feb 19 08:28:01 birdnetpi server.py[117147]:     self._target(*self._args, **self._kwargs)
Feb 19 08:28:01 birdnetpi server.py[117147]:   File "/usr/local/bin/server.py", line 385, in handle_client
Feb 19 08:28:01 birdnetpi server.py[117147]:     response = requests.post(url=soundscape_url, data=wav_data, headers={'Content-Type': 'application/octet-stream'})
Feb 19 08:28:01 birdnetpi server.py[117147]:   File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/requests/api.py", line 117, in post
Feb 19 08:28:01 birdnetpi server.py[117147]:     return request('post', url, data=data, json=json, **kwargs)
Feb 19 08:28:01 birdnetpi server.py[117147]:   File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/requests/api.py", line 61, in request
Feb 19 08:28:01 birdnetpi server.py[117147]:     return session.request(method=method, url=url, **kwargs)
Feb 19 08:28:01 birdnetpi server.py[117147]:   File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/requests/sessions.py", line 529, in request
[... cut ...]
Feb 19 13:18:19 birdnetpi server.py[117147]: MySQL connection is closed
Feb 19 13:18:19 birdnetpi server.py[117147]: 2022/02/19;08:27:15;Prunella modularis;Heckenbraunelle;0.92363447;51.0914;6.5716;0.7;7;1.25;0.0
```
Looks like there was a service restart right around 1:18pm that got things working again. I do expect this is all related to the system getting backed up and not having the resources available to run properly.

Like I said, keep an eye on things and please let me know anything else you see that is unusual or could use another set of eyes.

oliverk71, 2022-02-19:
> Your system's server.py service went down at 8:28am and didn't come back up until 1:18pm.

Yes, I restarted the service at this time.
 
> Like I said, keep an eye on things and please let me know anything else you see that is unusual or could use another set of eyes.

I will. Thanks again!

closed 2022-02-19 by mcguirepr89