mcguirepr89/BirdNET-Pi #174: Design of UI

joknjokn, 2022-03-27:
Hi,

I was wondering if it's possible implement support for custom design/UI in the webinterface?

I think the UI could benefit alot from using an open source design framework, i.e. Google's Material Design:
https://github.com/material-components/material-components-web

If it's not possible to do per-installation, I would volunteer to provide an overall layout suggestion, some styling of components (buttons etc.) and similar. I'm not sure how/where you've impemented the current UI, so not sure how to start. IF you think it could be interesting, maybe a way forward could be that you add the framework so it's available where you generate the UI, and then maybe let me know how I can play around with the UI myself.

Let me know if it could be interesting. :)

- Jonas

Svardsten53, 2022-03-27:
In the 'newif' version, V0.13, Patrick has put a lot of effort to create a new and more modern layout compared to the previous one. I like the new modern and clean style.

I have done some translations and small adjustments by editing files in the "homepage" and "scripts" directories. As long as it is not about more extensive changes, it is easily done there.

joknjokn, 2022-03-27:
@Svardsten53 Thanks for the reply. I did test the V0.13 of newif, and I'm definitely not saying that Patrick effort isn't great. I don't know if he's working with UI as a profession - I am - so just wanted to ask if he wanted my help with something that is technolocigally modern and easy to reuse with a lot of predifned UI componenets (not sure what he's using now), as well as responsive on multiple devices. It also comes down to taste. It's not that I don't like his UI, there are just some things that - in my opinion - could be optimized further. :)

mcguirepr89, 2022-03-27:
@joknjokn I 100% accept your offer to take part (lead) in UI design! I hate it! I've been learning it all as I go and it means I've slapped a lot of stuff together and have been going back through gradually to make things better as I realize how ugly things truly are :)

I'm not familiar at all with UI design, and don't know what the best (current, modern) approach is, so I've been doing my best to just make things work since I don't know the "right" way to do many things. What I would really like to do is set up a time to chat with you to see what your thoughts are and to explain as best I can the approach I have used thus far.

I'm hoping, too, that a smarter UI would allow for localization (I haven't forgotten your wonderful contributions!) in a clever way.

I'm very interested in optimizing the UI as much as possible and for allowing users the level of control and customization that FOSS allows (though I need to be sure to stop short of building a custom Bird CMS, as I'm no WordPress and nor should BirdNET-Pi be).

Please let me know what you would need from me and I will get it to you -- also, I'm very interested in a zoom/google meets chat if you might be interested. If so, please [email me](mailto:mcguirepr89@gmail.com) and we'll set something up.

Very excited to see what comes of this!

My best regards and gratitude for your kind offer,
Patrick

PS
Thank you for your kind words, @Svardsten53 !!

joknjokn, 2022-03-27:
@mcguirepr89 Just sent you an email :)

nilspupils, 2022-03-27:
@joknjokn: Thanks for the great offer, very much looking forward to seeing it. (I did really like "newif" though).
Best regards, 
Nils

joknjokn, 2022-03-28:
Aaaalright, so I figured out how I can mess around with the UI in my installation.

I imported the Material Design standard CSS and JS (modified theme colors) and a bit of official template code.

I had to rearrange some of the code you use to handle page content (i.e. I’m switching a lot of it to _GET instead of _POST) and a bunch of other things. Unsure if some of it breaks other things I’m not aware of. Also, it would need more tweeks for better responsibility etc. (somethings are responsive automatically, other stuff needs some classes/media-rules attached).

But – What do you think of the general idea? Screenshots are attached. One is a Quick attempt at the overview/FrontPage with some dashboard-ish-stuff, and another is with the Tools-menu expanded. Most buttons/links works.

Absolutely no hard feelings if this is not to your taste 😉

IF you think it’s worth continuing this, let me know, and we can discuss the approach.

  *   Jonas

Please let me know what you would need from me and I will get it to you -- also, I'm very interested in a zoom/google meets chat if you might be interested. If so, please email ***@***.***> and we'll set something up.

mcguirepr89, 2022-03-28:
Hi, @joknjokn!! I'm excited to see, but I don't think the images attached correctly. You may need to try again?

joknjokn, 2022-03-28:
@mcguirepr89 Sorry - The message was intended for your mail, not for GitHub. Trying again.

mcguirepr89, 2022-04-08:
Hi, Jonas -- I just wanted to put a little update here that I've got the new system's core services assembled and writing to the SQLite database. The next steps are:
- Adding more fields to the detections table (or creating a new table for system metadata)
- Adding a weather table that fetches weather data from wttr.in
- Exposing the DB through FastAPI
- Building out the Material Design UI

I'll add a Caddy webserver to the installer soon, but for now I'll be configuring that manually and adjusting the Caddyfile as needed for testing.

If you have any time to share a screen, I'd love to show you the new system in action so that you may have an idea of where development is in terms of setting the stage for Material Design.

My best!
Patrick

joknjokn, 2022-04-13:
Hi Patrick,

Thanks for the update, and sorry for the late response. We’re having a super busy month at my job (the app Monta EV Charging. An exploding startup).

I definitely would like to see it soon, and it’s nice to see the updates, but unsure about when I can focus on anything but work in the next few weeks to be honest. It will probably be a bit ad hoc, when I find a less stressful day. 😊

Also, don’t hesitate to ask me, if you want input about something, i.e. how to structure/what to call the api endpoints and parameters.

By the way, my BirdNET-Pi with my customized UI is still running without interruptions. It must be several weeks now. Super fun!

  *   Jonas

CaiusX, 2022-04-26:
Hi @joknjokn 

We're slowing back end development on the Analyzer path until the open source 3K model is released (probably within the next month), but feel we shouldn't let that hold  up thinking about UI.  My only experience thusfar is that Streamlit page and the ubiquitous daily_plot, but would be keen to learn & collaborate on Material Design/Fast API/JSON integration etc.

Best

mcguirepr89, 2022-08-22:
Howdy, @joknjokn 

I learned Laravel :)

The new repo will be built using the newest model and Laravel 9.

Thanks for the great inspiration! :)

closed 2022-08-22 by mcguirepr89

joknjokn, 2022-10-11:
Ohhh, that’s cool to hear! 😃

I assume you like it, then. If it’s going to support remote connection (rest API), I’ll probably try to play around with a basic Flutter-app shell that connects to the BirdNET Pi api (when I find time).

Did you also go for a UI library/framework, or?  (Like Material or similar)

Is  BirdNET model 2.0 still on hold?

  *   Jonas

mcguirepr89, 2022-10-11:
Howyd!

Yes, still on hold, but largely built and working.

I just am using Laravel and its UI framework (Mix I think it's called) and am using livewire also (and Tailwind for the CSS of course)

You can check out what I have so far here https://github.com/mcguirepr89/BirdNET-Analyzer-Pi

I'll likely be only releasing images for that (or a docker)