mcguirepr89/BirdNET-Pi #175: Feature request: time-scheduled recordings

DD4WH, 2022-03-30:
For research purposes it would be desirable to have the ability of BirdNET-Pi to perform one additional task beside the "normal" functions:

* make recordings on the basis of a user-defined time-schedule
* example: record in a recording window from 3:00 AM to 8:00 AM
* record a 60 sec-recording, save to SD card and pause for 240 seconds
* do this until time is outside the recording window
* the recordings should be saved in a separate folder for time-scheduled recordings separated from the extractions of BirdNET-Pi

User-defined settings could be:

* recording window 1: eg. 3:00AM-8:00AM
* recording window 2: eg. 17:00PM-20:00PM
* recording window 3: eg. 22:00PM-01:00AM
* recording length: eg. 60 seconds
* recording pause: eg. 240 seconds
* if recording pause  == 0, the BirdNET-Pi makes continuous recordings (length is == recording length) when time is inside one of the recording windows

Do you think this would be feasible ? 
It would be very helpful for researchers to have original (not yet classified) recordings from a specified recording schedule. You could then apply different Classifiers on them. Also very good, if models change, so you can rerun a newer model on the old recordings.

Would be awesome if this could be implemented!

mcguirepr89, 2022-03-30:
:thinking: interesting request!

My thoughts:
That should be doable -- even though I've just spent a lot of time reworking the web interface, I've recently been turned onto [a different framework altogether](https://material.io/design/introduction), which I'm hoping to be able to implement for the new model. Prior to thinking along the lines of making all of the data and tools available via an API, I would have put together a new page with input fields for these settings, and then have that feed variables into a (likely bash) script and cron (or systemd) to schedule things. Now I'm thinking that the API could have an endpoint for addressing the recording service, which may be scheduled some other way. Not sure, yet -- still learning what I might need to make the BirdNET-Pi data available as an API (probably as a connection to the SQLite database and then endpoints for some PHP and other server-side scripts).

I know this is a feature request, and telling you how to do it yourself isn't exactly fulfilling the feature request, but I want you to have a way to do this sooner than later and know that the "right" and "best" way to do this is not going to be in place very soon.
So, in true Patrick fashion, to get this working for you on the current BirdNET-Pi (`newif`),  here are some steps:

1. Make a copy of the current `birdnet_recording.service` that you can use for the new `custom_recording.service`
   ```
   cd ~/BirdNET-Pi
   sudo cp ./templates/birdnet_recording.service ./templates/custom_recording.service
   ```
2. Move the new service file to the system lib:
   ```
   sudo ln -sf /home/pi/BirdNET-Pi/templates/custom_recording.service /lib/systemd/system
   ```
3. Create a new file for the custom recording -- the command below will create the file and make it executable for you:
   ```
   cat << EOF > /home/pi/BirdNET-Pi/scripts/custom_recording.sh
   #!/usr/bin/env bash
   set -x
   source /etc/birdnet/birdnet.conf

   STAMP="%H:%M:%S"

   [ -z $RECORDING_LENGTH ] && RECORDING_LENGTH=15

   if ! pulseaudio --check;then pulseaudio --start;fi

   CUSTOM_WINDOW_START=(0 3 17 22)
   CUSTOM_WINDOW_END=(0 7 19 23)
   RECORDING_DURATION=60
   PAUSE_DURATION=240

   now=$(date +%H)

   while [ $now -ge ${CUSTOM_WINDOW_START[0]} ] && [ $now -le ${CUSTOM_WINDOW_END[0]} ];do
   echo $now
   # If you prefer no directory structure under "Raw", comment out the
   # lines below and uncommend the commands at the bottom.
   arecord -f S16_LE -c${CHANNELS} -r48000 -t wav -d $RECORDING_DURATION \
   --use-strftime ${EXTRACTED}/Raw/%B-%Y/%d-%A/%F-birdnet-${STAMP}.wav

   # Uncomment the lines below if you'd prefer having no directory scructure
   # under the "Raw" directory. Be sure to comment out the command above.
   #arecord -f S16_LE -c${CHANNELS} -r48000 -t wav -d $RECORDING_DURATION \
   #  --use-strftime ${EXTRACTED}/Raw/%F-birdnet-${STAMP}.wav
   sleep $PAUSE_DURATION
   now=$(date +%H)
   done
   while [ $now -ge ${CUSTOM_WINDOW_START[1]} ] && [ $now -le ${CUSTOM_WINDOW_END[1]} ];do
   echo $now
   # If you prefer no directory structure under "Raw", comment out the
   # lines below and uncommend the commands at the bottom.
   arecord -f S16_LE -c${CHANNELS} -r48000 -t wav -d $RECORDING_DURATION \
   --use-strftime ${EXTRACTED}/Raw/%B-%Y/%d-%A/%F-birdnet-${STAMP}.wav

   # Uncomment the lines below if you'd prefer having no directory scructure
   # under the "Raw" directory. Be sure to comment out the command above.
   #arecord -f S16_LE -c${CHANNELS} -r48000 -t wav -d $RECORDING_DURATION \
   #  --use-strftime ${EXTRACTED}/Raw/%F-birdnet-${STAMP}.wav
   sleep $PAUSE_DURATION
   now=$(date +%H)
   done
   while [ $now -ge ${CUSTOM_WINDOW_START[2]} ] && [ $now -le ${CUSTOM_WINDOW_END[2]} ];do
   echo $now
   # If you prefer no directory structure under "Raw", comment out the
   # lines below and uncommend the commands at the bottom.
   arecord -f S16_LE -c${CHANNELS} -r48000 -t wav -d $RECORDING_DURATION \
[... 32 lines cut ...]
   ```
   ExecStart=/usr/local/bin/custom_recording.sh
   ```
4. Enable your new `custom_recording.service`:
   ```
   sudo systemctl enable --now custom_recording.service
   ```

You'll probably intuitively understand how the script works just looking at it, but to give attention to the parts that will interest you that are also a little weird:
```
   CUSTOM_WINDOW_START=(0 3 17 22)
   CUSTOM_WINDOW_END=(0 7 19 23)
```
This section sets the start and end windows. You'll notice the not-so-elegant way of handling your request for the listening windows. The `CUSTOM_WINDOW_START` variables will be hours from 0-23, using the hour you'd like to start the window. The `CUSTOM_WINDOW_END` variables will also be hours from 0-23, but will use _the hour you'd like to stop the window **minus one**_. This basically says, start at the start window hour, and record _throughout_ the stop window hour. So for the the window from 3am to 8am, the `CUSTOM_WINDOW_START` is `3` while the `CUSTOM_WINDOW_END` is `7`. This tells the recording script to start when the hour hits 3am, and continue until the hour reads 8 (so recording _throughout_ the 7th hour).

The _very_ obvious weird part is the window of time from 10pm to 1am. To get that working, I just had to be a little more explicit and basically request: Record from 10pm to 12am, then from 12am until 1am.

Anywho, this works for now. 

To make this easier on yourself, definitely use PuTTY so that you can just copy and paste the steps you need.

Sorry for the epic response that doesn't even add any features! I hope, though, that it will improve your immediate research methods as the seasons change and data becomes more and more exciting.

My best regards as always,
Patrick

DD4WH, 2022-03-31:
Patrick, that is fantastic! Thanks a lot for this ready-to-use solution!
Very nice to hear that you are (again!) experimenting with a different UI solution!?

First test shows that this is working perfectly! I followed exactly your instructions and modified the current v0.13 from the `newif` branch. One problem during the modification was that copy-and-paste in putty seemed to substitute the `now=$(date +%H)` with the hard-coded-figures --> `now=9`. But copying the correct code for the `custom_recording.sh` in the nano editor solved that :-).

Thank you very much for that quick and working solution!
I will report later at this place when 24h have passed.
Keep up the good work!!!

DD4WH, 2022-03-31:
One question:
* do I have to restart the time-scheduled recording when I reboot the Pi ? [by typing `sudo systemctl enable --now custom_recording.service`] ?

DD4WH, 2022-03-31:
> Sorry for the epic response that doesn't even add any features!

Well, that is serious understatement ;-). Your new code turns the BirdNET-Pi into a very serious professional recording device enabling time-scheduled high quality recordings . . .

DD4WH, 2022-03-31:
Problem with downloading solved with WinSCP!

mcguirepr89, 2022-03-31:
That's great! I'm pretty sure that the ZIP function is just really bad at indicating that it is working. I'm trying it out right now and can see that it is still creating the archive, but that the web interface isn't doing anything. I suspect that it will appear when it is finished (take a look around in File Manager again after a little bit to see if the archive shows up eventually). My directory of about 50 recordings took several minutes to appear.

>One question:
do I have to restart the time-scheduled recording when I reboot the Pi ? [by typing sudo systemctl enable --now custom_recording.service] ?

Nope! This is now a full-fledged service that is part of your system. In fact, you'll need to _disable_ it if you want it to stop.
```
sudo systemctl disable --now custom_recording.service
```

Which brings me to one very important oversight:
You'll need to add `custom_recording.sh` to your `stop_core_services.sh` service list so that those recordings stop when your disk is full. That is super easy and can be done in the File Manager.

I'm so glad that this is fulfilling a need of yours and hope that it proves useful when comparing your dataset against various models!

DD4WH, 2022-03-31:
> You'll need to add `custom_recording.sh` to your `stop_core_services.sh` service list so that those recordings stop when your disk is full. That is super easy and can be done in the File Manager.

Thanks Patrick!

Do I need to add `custom_recording.service` to `/home/pi/BirdSongs/Extracted/scripts/restart_services.sh` ?

One additional problem: in the File Manager, I am not able to edit any file, because the file manager indicates I have no permission to do so.

mcguirepr89, 2022-03-31:
>Do I need to add custom_recording.service to /home/pi/BirdSongs/Extracted/scripts/restart_services.sh ?

You certainly can, though this custom service kind of operates outside of the rest of the system, so it is totally up to you and whether you want that service to be addressed when you call `restart_services.sh`.

>One additional problem: in the File Manager, I am not able to edit any file, because the file manager indicates I have no permission to do so. I am using putty instead, which works well.
UPDATE: not sure why, now it works editing and saving files in the File Manager !

I wonder if this had to do with File Manager still working on creating the ZIP archive? Did it ever appear?

DD4WH, 2022-03-31:
> I wonder if this had to do with File Manager still working on creating the ZIP archive? Did it ever appear?

No, I could not find any archive. Should it appear in the same folder where I requested the zip ?

mcguirepr89, 2022-03-31:
Yes, it should have eventually. :thinking: that is an oddity for sure. I suppose I'll have to chalk the permissions issue up to something with Tiny File Manger, since it is obvious that nothing changed from a system/OS perspective.

I'll keep thinking on this in case I can come up with an explanation that makes sense :thinking:

mcguirepr89, 2022-04-02:
I went ahead and added the `custom_recording.service` to the installer, though it is not enabled by default. (I did not, however, add it to "Tools" > "Services" at this point, since there is no GUI to configure the service at this time). This just means that at this point, an installation has the potential to flip this service on and would have the custom_recording.service you suggested.

DD4WH, 2022-04-03:
Patrick,

thanks a lot for including this into the installer!

However, it does not work, unfortunately.

* I installed the latest version
* I typed  `sudo systemctl enable --now custom_recording.service`

mcguirepr89, 2022-04-03:
Dang!!!

I suspect I know what may be going on -- 

Could you do me a favor, though, and either email me you installation log?
Do this:
```
cp ~/install* /home/pi/BirdSongs/Extracted
```
And then your installation log should be available to download in the "Tools" > "File Manager"

I'm going to be doing a test installation in the next half hour or so, so I should encounter the same issue, but your installation log will probably prove very helpful.

DD4WH, 2022-04-04:
Thanks a lot, Patrick!
I can now confirm, this works with the latest nistallation from the main branch !!!
All you have to do is:
* modify `/home/pi/BirdNET-Pi/scripts/custom_recording.sh` for your needs (filename, date format and time windows, recording duration and pause duration)
* in the Web Terminal, type `sudo systemctl disable --now custom_recording.service`

Perfect!

closed 2022-12-13 by mcguirepr89