mcguirepr89/BirdNET-Pi #177: Minor issue: Live Stream button on Android device not working properly

DD4WH, 2022-04-03:
* the Live stream button on Android devices used to be like the button in the webpage and worked very well
* since I installed the version from about three days ago, this button is only a round circle (not the usual play/pause button). Clicking on it starts the audio stream, but there is not way to stop it again

mcguirepr89, 2022-04-03:
Thank you so much for letting me know -- I'll look through the commits for that section of the web page banner to see what kinds of changes I may have made without thinking.

Any chance you might be able to provide a small screenshot of that? If not, no worries, your description ought to suffice.

DD4WH, 2022-04-03:
[image]
Thanks Patrick, for your quick response and never ending enthusiasm 👍 !

mcguirepr89, 2022-04-03:
Yikes!! The table shouldn't be visible on that tiny screen! and everything should be centered!!! Yikes!!!

Thanks, Frank! I'll see what I can find and then let you know

nilspupils, 2022-04-04:
The Live Stream works on my Android phone. 
Not working is Log and Web Terminal. If i click on them, another BirNET-Pi webinterface opens inside the old one....
My system was last updated a couple of days ago, with many iterations, so it might be a bit messy anyways. 
Will do the proper switch to main later today an will see if the problem persists. 
Best, 
Nils

mcguirepr89, 2022-04-04:
Hi, @nilspupils!

>Not working is Log and Web Terminal. If i click on them, another BirNET-Pi webinterface opens inside the old one....
My system was last updated a couple of days ago, with many iterations before that, so it might be a bit messy anyways.
Will do the proper switch to main later today and will see if the problem persists.

Sounds like a mixup from testing and updating and changes. Let me know how the switch goes for you and whether you'd like any additional help/notes.

>I have another issue with intermittent loss of connection to wlan, i first thought it was a problem of my router coverage, but i have since updated everything with a new router and repeater and so have full signal at the pi. The detection runs fine, just the connection is lost intermittently.

This sounds like power-management for the WiFi. Power-management is enabled by default for RPi network cards, so it turns off when there are no (system) users accessing the network. You can disable power-management for the network card by placing this command at the bottom of your `/etc/rc.local` just above `exit 0`:
```
iw wlan0 set power_save off
```

That will disable the power-saving feature for the network card each time your system starts (rather, each time the pi user logs in, which it does automatically via `getty`.)

Hopefully that will address the issue you're seeing with WiFi dropping out. Are you using the built-in NIC, or are you using a WiFi dongle?

nilspupils, 2022-04-04:
Thank you Patrick! 
Power management makes perfect sense, i will try that. (I use the inbuilt NIC). 
Just one last question before the switch: 
My birdsound library has become really nice lately so i want to backup it before the switch. Something must have changed as i cannot copy the files directly from the "by date" folder. Ithink it has to do with symbolic linking, but could't make it work. How can i do it? Many thanks!!!

mcguirepr89, 2022-04-04:
>My birdsound library has become really nice lately so i want to backup it before the switch. Something must have changed as i cannot copy the files directly from the "by date" folder. I think it has to do with symbolic linking, but couldn't make it work. How can i do it? Many thanks!!!

When you say " i cannot copy the files directly from the 'by date' folder," could you let me know what is getting in your way?
To make a guess, maybe you're trying to visit the "By_Date" folder in the web browser? If so, try adding the last forward slash '/' to the address URL, like so: http://birdnetpi.local/By_Date/

If that's not what you meant, but were referring to making a copy of the "By_Date" directory through the "File Manager," you should be able to open the "File Manager," click the check box next to "By_Date," then "ZIP" at the bottom. It will take a really long time and it will seem like nothing is happening, but just be super patient and let it run until you see the "By_Datexxxxxx.zip" archive appear in the "File Manager" at which point you should be able to download that .zip file.
[image]
Once it is created, you'll see it appear:
[image]

nilspupils, 2022-04-04:
I am using VNC and just copy the whole "by_date" folder in the file manager, when i want to paste it to my USB-stick it tells me file is already existing (it is not). If i press enter it seems to do something but the copied files have 0 byte. 
I will try the other way with compressing from the birdnet file manager. 
Thank you!!!

nilspupils, 2022-04-04:
Wasn't able to reinstall because of several errors, but managed to use the old data and extractions for the fresh install.
All issues resolved, stream is working on Android here. System is looking really nice!

mcguirepr89, 2022-04-05:
@DD4WH 
Thanks again for your help -- I've pushed the changes now and an update ought to get you that fix:
https://github.com/mcguirepr89/BirdNET-Pi/commit/5f4be58fb5d8d00be03696d03b88f0445f8bae03

closed 2022-04-05 by mcguirepr89