mcguirepr89/BirdNET-Pi #179: Minor issue: Species detected today --> wrong link

DD4WH, 2022-04-05:
Overview page:
* in the table at the left side of the page: if I click on the number of "Species Detected Today", I am redirected to the page wit ALL the species detected
* it should be the page with only the species detected today

mcguirepr89, 2022-04-05:
Great catch! I'll let you know when an update will fix that link for you :+1:

mcguirepr89, 2022-04-05:
Alright, I've made this change. Go ahead and do an update via "Tools" > "System Controls" > "Update" and then let me know if that link target is what you had in mind.

mcguirepr89, 2022-04-07:
Howdy, @DD4WH -- I just wanted to follow up on this to see if I've fixed the target the way you had in mind. To me, it now makes more sense where those links take you.

DD4WH, 2022-04-13:
Sorry for my late reply! I missed your fix and your response!
YES, now the link goes to the right page!
Thank you!

closed 2022-04-13 by DD4WH