mcguirepr89/BirdNET-Pi #180: Update script calls wrong filename

bane1202, 2022-04-05:
The update script calls sudo: /home/pi/BirdNET-Pi/scripts/update_birdnet2.sh: command not found .

/update_birdnet.sh exists but not "2"

Version 0.11.1

mcguirepr89, 2022-04-05:
Hi, @bane1202 
Thank you so much for you interest in BirdNET-Pi! I'm glad to learn that you have a system that you are interested in updating.

However, there have been very substantial changes from version 0.11.1 to the current version 0.13 which are not supported by an update. To name a few of the changes that prevent this update:

- The Extractions have a different directory structure and no longer use any symbolic links for sorting
- The MariaDB has been replaced with a single SQLite database file
- Custom URLs are now just 1 Custom URL
- GoTTY servers have a new `bind` policy that enforces `localhost` only . . .
- . . . for which Caddy handles using new reverse_proxies with path matchers
- The `detections` table now has a new field for `File_Name` (which is not present in the MariaDB's `detections` table), complicating the process of converting the MariaDB to an SQLite DB (which is possible, but you would have to manually add the new "File_Name" field to all you old `detections` table entries and then either generate the file names to populate the "File_Name" field, or set them to default `NULL`.)

If you are interested in updating your system from v0.11.1 to v0.13, you will actually, in essence, uninstall the old version and will reinstall the newest version.

If you are okay with starting your data-collection from scratch on a new system, you can get the updated system this way:
```
uninstall.sh
cd ~/BirdNET-Pi
git pull -f
./scripts/install_birdnet.sh && sudo reboot
```

That will uninstall the old system, will pull the updates from the repo, and then will re-install the new system.

I apologize that this particular update is not supported -- I always hope to be able to offer new versions to which folks can easily update, but the number and types of changes that went on under the hood for version 0.13 really have prevented my being able to offer an easy way to update.

Please let me know if you encounter any issues installing the new system, or if you have questions on how to maintain your older system.

My best regards,
Patrick

closed 2022-04-05 by bane1202

mcguirepr89, 2022-04-05:
Thank you for your kind words! Do you mind my asking where your BirdNET-Pi is listening?

bane1202, 2022-04-05:
Greenville, SC.  Hopefully will be able to share results once I settle on a microphone setup and other aspects such as powering the PI and weatherproofing. It will be set up in a large forest near a river so lots of opportunity for a wide variety of species.

mcguirepr89, 2022-04-05:
That's awesome! I'll be making a trip to Rock Hill, SC soon to visit my grandmother for her birthday -- she lives on the Catawba so I'm picturing that and am very excited to see what you come up with.

I can definitely recommend[ this ](https://github.com/mcguirepr89/BirdNET-Pi/discussions/39#discussioncomment-2180372)as a really great mic if you are comfortable with soldering. That discussion has a lot of really great info to get you started, and there are a few folks who have shared their setups, including [the setup I left in NC when I lived there](https://github.com/mcguirepr89/BirdNET-Pi/discussions/69#discussioncomment-2387820). It uses a lunch meat container for water proofing and little pine box, a spare guitar wall-mount, an elastic hair-tie to hold the mic in place, and some torn up t-shirt material push-pinned into the pine box to eliminate the slight 'boxy' sound the box makes. But there are several other setups there, too that could help with ideas.

Also, for what it is worth, I use [this case](https://www.argon40.com/products/argon-one-v2-case-for-raspberry-pi-4) inside the lunch mean container as the cooling method. It is an aluminum body that adheres to the CPU with a thermal pad and then inside is a fan which can be configured (though the default settings are good). The case exposes the GPIO if you ever want access to them, but the case has a shutdown/restart/poweron button and a trio of header pins inside with a jumper to toggle between being powering ON when initially plugged in, or remaining powered off when initially plugged in, needing to be powered on with the button.

Lastly, if you're interested in a birdnetpi.com subdomain, I am happy to get that setup for when you are ready to share the data. You could be SouthCarolina.birdnetpi.com or greenvillesc.birdnetpi.com.

bane1202, 2022-04-05:
So the uninstall and update seemed to break the installation. 
[image]

Reflashed fresh raspbian and ran install again and yields same issues exactly.  There seems to be piwheels timeouts and ssl errors in the log.  I can ping piwheels.org at port 443.  Looks like some certificate errors.

[installation-2022-04-05.txt](https://github.com/mcguirepr89/BirdNET-Pi/files/8420532/installation-2022-04-05.txt)

I will try and reflash and reinstall again in case it was a transient issue.

mcguirepr89, 2022-04-05:
I took a quick look through and I think the installation was successful -- you'll see those errors for now until there is at least 1 detection in the database. (I will suppress those errors so that it is not so alarming)

What does the log page look like? Are you seeing any errors there?

mcguirepr89, 2022-04-05:
I need to be more precise with my language!
> you'll see those errors for now until there is at least 1 detection in the database. (I will suppress those errors so that it is not so alarming)

You'll see _the errors on the BirdNET-Pi web page_ until there is at least one detection in the databse.

The SSL and timeouts _seemed_ to have resolved eventually during the retries, but the BirdNET-Pi log page will almost definitely let us know.

mcguirepr89, 2022-04-05:
I've encountered the piwheel.org issue with my testing on the other repo, so this is something going on with their servers/SSL certificates. Not sure what is going on, but I'm hoping it is temporary, because it is causing major delays. I'll let you know when I see things clear up on my end and hopefully it will be cleared up for everyone trying to use piwheel.org to fetch packages

bane1202, 2022-04-06:
Left it for the day yesterday and came back and ran the install on a fresh raspbian and it completed fully with no further ssl errors.  Gave it a test call and it identified it and seems to be operational.  Thanks again for the support and I'll circle back when I have time to get the sharing implemented.