mcguirepr89/BirdNET-Pi #181: 16gb filesystem full after 4 days; MariaDB won't start

mattluttrell, 2022-04-06:
I installed last week and have been running on and off with errors the past 4 days. Went to test new PS/mics and received a database connection error. Further digging showed MariaDB won't start because the disc is full:

```
birdnetpi:~/BirdNET-Pi$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        15G   14G     0 100% /
devtmpfs        1.7G     0  1.7G   0% /dev
tmpfs           1.9G     0  1.9G   0% /dev/shm
tmpfs           759M  920K  758M   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
/dev/mmcblk0p1  253M   31M  222M  12% /boot
tmpfs           380M  4.0K  380M   1% /run/user/1000
```

So I think I need to figure out:

1. What's filling it up? Is it related to me letting it run with errors? 
2. Can I delete something now to get going?
3. Some type of script to clean up / purge files or a "Max Data Retention in GBs" variable?

mattluttrell, 2022-04-06:
```
birdnetpi:/var$ sudo du -ahx / | sort -rh | head -n 20
14G	/
11G	/home/pi
11G	/home
9.1G	/home/pi/BirdSongs/April-2022
9.1G	/home/pi/BirdSongs
7.9G	/home/pi/BirdSongs/April-2022/04-Monday
1.9G	/var
1.7G	/usr
1.5G	/var/log
1.2G	/home/pi/BirdSongs/April-2022/05-Tuesday
1.1G	/home/pi/BirdNET-Pi
927M	/usr/lib

```

mattluttrell, 2022-04-07:
To get up and running I deleted all wave files and started MariaDB.

For a permanent solution:
1) 128GB SD card ordered
2) Pi moving to data closet in house w/50' microphone cable and wired internet.
3) Possibly attach 4TB NFS share and symlink to /home/pi/BirdSongs/

mcguirepr89, 2022-04-07:
Hi, @mattluttrell -- 

This is happening because of the update attempt. Starting the update removed your cleanup crontab (among other things) and so your system was able to fill up. 

Since at this point it seems there is nothing to recover, to repair the system, run the installer again.
```
bash /home/pi/BirdNET-Pi/newinstaller.sh
```

Though you may want to just use the new `main` now that there is nothing to recover of this system.

If I'm wrong about there being nothing to recover on this system, I do apologize and can recommend further steps to repair the system, just let me know.

mcguirepr89, 2022-04-08:
I'll go ahead and close this issue, but if you have any questions definitely let me know. Sorry, again, for the complications moving from version 0.11.1 to 0.13!

closed 2022-04-08 by mcguirepr89