mcguirepr89/BirdNET-Pi #186: Birdnet server service problem

oliverk71, 2022-04-15:
Hi @mcguirepr89, 

since update this morning, there is a problem with the birdnet analysis server: 
```
15:09:00 systemd[86567]: birdnet_server.service: Failed to execute /usr/local/bin/privacy_server.py: Exec format error
15:09:00 systemd[86567]: birdnet_server.service: Failed at step EXEC spawning /usr/local/bin/privacy_server.py: Exec format error
15:09:00 systemd[1]: Started BirdNET Analysis.
15:09:00 systemd[1]: birdnet_server.service: Main process exited, code=exited, status=203/EXEC
15:09:00 systemd[1]: birdnet_server.service: Failed with result 'exit-code'.
15:09:10 systemd[1]: birdnet_server.service: Scheduled restart job, restart counter is at 294.
```
I tried restarting birdnet server and client as well as reboot, but it did not help. 

What can I do to get it running again?

oliverk71, 2022-04-15:
I just recognized I had this problem before, but restarting services helped. Here is a screenshot of 9th of April: 
[image]
And here is one from today: 
[image]

mcguirepr89, 2022-04-15:
Hi, Oliver -- thanks for letting me know -- I suspect I may know what is going on, but I have a few commands I'd like you to run if you don't mind. They should help me get an idea of what has happened.

1. Did you see anything unusual during the update?

1. Could you please run the following and send me the output?

   ```
   ls -l /usr/local/bin/privacy_server.py &&
   ls -l $HOME/BirdNET-Pi/scripts/privacy_server.py
   cat $HOME/templates/birdnet_server.service
   ```

I'm hopeful that I will see the issue somewhere within those commands (the log is not really telling us much, is it!?!! Come on, log!!! That's what we need you for!!!)

oliverk71, 2022-04-15:
1. No, I can't remember anything unusual. 
2. ```
lrwxrwxrwx 1 root root 45 Apr 15 12:36 /usr/local/bin/privacy_server.py -> /home/pi/BirdNET-Pi/scripts/privacy_server.py
-rwxr-xr-x 1 pi pi 18858 Apr 15 12:36 /home/pi/BirdNET-Pi/scripts/privacy_server.py
cat: /home/pi/templates/birdnet_server.service: No such file or directory
```

oliverk71, 2022-04-15:
Ah, I see: the path is wrong 
` /home/pi/**BirdNET-Pi**/templates/birdnet_server.service`

mcguirepr89, 2022-04-15:
Oh great catch! I was scratching my head!

oliverk71, 2022-04-15:
```
[Unit]
Description=BirdNET Analysis Server
Before=birdnet_analysis.service
[Service]
Restart=always
Type=simple
RestartSec=10
User=pi
ExecStart=/usr/local/bin/privacy_server.py
[Install]
WantedBy=multi-user.target
```

mcguirepr89, 2022-04-15:
Thank you!

I think I know the issue:

Could you please also send the output of:
```
head -n1 $HOME/BirdNET-Pi/scripts/*.py
```

I think the shebangs are all gone from removing the default `pi` user

oliverk71, 2022-04-15:
Here it is: 
```
==> /home/pi/BirdNET-Pi/scripts/analyze.py <==
import argparse

==> /home/pi/BirdNET-Pi/scripts/daily_plot.py <==
import sqlite3

==> /home/pi/BirdNET-Pi/scripts/plotly_streamlit.py <==
import os

==> /home/pi/BirdNET-Pi/scripts/privacy_server.py <==
import socket 

==> /home/pi/BirdNET-Pi/scripts/server.py <==
import socket 
```

mcguirepr89, 2022-04-15:
YEP!

Ok I will send you a command soon that will fix things for you -- thank you so much for your help!

oliverk71, 2022-04-15:
Perfect! :)

oliverk71, 2022-04-15:
True, all the shebangs are deleted from the python scripts, but not from the shell scripts.

mcguirepr89, 2022-04-15:
This is all because I had to make the installation a bit more dynamic. Now, it detects the $HOME directory for the user and then creates the shebangs (the top line in scripts that tells the system which language interpreter to use to run the code) dynamically.

The update removes the hard-coded shebangs from when all installations had a `pi` user -- now that there is no default `pi` user, the update removed the shebangs from the scripts and then you'll need to run the installer again to fetch the values (even though they haven't changed on your particular installation, they changed upstream). I'm testing this right now on my Pi0W2 installation using just the Update button and then running ```$HOME/BirdNET-Pi/scripts/install_birdnet.sh``` to recreate the shebangs and ensure all your services are formatted correctly (they should be).

I will let you know when my test has finished and then will provide the exact steps you'll need.

mcguirepr89, 2022-04-15:
DON'T RUN THE INSTALLER AGAIN!! It will wipe away your database, as I've just learned/remembered :)

oliverk71, 2022-04-15:
Okay. I maybe can insert the shebangs manually, but don't know if it leads to other problems, later. 

Good to know! I won't start the installer again!

mcguirepr89, 2022-04-15:
This is the easiest solution:

```
my_dir=$HOME/BirdNET-Pi
echo "Ensure all python scripts use the virtual environment"
sed -si "1 i\\#\!$HOME/BirdNET-Pi/birdnet/bin/python3" $my_dir/scripts/*.py
```

That's just the excerpt from the installer that takes care of that step. I'll add it to the update script to be sure that this doesn't happen to everyone (it will, but they'll be able to run a second update to fix things)

Please let me know if that resolves the problems you're seeing.

mcguirepr89, 2022-04-15:
The recommended update procedure for installations that were installed with the default `pi` user:

1. Go to "Tools" > "Web Terminal"
2. Type ```git pull```
3. Go to "Tools" > "System Controls" > "Update" > "Ok"

oliverk71, 2022-04-15:
> Please let me know if that resolves the problems you're seeing.

Yes, it does! Although ... the recognition chart is not updated somehow. 

> The recommended update procedure for installations that were installed with the default `pi` user:
> 
>     1. Go to "Tools" > "Web Terminal"
> 
>     2. Type `git pull`
> 
>     3. Go to "Tools" > "System Controls" > "Update" > "Ok"

Do I need to do step 2 every time or only once?

oliverk71, 2022-04-15:
> Although ... the recognition chart is not updated somehow.

Sorry, it is being updated by now, but I lost the last 3 hours.

mcguirepr89, 2022-04-15:
I'm sorry, Oliver :/ looks like that data just wasn't able to collect while things were broken -- I HOPE YOU DIDN'T HAVE ANY RARE OR ENDANGERED VISITORS IN THE LAST THREE HOURS!!!!

oliverk71, 2022-04-15:
Most possibly not.

mcguirepr89, 2022-04-15:
I know this seems like a pattern wherein updates always cause issues, but the changes that happened upstream for the Raspberry Pi Foundation were to a standard that has been around since they came out in 2012. The changes I had to make to BirdNET-Pi were pretty substantial, so this wasn't a super straight-forward round of updates. I know I've said it before, but I don't expect updates to be such an issue in the future, there were simply too many fundamental changes to be able to make it smooth on the first shot (also super difficult and time consuming to test [install and old OS with the default Pi, install BirdNET-Pi from a commit that predates the changes which is SUPER difficult to get right in itself, then run the update and try to pinpoint errors]).

I appreciate your understanding and hope you accept my contrition for your lost data as consolation.

My best as always,
Patrick

PS I'm going to leave this up but rename it to "Issues updating" so that others might find it when hunting for their problem

oliverk71, 2022-04-15:
Would it make sense to do a clean install and copy the birds.db to get around possible problems with the changes made by the Rasperry Pi Foundation you mentioned?

closed 2022-04-24 by mcguirepr89