mcguirepr89/BirdNET-Pi #204: Ability to run this in a VM and use an rtsp stream for the audio

danmed, 2022-04-18:
I would like to run this on a Proxmox LXC (running bullseye) and some how pass it the audio from one of my external cameras (as it's already running 24/7)..

Do you know of any way i can make this work?

Many thanks

marclava, 2022-04-18:
If Pulseaudio can be used as a source: https://roc-streaming.org

danmed, 2022-04-18:
> If Pulseaudio can be used as a source: https://roc-streaming.org

Sorry, i don't understand your comment.. are you saying my method work if i can use roc-streaming to send the rtsp audio from the camera?

mcguirepr89, 2022-04-18:
Hmmm :thinking: I would bet this is possible . . .

I think you'd want to just change the `birdnet_recording.service` to use a new script that reads from an rtsp stream and writes to 15-second long files in the `RECS_DIR` that is configured in `$HOME/BirdNET-Pi/birdnet.conf`. Right now, things are expected to be in a nested directory structure by date, but you can also put single WAVE files in whatever is configured as the "`RECS_DIR`" (`$HOME/BirdSongs/`) and they will get picked up in analysis and extraction just like the rest.

I think the script for to read from the rtsp stream and write to a file might look something like this (using external camera IP 192.168.241.1 and a google search [with this result](https://stackoverflow.com/questions/16259588/how-to-dump-raw-rtsp-stream-to-file)):
```
#!/usr/bin/env bash
source /etc/birdnet/birdnet.conf
while true;do
  ffmpeg -i rtsp://@192.168.241.1:62156 -acodec copy  $(date "+%F-%H:%M:%S") -t15
done
```
or from @marclava's comment you could use `roc-recv` is seems instead of ffmpeg. 

Other than that, you'd just expose port 80 (and 443 if you want SSL certificates as well) of the container to the host via a bridged network I suppose?

I'm happy to help out with some tinkering to get it to work :)

danmed, 2022-04-18:
> Hmmm 🤔 I would bet this is possible . . .
> 
> I think you'd want to just change the `birdnet_recording.service` to use a new script that reads from an rtsp stream and writes to 15-second long files in the `RECS_DIR` that is configured in `$HOME/BirdNET-Pi/birdnet.conf`. Right now, things are expected to be in a nested directory structure by date, but you can also put single WAVE files in whatever is configured as the "`RECS_DIR`" (`$HOME/BirdSongs/`) and they will get picked up in analysis and extraction just like the rest.
> 
> I think the script to read from the rtsp stream and write to a file might look something like this (using external camera IP 192.168.241.1 and a google search [with this result](https://stackoverflow.com/questions/16259588/how-to-dump-raw-rtsp-stream-to-file)):
> 
> ```
> #!/usr/bin/env bash
> source /etc/birdnet/birdnet.conf
> while true;do
>   ffmpeg -i rtsp://@192.168.241.1:62156 -acodec copy  $(date "+%F-%H:%M:%S").wav -t15
> done
> ```
> 
> or from @marclava's comment you could use [`roc-recv`](https://roc-streaming.org/toolkit/docs/manuals/roc_recv.html) it seems instead of ffmpeg.
> 
> Other than that, you'd just expose port 80 (and 443 if you want SSL certificates as well) of the container to the host via a bridged network I suppose?
> 
> I'm happy to help out with some tinkering to get it to work :)

Nice, thankyou..

I'll definately take a look at this method and get back to you with any results!

marclava, 2022-04-18:
Roc-streaming is tp stream remote pulseaudio sources, so I don't know if it'd work with rtsp sources...
I was able to stream audio from an Android phone, using this app:
https://github.com/roc-streaming/roc-droid
That could be an easy way to deploy microphone(s) using wifi.

mgottholsen, 2022-04-19:
I came to request this same feature :) I have a network that spans 8 acres with roughly 13 cameras that get real time audio/video and I experimented converting their audio streams to spectrograms to run machine learning on for recognition.

I would love to help test or put together a guide for how to do this. I'll test out the method in this thread.

Sieboldianus, 2022-04-19:
I would also be interested in this. I have a Raspberry PI Zero WH that I could use to capture audio, but I would like to forward it ot my Proxmox cluster with VMs for processing, since the Raspi Zero is too slow. I also doN't want to add another energy consuming device, when I already have all the hardware needed.

evanbills, 2022-04-20:
The ability to leverage a stream from a security camera is _**highly**_ desirable.

ehpersonal38, 2022-04-20:
Okay, I've spent a bit of time on this (although this solution is very rough, I have very limited bash knowledge).

1. In BirdNet-Pi, go to Tools -> Services, then DISABLE Recording Service.
2. Create a new .sh file anywhere, paste this in and replace {YOUR_STREAM_URL} with your rtsp stream url:

```
#!/usr/bin/env bash
source /etc/birdnet/birdnet.conf

if [ "${TIMESTAMP_FORMAT}" == "12" ];then
  STAMP="%I:%M:%S%P"
else
  STAMP="%H:%M:%S"
fi

while true;do
  ffmpeg -i {YOUR_STREAM_URL} -t 15 -acodec copy file:${RECS_DIR}/birdnet-$(date "+%F-%H:%M:%S").wav
done
```

Run this file. Hopefully it will start pulling audio from the URL and putting it where it needs to go for extraction. Ideally this should run as a service not a sh file, but I have no idea how to implement that. Maybe someone can come along and help pick up the pieces?

mgottholsen, 2022-04-21:
Any thoughts on multiple streams? That's probably a separate enhancement?

ehpersonal38, 2022-04-21:
> Any thoughts on multiple streams? That's probably a separate enhancement?

Try this out:

```
#!/usr/bin/env bash
source /etc/birdnet/birdnet.conf

if [ "${TIMESTAMP_FORMAT}" == "12" ];then
  STAMP="%I:%M:%S%P"
else
  STAMP="%H:%M:%S"
fi

while true;do   

ffmpeg -nostats -loglevel 0 -i {YOUR_STREAM_URL} -t 15 -acodec copy file:$(date "+%F")-birdnet-$(date "+%H:%M:%S").wav </dev/null > /dev/null 2>&1 & sleep 1; 
ffmpeg -nostats -loglevel 0 -i {YOUR_STREAM_URL} -t 15 -acodec copy file:$(date "+%F")-birdnet-$(date "+%H:%M:%S").wav </dev/null > /dev/null 2>&1 & sleep 1; 
ffmpeg -nostats -loglevel 0 -i {YOUR_STREAM_URL} -t 15 -acodec copy file:$(date "+%F")-birdnet-$(date "+%H:%M:%S").wav </dev/null > /dev/null 2>&1

done
```

ehpersonal38, 2022-04-24:
Okay I did a bit more testing, and I believe I've got it working. Here are the solutions for now:

**Using a single RTSP stream for BirdNET-Pi**: https://github.com/mcguirepr89/BirdNET-Pi/issues/204#issuecomment-1104298534
**Using multiple RTSP streams for BirdNET-Pi**: https://github.com/mcguirepr89/BirdNET-Pi/issues/204#issuecomment-1105537589

danmed, 2022-04-24:
> Okay, I've spent a bit of time on this, but the solution is a bit hacky for now. Hoping to make this native in a future release of BirdNet-Pi.
> 
> 1. Edit BirdNet-Pi/scripts/birdnet_recording.sh, and replace its entire contents with:
> 
> ```
> #!/usr/bin/env bash
> source /etc/birdnet/birdnet.conf
> 
> while true;do
>   ffmpeg -i (YOUR STREAM URL) -t 15 -vn -acodec pcm_s16le -ac 2 -ar 44100 file:${RECS_DIR}/$(date "+%F")-birdnet-$(date "+%H:%M:%S").wav
> done
> ```
> 
> Make sure you edit (YOUR STREAM URL) and replace it with your rtsp stream url.
> 
> Execute `sudo reboot` and wait for the Pi to reboot.
> 
> Hopefully it will start pulling audio from the URL and putting it where it needs to go for extraction. Please test this and let me know if this works for any of you.

Thank you.. i've finally got this set up.. unfortunately it's now night time here lol..

One quick qiestion though.. it has detected 3 birds.. but i'm unable to playback the sounds.. are you able to?

ehpersonal38, 2022-04-24:
> One quick qiestion though.. it has detected 3 birds.. but i'm unable to playback the sounds.. are you able to?

Hmm... I think it may have to do with the fact that the raw wavs from the rtsp stream (pre-analysis) aren't being written to date specific folders, they're just being chucked in rec_dirs. Any thoughts? @mcguirepr89

mcguirepr89, 2022-04-24:
> chucked in rec_dirs. 

That's good enough :+1: 
line 320 in `birdnet_analysis.sh`
```
if [ $(find ${RECS_DIR} -maxdepth 1 -name '*wav' | wc -l) -gt 0 ];then
  run_birdnet "${RECS_DIR}"
fi
```

danmed, 2022-04-24:
> Hmm... I think it may have to do with the fact that the raw wavs from the rtsp stream (pre-analysis) aren't being written to date specific folders, they're just being chucked in rec_dirs. Any [...]

From the cmdline... where should i be able to find the mp3's?

ehpersonal38, 2022-04-24:
They'd be in /home/pi/BirdSongs/Extracted/By_Date/(species)

danmed, 2022-04-24:
> They'd be in /home/pi/BirdSongs/Extracted/By_Date/(date) /(species)

Ok.. my Extracted By_Date folder is empty for some reason...

ehpersonal38, 2022-04-24:
> > They'd be in /home/pi/BirdSongs/Extracted/By_Date/(date) /(species)
> 
> Ok.. my Extracted By_Date folder is empty for some reason...

I'm not sure what the issue is - but you could try adding `set - x` to the beginning of scripts/extract_new_birdsounds.sh, then restart the extraction service in the web interface and checking the log

danmed, 2022-04-24:
> I'm not sure what the issue is - but you could try adding `set -x` to the beginning of scripts/extract_new_birdsounds.sh, then restart the extraction service in the web interface and checking the log

Are you seeing the same problem?

ehpersonal38, 2022-04-24:
> Are you seeing the same problem?

Nope! All's well with my installation. Does it happen with the microphone too?

danmed, 2022-04-24:
> Nope! All's well with my installation. Does it happen with the microphone too?

I don't have a microphone attached.. the pi is no where near outside :)

mcguirepr89, 2022-04-24:
If anyone knows of a public RTSP stream that can be used for development, it would help tremendously --  just sayin

ehpersonal38, 2022-04-24:
> If anyone knows of a public RTSP stream that can be used for development, it would help tremendously -- just sayin

I did manage to find one! `rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mp4`

mcguirepr89, 2022-04-24:
That's something to work with -- now once you can get audible files written to the disk from that stream in .wav format, we'll be cookin something -- I'll let you know what I come up with when I do take a look here

ehpersonal38, 2022-04-24:
> That's something to work with -- now once you can get audible files written to the disk from that stream in .wav format

Done! Check https://github.com/mcguirepr89/BirdNET-Pi/issues/204#issuecomment-1104298534

mcguirepr89, 2022-04-24:
hahahahaha me too, it was hilarious :laughing: it's nature disney

mcguirepr89, 2022-04-24:
Let me see what I can get working with ```rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mp4```

mcguirepr89, 2022-04-24:
You can see my new rtsp stream reading services running while I'm troubleshooting if you're interested https://virginia.birdnetpi.com/log -- so far things are working -- nice jon @ehpersonal38

danmed, 2022-04-24:
@mcguirepr89 whilst you're active here.. what's the email i should send to to get a subdomain? the one in the gui is just giving me @gmail.com

mcguirepr89, 2022-04-24:
my githubhandle @ gmail.com

mcguirepr89, 2022-04-25:
Ok it is working for me on my end -- I probably shouldn't have used my main system to test this, since now it's full of this stream's data!!!

All that is left now is a template for an `rtsp_stream.service` 
```
[Unit]
Description=BirdNET RTSP Stream Fetcher
[Service]
Restart=always
Type=simple
RestartSec=3
User=birder
ExecStart=/usr/local/bin/rtsp_stream.sh
[Install]
WantedBy=multi-user.target
```

The new `rtsp_stream.sh` script needs to be in the repo and properly symlinked to `/usr/local/bin`
`rtsp_stream.sh`
```
#!/usr/bin/env bash
source /etc/birdnet/birdnet.conf

while true;do
  ffmpeg -i  $RTSP_STREAM -t 15 -vn -acodec pcm_s16le -ac 2 -ar 44100 file:${RECS_DIR}/$(date "+%F")-birdnet-$(date "+%H:%M:%S").wav
done
```

And add the new `RTSP_STREAM` variable to the `birdnet.conf-defaults`, `install_config.sh`, and make it webGUI addressable and we'll have outselves a new interface. 

I had to adjust `extract_new_birdsounds.sh` to check in the correct places for audio and then put the extractions where they belong for the PHP to know where to find it.

Pretty cool

mcguirepr89, 2022-04-25:
I'll update here when this is further along -- I'm switching back to my microphone now!

danmed, 2022-04-25:
So i've not switched to the new method above but i've woken up and there's 230 odd birds detected which is great., but i', unable to download any of the mp3's still :(

mcguirepr89, 2022-04-25:
Hi, @danmed -- I had not yet pushed the changes to make this feature work for folks --

I've though about this a bit more, and the `birdnet_recording.sh` services will check for the `RTSP_STREAM` variable in `birdnet.conf` and will be set in "Advanced Settings" in the webGUI. There is also a small change to  `extract_new_birdsounds.sh` that allows the extractions to get properly organized from the base of the `RECS_DIR`

I will let you know when an update will suffice for you to test this feature.

In the meantime, you can make the adjustment to `extract_new_birdsounds.sh` manually ---
in `extract_new_birdsounds.sh`, change your "Set Variables" section to look like this:
```
ANALYZED=${RECS_DIR}/*/*Analyzed
#SCAN_DIRS are all directories marked "Analyzed"
SCAN_DIRS=($(find $HOME -type d -name '*Analyzed' 2>/dev/null | sort ))
```
and then change line 41 from this:
```
```
to this
```
echo "$(basename ${i})" >> ${TMPFILE}
```
Then things will work for you. 

Understand, however, that making customizations to your system can make updates difficult (perhaps you will have to manually reconcile divergent branches and handle merging manually)

Again, I will let you know when this is a true feature and is past the proof-of-concept stage

ehpersonal38, 2022-04-25:
> I've though about this a bit more, and the `birdnet_recording.sh` services will check for the `RTSP_STREAM` variable in `birdnet.conf` 

So no separate service for RTSP stream? I guess that makes my job a bit easier

ehpersonal38, 2022-04-25:
> In the meantime, you can make the adjustment to `extract_new_birdsounds.sh` manually --- in `extract_new_birdsounds.sh`, change your "Set Variables" section to look like this:
> 
> ```
> ANALYZED=${RECS_DIR}/*/*Analyzed
> #SCAN_DIRS are all directories marked "Analyzed"
> SCAN_DIRS=($(find $HOME -type d -name '*Analyzed' 2>/dev/null | sort ))
> ```
> 
> and then change line 41 from this:
> 
> ```
> echo "${i}" | cut -d'/' -f7 >> ${TMPFILE}
> ```
> 
> to this
> 
> ```
> echo "$(basename ${i})" >> ${TMPFILE}
> ```
> 
> Then things will work for you.
> 
> Understand, however, that making customizations to your system can make updates difficult (perhaps you will have to manually reconcile divergent branches and handle merging manually)
> 
> Again, I will let you know when this is a true feature and is past the proof-of-concept stage

I've tried making these adjustments, but the mp3s still aren't being created, here's the log. Any ideas?:
[image]

mcguirepr89, 2022-04-25:
I've got the new service put together in the `rtsp` branch -- you can `git pull && git checkout rtsp` then "Tools" > "System Controls" > "Update" 

Then "Tools" > "Settings" > "Advanced Settings" will let you input an RTSP stream URL and then when you click "Update Settings," the system will switch over to using the RTSP stream.

Let me know @ehpersonal38 if you have trouble with that

danmed, 2022-04-25:
I've switched to the rtsp branch and i *think* it's working now.. but watching the logs it's almost like it's either going back and creating all the MP3's that weren't created previously... is that possible?

Otherwise something is WAAAYY off ..

Apr 25 14:56:30 BIRDNET-PI env[446]: NEWFILE=Eurasian_Blackbird-72-2022-04-25-birdnet-05:18:08.mp3
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]: [rtsp @ 0x558baa4020] max delay reached. need to consume packet
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]: [rtsp @ 0x558baa4020] RTP: missed 44 packets
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]: [rtsp @ 0x558baa4020] max delay reached. need to consume packet
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]: [rtsp @ 0x558baa4020] RTP: missed 5 packets
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]: Input #0, rtsp, from 'rtsp://admin:312217@192.168.2.125/h264Preview_01_main':
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]:   Metadata:
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]:     title           : Session streamed by "preview"
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]:     comment         : h264Preview_01_main
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]:   Duration: N/A, start: 0.000000, bitrate: N/A
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]:     Stream #0:0: Video: h264 (High), yuv420p(progressive), 2560x1440, 30 tbr, 90k tbn, 180k tbc
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]:     Stream #0:1: Audio: aac (LC), 16000 Hz, mono, fltp
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]: Stream mapping:
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]:   Stream #0:1 -> #0:0 (aac (native) -> pcm_s16le (native))
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]: Press [q] to stop, [?] for help
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]: Output #0, wav, to 'file:/home/pi/BirdSongs/2022-04-25-birdnet-14:56:28.wav':
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]:   Metadata:
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]:     INAM            : Session streamed by "preview"
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]:     ICMT            : h264Preview_01_main
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]:     ISFT            : Lavf58.45.100
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]:     Stream #0:0: Audio: pcm_s16le ([1][0][0][0] / 0x0001), 48000 Hz, stereo, s16, 1536 kb/s
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]:     Metadata:
Apr 25 14:56:31 BIRDNET-PI birdnet_recording.sh[20649]:       encoder         : Lavc58.91.100 pcm_s16le

mcguirepr89, 2022-04-25:
It found all the analyzed files and is extracting them now -- so what you are seeing is correct, it is post-processing

danmed, 2022-04-25:
> It found all the analyzed files and is extracting them now -- so what you are seeing is correct, it is post-processing

Nice!.. this will please the wife.. 

I really appreciate the work you guys are putting in to this! 

Do you think the ability to run this in an x86 VM or LXC will ever be an option btw?

mcguirepr89, 2022-04-25:
@danmed please be mindful when posting log output as it can often contain sensitive information

mcguirepr89, 2022-04-25:
You're correct that the spectrogram is just showing the frequency limitations of the original stream -- it's working for sure :+1:

ehpersonal38, 2022-04-25:
### **For anyone wanting to test RTSP now:**
1. Install BirdNet-PI as normal
2. Type `git pull && git checkout rtsp`
3. In the web GUI, navigate to "Tools" > "System Controls" > "Update", and confirm
4. Input your RTSP stream URL in "Tools" > "Settings" > "Advanced Settings", click update at the bottom, and the system will switch over to using the RTSP stream.

MrksHfmn, 2022-05-06:
is it somehow possible to run BirdnetPi in a VM? I would love to run it on x86 or in a VM ...

mcguirepr89, 2022-05-06:
Hi, @MrksHfmn 
>is it somehow possible to run BirdnetPi in a VM? I would love to run it on x86 or in a VM ...

It should be possible. Here are some things to know:
1. The installation user needs to have a `NOPASSWD` sudo policy and of course be a member of the `sudo` group
1. You'll want to replace the `tflite_runtime.wheel` for `aarch64` to one for `x86_64`, or you may be able to install the full `tensorflow` instead (make these changes in `requirements.txt`).
1. The OS needs to be Debian based (using `systemd` -- indeed it may be best to use the RaspiOS for x86 to ensure similar default packaging, though there are users on vanilla Debian who do not report any issues).
1. For a VM, you'll want to be sure to expose ports 80 and 443. ([There is a fork of the repo which may interest you](https://github.com/mcguirepr89/BirdNET-Pi/issues/211) -- it has dockerized the system.)

Since this issue is in regards to RTSP streams and since that branch was merged into `main`, I was going to close this issue.

I think is would great to have an ongoing discussion for VMs and x86 -- please start a discussion for that topic and I will do my best to funnel folks there who may have ways of fulfilling this port request! Also, I'm happy to help you with troubleshooting were you to take on porting the system to x86 yourself, just let me know how I can help :+1:

closed 2022-05-06 by mcguirepr89