mcguirepr89/BirdNET-Pi #209: Overview sometimes doesn't show Most Recent Detection

mcguirepr89, 2022-04-19:
Not sure why yet -- must have something to do with checking the "`previous_detection_identifier`" I'll be taking a look at this more closely and will update here

ehpersonal38, 2022-04-19:
> That probably is the most elegant way to handle things right now -- this could possibly be related to the still-occurring huge spike in traffic (+3,000 views since this morning). It definitely works once there is a new detection, but sometimes doesn't load the most recent if it is older. This isn't a super high priority and is only happening on the installation that is getting the high traffic -- the RPi0W2 installation (getting less traffic) is not seeing the same issue.
> 
> Don't spend any time on this, as I do think this is probably the best method for now and the solution to the database locking is probably going to be in `server.py` 👍

Oh I see, I didn't realize this only occured on the high traffic system. 

Also look at the stars! Over 250!!! 😮

mcguirepr89, 2022-04-19:
>Oh I see, I didn't realize this only occured on the high traffic system. (I can see your installation is running really slow too, probably due to all the images it's serving?)

Yes and it's my test system, so I'm in there a lot doing stuff (I need a dedicated test system that is not also my public installation!!!).

Here are the changes I'm testing now that I'll probably pull in later after I've confirmed they're doing what I need:
line 360 in `server.py`
```
                                for attempt_number in range(3):
                                    try:
                                        con = sqlite3.connect(userDir + '/BirdNET-Pi/scripts/birds.db')
                                        cur = con.cursor()
                                        cur.execute("INSERT INTO detections VALUES (?, ?, ?, ?, ?, ?, ?, ?, ?, ?, ?, ?)", (Date, Time, Sci_Name, Com_Name, str(score), Lat, Lon, Cutoff, Week, Sens, Overlap, File_Name))

                                        con.commit()
                                        con.close()
                                        break
                                    except:
                                        print("Database busy")
                                        time.sleep(2)
```

So that it will try to write to the database 3 times before moving on

>Also look at the stars! Over 250!!! open_mouth

I KNOW!!! So crazy!!!! The project has changed scope and scale in just the last 24 hours :exploding_head:

ehpersonal38, 2022-04-19:
> Yes and it's my test system, so I'm in there a lot doing stuff (I need a dedicated test system that is not also my public installation!!!).
>
> Here are the changes I'm testing now that I'll probably [...]

Nice! Hopefully that improves things. There's also something interesting regarding SQLite3 I found, may be of use, may not be:

https://www.sqlite.org/cgi/src/doc/wal2/doc/wal2.md
`In legacy wal mode, when a writer writes data to the database, it doesn't modify the database file directly. Instead, it appends new data to the "<database>-wal" file. Readers read data from both the original database file and the "<database>-wal" file. At some point, data is copied from the "<database>-wal" file into the database file, after which the wal file can be deleted or overwritten. Copying data from the wal file into the database file is called a "checkpoint", and may be done explictly (either by "PRAGMA wal_checkpoint" or sqlite3_wal_checkpoint_v2()), or automatically (by configuring "PRAGMA wal_autocheckpoint" - this is the default).`

So theoretically the database will never get tied up when writing because it's all basically reading and writing a text file instead?

mcguirepr89, 2022-04-19:
Thanks, I'll definitely check that out! SQLite is new to me -- a file versus a server has introduced some novelty for how/when to access the database, but lightweight is the way to go, so I know we'll figure it out!

I'm closing this since I know that the issue is in traffic and not in the code

closed 2022-04-19 by mcguirepr89

mcguirepr89, 2022-04-19:
Looks like I may also start using either the [`sqlite3_busy_timeout`](https://www.sqlite.org/c3ref/busy_timeout.html)
or the timeout parameter in the 
[`sqlite3.connect(database, [**timeout=FLOAT**])`](https://docs.python.org/3/library/sqlite3.html)