mcguirepr89/BirdNET-Pi #211: Docker support

olipayne, 2022-04-19:
I'm making this issue and working on it myself, just in case anyone else here is interested in following the progress :)

mcguirepr89, 2022-04-19:
What synchronicity!! I[ just posted this discussion for needs](https://github.com/mcguirepr89/BirdNET-Pi/discussions/212), and at the top is . . . 

I want to help however I can!!! I know that this thing is not put together the way it would were I to be anything other than a hobbyist when it comes to developing a web application, so I will likely be of most value just by fixing things I've done so that they're not bad and to make things as easy for you as I can -- please just let me know how to help and I will!

olipayne, 2022-04-19:
At this point the simplest thing to do is emulate an entire Raspberry Pi image and then throw that into a container registry, but it's quite bloaty (so far my build is taking more than 20 mins 😿)

If you're interested and you want to rewrite the entire web-end in Laravel, and then just querying the existing 'server.py' service internally, you'd get a lot of support from me ;)

olipayne, 2022-04-19:
So far the initial issues are as you described in #104, the pre-built TFLite binaries are only for aarch64. This part of the stack I'm not too familiar with.

How feasible is it to get similar binaries for x86_64 architecture?

olipayne, 2022-04-19:
I've managed to get a Docker image build running, https://github.com/olipayne/BirdNET-Pi-Docker/runs/6085887119 will leave that for tonight and get back to it tomorrow 🥳

mcguirepr89, 2022-04-19:
>How feasible is it to get similar binaries for x86_64 architecture? If so we can dump a lot of the 'build' work to GitHub Action

Pretty easy, but I think maybe the easiest thing to do would be to try adjusting the `requirements.txt` file's line for the tflite aarch64 binary to just say `tflite_runtime`, which ought to grab those for the platform you're building on -- I will try to track down a pre-build binary for x86 and will let you know if I can come up with it. If you get to setting up a build action before I get the chance, let me know how it goes! :+1:

lpasselin, 2022-04-26:
Hey this is just a comment but imo, the cleanest way to "dockerize" this would be to refactor the whole thing into docker-compose and have each service (db, ui, server, analysis, recording, etc) running in a different container.

There are lots of things happening during BirdNET-Pi installation (new files in /usr/local/bin, scripts to add custom shebangs, scripts writing templates, etc). These hacks complicate the maintenance and updates. All of this would be much simpler when using docker images.

This is not an easy task if you've never worked with containers.

I am not saying dockerizing is the way to go. But if you want to do it properly, I would suggest to refactor at the same time instead of installing into docker image based on raspberry-pi image.

By the way, BalenaOS might be worth looking into to deploy docker containers on a raspberry-pi.

maxogden, 2022-04-29:
@olipayne I found an x86_64 whl online https://github.com/maxogden/BirdNET-Pi/blob/main/tflite_runtime-2.6.0-cp39-cp39-linux_x86_64.whl but I am not sure if it works... I tried building your Dockerfile locally and I think because `balenalib/raspberrypi4-64-debian:bullseye` is not amd64 it won't let me build on an amd64 machine. How were you planning on running raspbian on amd64?

xginn8, 2022-05-18:
@lpasselin @olipayne former balena employee here happy to help with the containerization push! I actually tried to deploy this today before realizing it was more complicated than I originally thought. It can certainly be done without much balena lockin (generic docker-compose and service definitions), but the open fleets stuff makes it stupid simple to deploy as well.

Again I'm no longer affiliated with balena, just happy to help to make this easier to get up and running!

olipayne, 2022-05-27:
Since this entire project is built on the premise that it's running on a Raspberry Pi of some sort, it might be a bigger job than initially expected. I'm looking into seeing what can be chopped up so that in theory, this entire project could run in separate places (e.g., recording on a Pi somewhere, sending the files to a more powerful machine for analysis and webpage hosting)

olipayne, 2022-06-07:
Quick update, I'm separating out the Tensorflow stuff first, it seems like the easiest route for now is to build a dedicated image per CPU arch, since there are a lot of hardware level optimisations you miss out on if you build an all-in-one package.

I've got the `server.py` service running on a local machine here, decided to open it to the world for testing: https://bird-analyze.oli.xyz/analyze

Feel free to post to it as much as you like, it'll just slow down but it wont crash anything else.

Naturally, this comes with a privacy warning, you'd be sending raw audio to a third party (me). Once I've got a bit more running, I'll stitch together a docker-compose file so you can host this yourself.

aderusha, 2022-06-21:
Getting this running on a container platform + RTSP support would mean that people with home NAS devices etc could run the solution with existing security cameras without having to try and source an RPi for sale.  Container support brings a whole new set of deployment options to the table and I'm pretty stoked that it's being worked out!

supajason, 2022-06-22:
I'm going to give this an attempt as it will give me the opportunity to learn about docker.
Not sure how it's going to work yet but I think splitting the analyse "server" into one docker and everything else on another for now will be a great start.

olipayne, 2022-06-25:
@supajason is on the right track, using https://github.com/kahst/BirdNET-Analyzer/blob/main/Dockerfile (slightly modified to run server.py though) and then forking everything else over to a normal 'web app' seems like the right way forward.

mgottholsen, 2022-06-25:
> Quick update, I'm separating out the Tensorflow stuff first, it seems like the easiest route for now is to build a dedicated image per CPU arch, since there are a lot of hardware level optimisations you miss out on if you build an all-in-one docker image.
> 
> I've got the `server.py` service running on a local machine here, decided to open it to the world for testing: https://bird-analyze.oli.xyz/analyze
> 
> Feel free to post to it as much as you like, it'll just slow down but it wont crash anything else.
> 
> Naturally, this comes with a privacy warning, you'd be sending raw audio to a third party (me). Once I've got a bit more running, I'll stitch together a docker-compose file so you can host this yourself.

If this could be modularized further, I could see an end result that you could plug in many different models, with applications outside of bird identification. I've always wanted to be able to identify sounds & trilaterate them in 3D space using an array of microphones. Similar to gunshot detection that cities use.

mcguirepr89, 2022-06-25:
Hi @mgottholsen! I just wanted to [show you this](https://github.com/mcguirepr89/BirdNET-Pi-assets/blob/main/G23_Report_ModelBasedSysEngineering_FarmMarkBirdDetector_V1__Copy_.pdf) since you might be interested in their findings.

My best,
Patrick

mgottholsen, 2022-06-25:
@mcguirepr89 This is really awesome stuff! IT conceptually overlaps with some other stuff I've been looking at Rhasspy voice assistant satellite devices that run on similarly specc'ed hardware. The multilateration is a killer feature as well. I noticed you helped them with their setup, were there large differences in their implementation that would make sense here?

mcguirepr89, 2022-06-25:
@mgottholsen My contributions to that DTU project were not around any of the cool engineering, unfortunately -- I just made myself available to them, helped get the installation working, answer questions, etc., and was able to make little adjustments here and there to make things easier for their needs.

The system can be noded without too much hassle so long as you know a little bit about networking. Also, since the system can run on the $15 Raspberry Pi 0W2, a triangle of nodes could be ~$45 + $cost of (3)three USB mics, and (3)three USB-A to micro-USB adapters.

Do you have any recommendations on some 101, introductory type sources for getting started in being able to gather the info needed to start tinkering with this?

AngellusMortis, 2022-06-28:
+1 for breaking out the different services into different containers. The way I would refactor this (I am not a PHP person so probably not much help in actually doing it):

1. Get everything running in a single container (sounds like already got that kind of working)
2. Start identifying the core "pieces" of the project that you can break out. My initial guess would be...
    * Audio capture
    * TensorFlow / audio processing
    * Web application (PHP)
    * HTTP server
    * Database maintenance/misc background cleanup/tasks
    * Everything else
3. Start splitting out the different pieces from above into their own containers with docker compose. Leave out everything under "everything else" to start with. The prototype/MVP should not have a Web terminal, file manager, FTP server, phpSysInfo, etc. You can slow add those back in later
4. Update the install scripts to basically install docker -> pull compose setup -> configure it -> start
5. Add back in optional components from 3.
6. Add new things that make sense as you scale it out (Postgres/MySQL support?, audio from multiple RSTP sources, multi-node support within the Web app, running multiple Web apps, etc.)

supajason, 2022-06-29:
Audio capture is a bit of sticking point. I don't know enough about docker for USB/device recording.
For not I might focus just on using streams then try and add it later.
Next up will be understanding how the front end works and see about running it on its own container.

AngellusMortis, 2022-06-29:
I think most people using Docker/containers will likely be using RTSP sources. For the RPI instance, that is likely something you would want to keep on the host machine. The _only_ service on the host machine to record off audio to a folder that docker can then read from. 

You can try to mount the audio device into the container, but that is likely more trouble than it is worth.

supajason, 2022-07-05:
So I have the analyse and stream_capture containers running.
Trying to plug these into this repo isn't going to be possible with the amount of time I have and knowledge.
What I might do is re-create it in another repo, that way should be easier*

mcguirepr89, 2022-07-05:
@supajason that sounds great! If on another repo, please let us know how to find it and I will do my best to spotlight the repo for folks interested in what you have working. :+1:

supajason, 2022-07-25:
Has anyone used the docker version? I've got it up and running but a 10 second clip is taking 12 seconds to analyse which isn't going to work.
I raised the [issue](https://github.com/kahst/BirdNET-Analyzer/issues/50) on the main BirdNET repo but wanted to see if anyone here had seen similar performance issues with the docker version?

cam-el, 2022-07-25:
> Has anyone used the docker version? I've got it up and running but a 10 second clip is taking 12 seconds to analyse which isn't going to work. I raised the [...]

I am going to give it a go over the next weekend if my Pi0W2 stabilises. I am trying it out on a Rockchip A55 machine but have poor detection results on that device.

matthew73210, 2022-08-07:
> 

Could you point me to a guide for your docker idea, I'll give it a whirl on my nuc / home assistant?

supajason, 2022-08-09:
Sorry been busy. I'll tidy what I have and pop the link in. Conscious that this isn't strictly on topic with this repo. (But appreciation it links in)

matthew73210, 2022-08-11:
> Sorry been busy. I'll tidy what I have and pop the link in. Conscious that this isn't strictly on topic with this repo. (But appreciation it links in)

Theres interest on the home assistant side, if this could become an integration. Could really up the game.

matthew73210, 2022-08-23:
FYI just tried installing in a clean simple debian bulleye arm64 docker container. Was going fin till it hit the systemD part which, I gather, is blasphemy on docker. Might see If I run the services under supervisord but I don't really know where to start. 

```
Installation completed successfully
System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: Host is down
Failed to talk to init daemon.
````

What are you using or are you going the docker compose route?

JayRama, 2022-08-24:
How about a .sh entrypoint script for the Dockerfile which just runs the commands from the service init scripts?

matthew73210, 2022-08-25:
> How about a .sh entrypoint script for the Dockerfile which just runs the commands from the service init scripts?

That's what I did, systemD isn't really supported on docker so it can't start the services. With supervisord it can though. Or go the proper route and have a container per service and use docker compose.

mcguirepr89, 2022-09-01:
>Or there is the option of using a systemD ready container, I tried that and the installation went fine, but for some reason the services wouldn't start.

One of RaspiOS's non-Debian-isms is that the default user (`UID 1000`) is automatically added to the pre-installed `sudo` group. Debian doesn't come with sudo installed by default -- furthermore, it would need the `UID 1000` user to be added to the `sudo` group _and_ given the `sudoers` NOPASS rule:

`/etc/sudoers.d/010_pi-nopasswd`:
```
pi ALL=(ALL) NOPASSWD: ALL
```
Naturally, the `pi` user will actually be whichever `UID 1000` user you create and add to the `sudo` group.

:thinking: not sure if that is the issue you're encountering since the systemd services are run as the `UID 1000` user.

Let me know if having those sorts of permissions fixes the systemd shortcomings. (In theory, you could probably move the "`ExecStart`" portions of the systemd service files into cron jobs??? I haven't tried that, but it would likely work -- you would just loose your journal logging until you installed an MTA to get cron emails (if you want logs). This would also break your "View Log" page, since that reads from the journal.

My very best regards,
Patrick

JayRama, 2022-09-01:
> > How about a .sh entrypoint script for the Dockerfile which just runs the commands from the service init scripts?
> 
> The issue is with systemD, on the classic debian container systemD doesn't work, you need to use supervisorD. Or separate each service into a separate container and use docker compose. Which is the 'right' way of doing it. Or there is the option of using a systemD ready container, I tried that and the installation went fine, but for some reason the services wouldn't start.

putting the commands from the execstart lines in the systemd files into a .sh with an & at the end of each command (additionally we can output to a /var/log like this too so docker logs works), then setting the .sh as the entrypoint for the Dockerfile (hence starting everything when the container starts). this doesnt need systemd or initd.

I agree in theory that separating the services out into their own containers appears to be the right way, however this is pretty close to working in its current form and could be a good starting point - would bring in some new contributors and sets of eyes  to tidy it up.

matthew73210, 2022-09-03:
> putting the commands from the execstart lines in the systemd files into a .sh with an & at the end of each command (additionally we can output to a /var/log like this too so docker logs works), then [...]

If I understand correctly we are going to move all the services away from systemd and into a script that starts when the container is run? All the services will need to be installed beforehand so it doesn't really change the installation process, we just need to create a new entrypoint after the container has finished doing its business the first time. 
I may be a bit rusty on my docker terms. Create an image after birdnet has finished installing and run as container with said entrypoint.

matthew73210, 2022-10-26:
I'll gove this a go next week, on holiday then. Cheers

milch, 2022-11-10:
Hey @matthew73210, did you end up trying this out? If yes, did you get anywhere?

closed 2022-12-13 by mcguirepr89

fightforlife, 2023-05-18:
Could this be a replacement for a docker variant? https://github.com/mmcc-xx/BirdCAGE

srd424, 2023-07-10:
Did everyone settle on a location to discuss docker and other container approaches? I haven't really found anywhere active that might be relevant except #251. I run my own homebrew container system and have managed to get BirdNET-Pi running in it OK, it should be fairly easy to translate the build script into a Dockerfile.

evanbills, 2023-07-10:
Birdcage looks promising:  https://github.com/mmcc-xx/BirdCAGE

frostworx, 2023-07-16:
Thanks for the heads-up, @evanbills!

srd424, 2023-07-17:
I think BirdCAGE includes a modified `server.py` to allow BirdNET-Pi to talk to it .. that's probably something I could add as an option relatively easily to my docker containers if anyone's interested?

matthew73210, 2023-07-17:
It does, it's because the docker container was restarting too often. 

> I think BirdCAGE includes a modified `server.py` to allow BirdNET-Pi to talk to it .. that's probably something I could add as an option relatively easily to my docker containers if anyone's interested?

alexbelgium, 2024-04-27:
Hi, I've made a working aarch64 and amd64 docker container based on https://github.com/Nachtzuster/BirdNET-Pi from @Nachtzuster.

It is a homeassistant addon, that can be installed using my repo here : https://github.com/alexbelgium/hassio-addons

If you don't use homeassistant, the containers can be assessed directly from the packages : https://github.com/alexbelgium/hassio-addons/pkgs/container/birdnet-pi-aarch64 and https://github.com/alexbelgium/hassio-addons/pkgs/container/birdnet-pi-amd64

frostworx, 2024-04-28:
> Hi, I've made a working aarch64 and amd64 docker container based on https://github.com/Nachtzuster/BirdNET-Pi from @Nachtzuster.

Hey @alexbelgium! Wow, thank you very much! Your hassio-addons collection is a must-have now even more :)

gotschi, 2025-02-11:
> https://github.com/alexbelgium/hassio-addons/pkgs/container/birdnet-pi-amd64

Thank you, I think I found a working compose config with your image. For anyone else:

```yaml
services:
  birdnet-pi:
    container_name: birdnet-pi
    image: ghcr.io/alexbelgium/birdnet-pi-amd64
    restart: unless-stopped
    ports:
      - "8004:8081"
    environment:
      - TZ=Europe/Vienna
      - BIRDSONGS_FOLDER=/config/BirdSongs
      - LIVESTREAM_BOOT_ENABLED=false
      - MQTT_DISABLED=true
      - SSL=false
      - CERTFILE=fullchain.pem
      - KEYFILE=privkey.pem
    volumes:
      - ./config:/config
      - ./media:/media
      - ./share:/share
      - ./ssl:/ssl
      - /dev/shm:/dev/shm  # Shared memory
      - ./nginx/:/etc/nginx/servers
    tmpfs:
      - /tmp
```

alexbelgium, 2025-02-11:
> > https://github.com/alexbelgium/hassio-addons/pkgs/container/birdnet-pi-amd64
> 
> Thank you, I think I found a working compose config with your image. For anyone else:
> 
> services:
>   birdnet-pi:
>     container_name: birdnet-pi
>     image: ghcr.io/alexbelgium/birdnet-pi-amd64
>     restart: unless-stopped
>     ports:
>       - "8004:8081"
>     environment:
>       - TZ=Europe/Vienna
>       - BIRDSONGS_FOLDER=/config/BirdSongs
>       - LIVESTREAM_BOOT_ENABLED=false
>       - MQTT_DISABLED=true
>       - SSL=false
>       - CERTFILE=fullchain.pem
>       - KEYFILE=privkey.pem
>     volumes:
>       - ./config:/config
>       - ./media:/media
>       - ./share:/share
>       - ./ssl:/ssl
>       - /dev/shm:/dev/shm  # Shared memory
>       - ./nginx/:/etc/nginx/servers
>     tmpfs:
>       - /tmp

Wow that's grea!!! and you did not run into any issues while starting due to bashio not founding the config.json ?
If you paste a startup log I can adapt the script to avoid this. Also, tmpfs is likely optional (should be)

gotschi, 2025-02-11:
> Wow that's grea!!! and you did not run into any issues while starting due to bashio not founding the config.json ? If you paste a startup log I can adapt the script to avoid this. Also, tmpfs is likely optional (should be)

what is needed is ingress.conf for nginx with set %%interface%%:%%port%%,
e.g.
```
   server {
    listen 0.0.0.0:8005 default_server;
```

as you can see I mounted /etc/nginx/servers containing my ingress.conf as I think this is set by HA supervisor and did not see any chance doing so using env vars?

other than that it should work fine, I use a rtsp stream so the device permissions are missing in the compose.yml for me. anyone using hardware input may need to add these also.

here's the log

```
birdnet-pi  | Starting...
birdnet-pi  | /etc/cont-init.d/00-banner.sh: executing
birdnet-pi  | curl: (6) Could not resolve host: supervisor
birdnet-pi  | [09:41:03] ERROR: Something went wrong contacting the API
birdnet-pi  | curl: (6) Could not resolve host: supervisor
birdnet-pi  | [09:41:03] ERROR: Something went wrong contacting the API
birdnet-pi  | /etc/cont-init.d/00-global_var.sh: executing
birdnet-pi  | jq: error: Could not open file /data/options.json: No such file or directory
birdnet-pi  | Timezone set from  to Europe/Vienna
birdnet-pi  | /etc/cont-init.d/00-local_mounts.sh: executing
birdnet-pi  | /etc/cont-init.d/00-smb_mounts.sh: executing
birdnet-pi  | /etc/cont-init.d/01-config_yaml.sh: executing
birdnet-pi  | Setting permissions for the config.yaml directory
birdnet-pi  |
birdnet-pi  | Load environment variables from /config/config.yaml if existing
birdnet-pi  | If accessing the file with filebrowser it should be mapped to /addon_configs/a3dff44168cd/config.yaml
birdnet-pi  | ---------------------------------------------------------
birdnet-pi  | Wiki here on how to use : github.com/alexbelgium/hassio-addons/wiki/Add‐ons-feature-:-add-env-variables
birdnet-pi  |
birdnet-pi  | ... no env variables found, exiting
birdnet-pi  | /etc/cont-init.d/01-custom_script.sh: executing
birdnet-pi  | Execute /addon_configs/a3dff44168cd/a3dff44168cd.sh if existing
birdnet-pi  | Wiki here : github.com/alexbelgium/hassio-addons/wiki/Add-ons-feature-:-customisation
birdnet-pi  | /etc/cont-init.d/01-structure.sh: executing
birdnet-pi  | [09:41:03] INFO: Ensuring the file structure is correct:
birdnet-pi  | ... creating default files
birdnet-pi  | ... creating default folders; it is highly recommended to store these on an SSD
birdnet-pi  | ... tmpfs detected, using it for StreamData and Processed to reduce disk wear
birdnet-pi  | ... setting permissions for user pi
birdnet-pi  | ... creating symlinks for configuration files
birdnet-pi  | ... creating symlink for /config/BirdSongs/By_Date
birdnet-pi  | ... creating symlink for /config/BirdSongs/Charts
birdnet-pi  | ... checking and setting permissions
birdnet-pi  | ... setting up Matplotlabdir
birdnet-pi  | /etc/cont-init.d/02-restorestreamdata.sh: executing
birdnet-pi  | /etc/cont-init.d/31-checks.sh: executing
birdnet-pi  | [09:41:03] INFO: Checking your birdnet.conf file integrity
birdnet-pi  | curl: (6) Could not resolve host: supervisor
birdnet-pi  | [09:41:03] ERROR: Something went wrong contacting the API
birdnet-pi  | [09:41:03] INFO: Performing potential updates
birdnet-pi  | BIRDNET_USER: pi
birdnet-pi  | cp: '/config/birdnet.conf' and '/home/pi/BirdNET-Pi/birdnet.conf' are the same file
birdnet-pi  | ln: '/home/pi/BirdNET-Pi/birdnet.conf' and '/config/birdnet.conf' are the same file
birdnet-pi  | yes
birdnet-pi  | /etc/cont-init.d/33-mqtt.sh: executing
birdnet-pi  | /etc/cont-init.d/71-newfeatures.sh: executing
birdnet-pi  | [09:41:05] INFO: Adding optional features
birdnet-pi  | /etc/cont-init.d/81-modifications.sh: executing
birdnet-pi  | [09:41:05] INFO: Adapting webui
birdnet-pi  | ... removing System Controls from webui as should be used from HA
birdnet-pi  | ... removing Ram drive from webui as it is handled from HA
birdnet-pi  | ... updating services to start as user pi
birdnet-pi  | ... redirecting services logs to container logs
birdnet-pi  | ... disabling preselecting options in include and exclude lists
[... 133 lines cut ...]
birdnet-pi  | [birdnet_analysis][INFO] Analyzing /home/pi/BirdSongs/StreamData/2025-02-11-birdnet-RTSP_1-09:41:35.wav
birdnet-pi  | [server][INFO] READING AUDIO DATA...
birdnet-pi  | [rtsp @ 0x654f40322f40] method SETUP failed: 461 Unsupported transport
birdnet-pi  | 2025/02/11 09:41:53 127.0.0.1:59054 200 GET /log/
birdnet-pi  | 2025/02/11 09:41:53 127.0.0.1:59054 200 GET /log/css/index.css
birdnet-pi  | 2025/02/11 09:41:53 127.0.0.1:59054 200 GET /log/css/xterm_customize.css
birdnet-pi  | 2025/02/11 09:41:53 127.0.0.1:59070 200 GET /log/auth_token.js
birdnet-pi  | 2025/02/11 09:41:53 127.0.0.1:59054 200 GET /log/css/xterm.css
birdnet-pi  | 2025/02/11 09:41:53 127.0.0.1:59070 200 GET /log/config.js
birdnet-pi  | 2025/02/11 09:41:53 127.0.0.1:59054 200 GET /log/js/gotty.js
birdnet-pi  | 2025/02/11 09:41:53 New client connected: 127.0.0.1:59102, connections: 1/0
birdnet-pi  | 2025/02/11 09:42:16 Connection closed by client: 127.0.0.1:59102, connections: 0/0
birdnet-pi  | [birdnet_analysis][INFO] Analyzing /home/pi/BirdSongs/StreamData/2025-02-11-birdnet-RTSP_1-09:42:05.wav
birdnet-pi  | [server][INFO] READING AUDIO DATA...
birdnet-pi  | [server][INFO] READING DONE! READ 5 CHUNKS.
birdnet-pi  | [server][INFO] ANALYZING AUDIO...
[... cut ...]
birdnet-pi  | [server][INFO] 0.0;3.0-('Empidonax flaviventris_Yellow-bellied Flycatcher', 0.07126828)
birdnet-pi  | [server][INFO] 3.0;6.0-('Caprimulgus europaeus_Eurasian Nightjar', 0.074125595)
birdnet-pi  | [server][INFO] 6.0;9.0-('Caprimulgus europaeus_Eurasian Nightjar', 0.08748733)
birdnet-pi  | [server][INFO] 9.0;12.0-('Chordeiles minor_Common Nighthawk', 0.23678511)
birdnet-pi  | [server][INFO] 12.0;15.0-("Empidonax hammondii_Hammond's Flycatcher", 0.40461805)
birdnet-pi  | [birdnet_analysis][INFO] Analyzing /home/pi/BirdSongs/StreamData/2025-02-11-birdnet-RTSP_1-09:42:21.wav
```

alexbelgium, 2025-02-11:
Thanks, I can actually skip those elements if started without HA. I'll look to do that

Bongo2, 2025-02-12:
Birdnet in the Docker container. I waited for that. Great.

@gotschi
I started your compose with the alexbelgium container on my Synology with portainer. It runs.
Now it would be good if there were a container with the current feature.

alexbelgium, 2025-02-12:
> Birdnet in the Docker container. I waited for that. Great.
> 
> [@gotschi](https://github.com/gotschi) I started your compose with the alexbelgium container on my Synology with portainer. It runs. Now it would be good if there were a container with the current feature.

Which current feature are you referring to? My container is based on Nachtzuster's repo which is the most up to date with some additional scripts and additions in my repo that are awaiting to be merged. I'm currently updating to avoid the curl issues due to the lack of homeassistant. MQTT also works provided that the env variables are set ; once the code is updated I'll add the list of env variables supported (expanding on @gotschi 's list - globally the mqtt and TZ ones)

gotschi, 2025-02-13:
the new image doesn't seem to start without setting `pi_password` and then it doesn't start caddy for some reason, so no webui on 8081

alexbelgium, 2025-02-15:
Should be good now :)

Bongo2, 2025-02-26:
@gotschi @alexbelgium 

I am enthusiastic about the Docker container.

[image]

I use Synology with portainer. My docker compose looks like this:
```
services:
  birdnet-pi:
    container_name: birdnet-pi
    image: ghcr.io/alexbelgium/birdnet-pi-amd64
    restart: unless-stopped
    ports:
      - "8004:8081"
    environment:
      - TZ=Europe/Berlin
      - BIRDSONGS_FOLDER=/config/BirdSongs
      - LIVESTREAM_BOOT_ENABLED=false
      - MQTT_DISABLED=true
      - SSL=false
      - CERTFILE=fullchain.pem
      - KEYFILE=privkey.pem
    volumes:
      - /volume1/docker/birdnet-pi/config:/config
      - /volume1/docker/birdnet-pi/media:/media
      - /volume1/docker/birdnet-pi/share:/share
      - /volume1/docker/birdnet-pi/ssl:/ssl
      - /dev/shm:/dev/shm  # Shared memory
      - /volume1/docker/birdnet-pi/nginx/:/etc/nginx/servers
    tmpfs:
      - /tmp
```
I just have a problem. When I restart the container, it looks like the container is running in the log. But I can't get it on the web interface. I have now tried to restart about 10 times. With re-pull and without a re-pull of the image. It didn't work 9 times.
Here is a log where it didn't work:

[birdnet-pi_logs.txt](https://github.com/user-attachments/files/18981243/birdnet-pi_logs.txt)

Maybe you can see a problem in the log that you could correct.

One more question. Where do I see when there is a new version?

alexbelgium, 2025-02-26:
Hi, I'll write a full guide soon indeed this will be needed. So to understand clearly, it works when doing a clean stop and pulling the image when starting (all scripts run) but not when restarting (only the run script runs)?

Bongo2, 2025-02-26:
@alexbelgium 
Thanks for the reply.
I've tested again now. I have not been able to reproduce the missing web interface.

But there were other mistakes.
Here is the log after Stop Start:

[_birdnet-pi_logs stop-start Container 1.txt](https://github.com/user-attachments/files/18987684/_birdnet-pi_logs.stop-start.Container.1.txt)

This message comes a few times:
File:/Home/Pi/BirdSongs/StreamData/2025-02-26-BirdNet-RTSP_1-15: 18: 01.WAV: No Such File Or Directory
But then everything seems to work.

Then Stop Start again:

[_birdnet-pi_logs stop-start Container 2.txt](https://github.com/user-attachments/files/18987695/_birdnet-pi_logs.stop-start.Container.2.txt)
This message comes:
```
2025-02-26T14:31:14.030890597Z Starting service: avahi daemon
2025-02-26T14:31:14.035284130Z Found user 'avahi' (UID 104) and group 'avahi' (GID 106).
2025-02-26T14:31:14.035555594Z Successfully dropped root privileges.
2025-02-26T14:31:14.036340723Z avahi-daemon 0.8 starting up.
2025-02-26T14:31:14.037230124Z dbus_bus_get_private(): Failed to connect to socket /run/dbus/system_bus_socket: Connection refused
2025-02-26T14:31:14.037355985Z WARNING: Failed to contact D-Bus daemon.
2025-02-26T14:31:14.037400256Z avahi-daemon 0.8 exiting.
```

Then a Restart:

[_birdnet-pi_logs Restart Container 3.txt](https://github.com/user-attachments/files/18987723/_birdnet-pi_logs.Restart.Container.3.txt)

The same as before.

Then an update stack. The compose file is started in portainer with a stack:

[_birdnet-pi_logs Update Stack 4.txt](https://github.com/user-attachments/files/18987746/_birdnet-pi_logs.Update.Stack.4.txt)

The same as before.

Then a Stop Start stack:

[_birdnet-pi_logs Stop-Start Stack 5.txt](https://github.com/user-attachments/files/18987759/_birdnet-pi_logs.Stop-Start.Stack.5.txt)

Now everything is running as it should be.

I have the impression that everything is not properly ended or started at the stop or start. At Stop start it is very unreliable. But if it started properly, things go very well.

I hope you can find something from the logs that you can improve.

alexbelgium, 2025-02-26:
Thanks! I don't have much time now but I'll look to improve it soon

alexbelgium, 2025-03-02:
Hi all I've written a guide but am not a super expert of docker compose so please don't hesitate to improve it

milch, 2025-03-02:
I used a very similar setup and can confirm it works

Bongo2, 2025-03-05:
@alexbelgium 
Thanks for the documentation.

I have the version 2025.02.23 on my Synology DS1522+. As described above, it has difficulties. But when it works it works.

Now I see there is a new version 2025.03.13. Before I install it on my DS1522+, I wanted to test it on my DS918+. But there are error messages here. Here is my log:

[_birdnet-pi_logs(1).txt](https://github.com/user-attachments/files/19090496/_birdnet-pi_logs.1.txt)

My friend has a DS916+. It doesn't work either.
Could it be that there is a problem with older hardware? Just like here:
https://github.com/alexbelgium/haassio-addons/issues/1798
My DS1522+ has an AMD processor.
My DS918+ an Intel Celeron processor.

alexbelgium, 2025-03-05:
Indeed, there can be 2 potential issues : 
- indeed I see "Your cpu doesn't support avx2" -> therefore it tries to install Tensorflow instead of Tflite. I'm not 100% sure why sometimes it works or not. It worked according to this thread : https://github.com/Nachtzuster/BirdNET-Pi/discussions/111 but as I'm lucky enough NOT to have a cpu without avx2 I can't really investigate why...
- However, I also see `[aac @ 0x559024c7f0c0] channel element 2.14 is not allocated 2025-03-05T13:10:07.586022219Z Error while decoding stream #0:1: Invalid data found when processing input` : which indicates that your rtps feed using aac visibly is not working. Could it be that it is not accessible for your synology ?

DS1522+ has as cpu a ryzen 1600 with avx2 -> working with tflite
DS918+ has as cpu a Intel Celeron J3455 without avx2
DS916+ has a cpu a Intel Pentium N3710 without avx2

Bongo2, 2025-03-05:
>     * indeed I see "Your cpu doesn't support avx2" -> therefore it tries to install Tensorflow instead of Tflite. I'm not 100% sure why sometimes it works or not. It worked according to this thread : [Install BirdNET-Pi to Proxmox LXC container Nachtzuster/BirdNET-Pi#111](https://github.com/Nachtzuster/BirdNET-Pi/discussions/111) but as I'm lucky enough NOT to have a cpu without avx2 I can't really investigate why...

I tried the code. But I think that doesn't work in Docker. The code is for VM.

> * However, I also see `[aac @ 0x559024c7f0c0] channel element 2.14 is not allocated 2025-03-05T13:10:07.586022219Z Error while decoding stream #0:1: Invalid data found when processing input` : which [...]

I can ping the camera in the Docker container.
I also see the message on my DS1522+. There are many bird recognitions here.

[image]

> Edit : I have access to a non-avx2 system I'll do some tests

It's great. Thank you.

[21 more replies]