mcguirepr89/BirdNET-Pi #218: BirdNET-Pi stops processing/analyzing at midnight

mcguirepr89, 2022-04-23:
### Discussed in https://github.com/mcguirepr89/BirdNET-Pi/discussions/217

<div type='discussions-op-text'>

<sup>Originally posted by **pentode1** April 23, 2022</sup>
I installed BirdNET-Pi a few days ago on a 3B+ and it's working great other than that it stops analyzing data at exactly midnight (last analyzed sample is from a few seconds before midnight)

Everything else seems to work, the web interface works, the web terminal works... the streaming audio works so I presume all the sound subsystem and recording is working too.  Even with a reboot, though, it still doesn't process any data.  The only method I've found to resurrect it so far is to wipe all the data, which I'm guessing isn't the way it's intended to be used ;)

In fact if I shut down before midnight and restart in the morning all the data remains intact and functions as expected-  it's only if it's running _at_ midnight, or so it appears.

Anyone have any ideas what's going on?  Cheers, great project!</div>

mcguirepr89, 2022-04-23:
Hi, @Pentode1

Thank you for your kind words about the project. I'm really sorry you're seeing a midnight bug ~~ and you're right, clearing all data isn't the intended strategy to get from one day to the next :)

If I were to guess, I would thing this may have something to do with timezone settings? But that is a guess at this point.

The logs, however, will tell us what we need to know about what is going on. 

If you could please provide the output of the following command, it should help describe the issue from the system's perspective:

This command will output the log info for the `birdnet_analysis`, `birdnet_server`, `birdnet_recording`, and the `extraction` services (the core services) from 11:58pm to 12:02am so that I can take a look at what is going on. If there isn't enough information for this, we can increase some verbosity in the logs and check again tonight.

```
date && # To check your system time against the actual time -- if this is set to UTC, we've found our culprit
journalctl --no-hostname --no-pager --since="2022-04-22 23:58:00" --until="2022-04-23 00:02:00" -ubirdnet_analysis -ubirdnet_server -ubirdnet_recording -uextraction -u
```

Quick follow up question: Did you set the system time using the Raspberry Pi Imager as described in the [Installation Guide](https://github.com/mcguirepr89/BirdNET-Pi/wiki/Installation-Guide)?

My best regards,
Patrick

pentode1, 2022-04-23:
Much obliged for the quick help!  Answer to 2nd question first-  I believe I followed the installation guide precisely... it's possible I set the time zone in raspi-config though, can't recall.  I did start by following the guide though.   Once again sorry about the code formatting.

`04-23 00:02:00" -ubirdnet_analysis -ubirdnet_server -ubirdnet_recording -uextraction
Sat Apr 23 10:34:39 ADT 2022
-- Journal begins at Mon 2022-04-04 11:41:41 ADT, ends at Sat 2022-04-23 10:34:39 ADT. --
Apr 22 23:58:00 birdnet_analysis.sh[239938]: 0.0;3.0-('Scolopax minor_American Woodcock', 0.027448025)
Apr 22 23:58:00 birdnet_analysis.sh[239938]: 3.0;6.0-('Scolopax minor_American Woodcock', 0.025703466)
Apr 22 23:58:00 birdnet_analysis.sh[239938]: 6.0;9.0-('Scolopax minor_American Woodcock', 0.02373872)
Apr 22 23:58:00 birdnet_analysis.sh[239938]: 9.0;12.0-('Scolopax minor_American Woodcock', 0.020475946)
Apr 22 23:58:00 birdnet_analysis.sh[239938]: 12.0;15.0-('Scolopax minor_American Woodcock', 0.025214082)
Apr 22 23:58:00 systemd[1]: birdnet_analysis.service: Succeeded.
Apr 22 23:58:00 systemd[1]: birdnet_analysis.service: Consumed 4.312s CPU time.
Apr 22 23:58:02 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 1217.
Apr 22 23:58:02 systemd[1]: Stopped BirdNET Analysis.
Apr 22 23:58:02 systemd[1]: birdnet_analysis.service: Consumed 4.312s CPU time.
Apr 22 23:58:02 systemd[1]: Started BirdNET Analysis.
Apr 22 23:58:03 birdnet_analysis.sh[239978]: Files loaded
Apr 22 23:58:07 birdnet_analysis.sh[239978]: RECORDING_LENGTH set to 15
Apr 22 23:58:11 birdnet_analysis.sh[239978]: analyze.py --i /home/sierra/BirdSongs/April-2022/22-Friday/2022-04-22-birdnet-23:57:55.wav --o /home/sierra/BirdSongs/April-2022/22-Friday/2022-04-22-birdnet-23:57:55.wav.csv --lat 46.0878 --lon -64.7782 --week 16 --overlap 0.0 --sensitivity 1 --min_conf 0.55
Apr 22 23:58:12 server.py[367]: ANALYZING AUDIO... DONE! Time 3.2 SECONDS
Apr 22 23:58:12 server.py[367]: WRITING RESULTS TO /home/sierra/BirdSongs/April-2022/22-Friday/2022-04-22-birdnet-23:57:40.wav.csv ... DONE! WROTE 0 RESULTS.
Apr 22 23:58:12 server.py[367]: [NEW CONNECTION] ('127.0.0.1', 57714) connected.
Apr 22 23:58:12 server.py[367]: [ACTIVE CONNECTIONS] 1
Apr 22 23:58:12 server.py[367]: READING AUDIO DATA... DONE! READ 5 CHUNKS.
Apr 22 23:58:12 server.py[367]: FULL FILENAME: -/home/sierra/BirdSongs/April-2022/22-Friday/2022-04-22-birdnet-23:57:55.wav-
Apr 22 23:58:12 server.py[367]: Date-time: 2022-04-22 23:57:55
Apr 22 23:58:15 birdnet_analysis.sh[240122]: 0.0;3.0-('Scolopax minor_American Woodcock', 0.021221774)
Apr 22 23:58:15 birdnet_analysis.sh[240122]: 3.0;6.0-('Scolopax minor_American Woodcock', 0.021289319)
Apr 22 23:58:15 birdnet_analysis.sh[240122]: 6.0;9.0-('Scolopax minor_American Woodcock', 0.023365749)
Apr 22 23:58:15 birdnet_analysis.sh[240122]: 9.0;12.0-('Scolopax minor_American Woodcock', 0.024875088)
Apr 22 23:58:15 birdnet_analysis.sh[240122]: 12.0;15.0-('Scolopax minor_American Woodcock', 0.0232173)
Apr 22 23:58:15 systemd[1]: birdnet_analysis.service: Succeeded.
Apr 22 23:58:15 systemd[1]: birdnet_analysis.service: Consumed 4.278s CPU time.
Apr 22 23:58:17 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 1218.
Apr 22 23:58:17 systemd[1]: Stopped BirdNET Analysis.
Apr 22 23:58:17 systemd[1]: birdnet_analysis.service: Consumed 4.278s CPU time.
Apr 22 23:58:17 systemd[1]: Started BirdNET Analysis.
Apr 22 23:58:18 birdnet_analysis.sh[240166]: Files loaded
Apr 22 23:58:22 birdnet_analysis.sh[240166]: RECORDING_LENGTH set to 15
Apr 22 23:58:26 birdnet_analysis.sh[240166]: analyze.py --i /home/sierra/BirdSongs/April-2022/22-Friday/2022-04-22-birdnet-23:58:10.wav --o /home/sierra/BirdSongs/April-2022/22-Friday/2022-04-22-birdnet-23:58:10.wav.csv --lat 46.0878 --lon -64.7782 --week 16 --overlap 0.0 --sensitivity 1 --min_conf 0.55
Apr 22 23:58:27 server.py[367]: ANALYZING AUDIO... DONE! Time 3.2 SECONDS
Apr 22 23:58:27 server.py[367]: WRITING RESULTS TO /home/sierra/BirdSongs/April-2022/22-Friday/2022-04-22-birdnet-23:57:55.wav.csv ... DONE! WROTE 0 RESULTS.
Apr 22 23:58:27 server.py[367]: [NEW CONNECTION] ('127.0.0.1', 57716) connected.
Apr 22 23:58:27 server.py[367]: [ACTIVE CONNECTIONS] 1
Apr 22 23:58:27 server.py[367]: READING AUDIO DATA... DONE! READ 5 CHUNKS.
Apr 22 23:58:27 server.py[367]: FULL FILENAME: -/home/sierra/BirdSongs/April-2022/22-Friday/2022-04-22-birdnet-23:58:10.wav-
Apr 22 23:58:27 server.py[367]: Date-time: 2022-04-22 23:58:10
Apr 22 23:58:30 birdnet_analysis.sh[240307]: 0.0;3.0-('Scolopax minor_American Woodcock', 0.020421924)
Apr 22 23:58:30 birdnet_analysis.sh[240307]: 3.0;6.0-('Scolopax minor_American Woodcock', 0.023533132)
Apr 22 23:58:30 birdnet_analysis.sh[240307]: 6.0;9.0-('Scolopax minor_American Woodcock', 0.023428125)
Apr 22 23:58:30 birdnet_analysis.sh[240307]: 9.0;12.0-('Scolopax minor_American Woodcock', 0.022610942)
Apr 22 23:58:30 birdnet_analysis.sh[240307]: 12.0;15.0-('Scolopax minor_American Woodcock', 0.020056667)
Apr 22 23:58:30 systemd[1]: birdnet_analysis.service: Succeeded.
Apr 22 23:58:30 systemd[1]: birdnet_analysis.service: Consumed 4.228s CPU time.
Apr 22 23:58:32 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 1219.
Apr 22 23:58:32 systemd[1]: Stopped BirdNET Analysis.
Apr 22 23:58:32 systemd[1]: birdnet_analysis.service: Consumed 4.228s CPU time.
Apr 22 23:58:32 systemd[1]: Started BirdNET Analysis.
[... cut ...]
Apr 23 00:01:51 systemd[1]: birdnet_analysis.service: Succeeded.
Apr 23 00:01:53 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 1244.
Apr 23 00:01:53 systemd[1]: Stopped BirdNET Analysis.
Apr 23 00:01:53 systemd[1]: Started BirdNET Analysis.
Apr 23 00:01:56 systemd[1]: birdnet_analysis.service: Succeeded.
Apr 23 00:01:58 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 1245.
Apr 23 00:01:58 systemd[1]: Stopped BirdNET Analysis.
Apr 23 00:01:58 systemd[1]: Started BirdNET Analysis.`

mcguirepr89, 2022-04-23:
Thanks for your answer and for the log output. It doesn't show very much, but still points to timestamping issues.
I have a few more questions and commands to request before increasing verbosity for tonight:

- Was the output for `date` accurate?
- `cat /etc/timezone`
- `ls -l /etc/localtime`

pentode1, 2022-04-23:
Yes, date, time and time zone are correct.
`birdnetpi:/var/log$ date
Sat Apr 23 10:46:33 ADT 2022
birdnetpi:/var/log$ cat /etc/timezone
America/Moncton
birdnetpi:/var/log$ ls -l /etc/localtime
lrwxrwxrwx 1 root root 35 Apr 18 18:43 /etc/localtime -> /usr/share/zoneinfo/America/Moncton
`

mcguirepr89, 2022-04-23:
Hmmmm :thinking: thank you again for confirming those things.

We'll need to increase verbosity on the `birdnet_analysis` service and let it fail again tonight so that we can see more closely what is going on.

If you would be so kind as to make a small edit in the `$HOME/BirdNET-Pi/scripts/birdnet_analysis.sh`, we will have much more to work with.

Near the very top of the script is a commented out line that currently looks like this:
```
#!/usr/bin/env bash
# Runs BirdNET-Lite
#set -x
source /etc/birdnet/birdnet.conf
{. . .} 
```
if you could remove the `#` on line 3, that will set the debugging to "on" and you should notice VERY verbose logging in your "View Logs" page very soon thereafter.

Thank you for your help tracking this down -- it is sincerely appreciated.

pentode1, 2022-04-23:
Sincerely glad to help and very grateful for your help as well.  I've made that edit.  
My remaining question for now is:  given that it's not working now, I would think I'm going to have to wipe it and get it running again to get anything meaningful out of the logs, correct?  I don't suppose there's a simple way to back up the data and either reintegrate it later or just keep it for posterity?  It's not important at all, just curious what my options are.

mcguirepr89, 2022-04-23:
Before Clearing all data, you can try first to "Restart ALL Services" at the very bottom in "Tools" > "Services." Be very patient with this, as it does take at least 90 seconds to complete, sometimes more (rarely less).

If thing aren't working after the service restart, then you may need to "Clear ALL Data." If you want to backup the data, however, you can simply:

```
cp ~/BirdNET-Pi/scripts/birds.db{,.backup}
cp ~/BirdSongs{,.backup}
```
then you can run "Clear ALL Data" and will still have a copy of all of your extractions and a copy of your database from which you can restore.

mcguirepr89, 2022-04-23:
Worst case scenario, we don't find anything over night tonight, in which case I will run an installation tomorrow using your timezone and will try to replicate the problem. If I can make it break this same way, I'll be able to find the solution and will be able to get your system working. Not a terrible worst case scenario

pentode1, 2022-04-23:
I've tried to restart all services on both occasions where this happened-  it makes no difference, nor does rebooting.  

Thanks for the tips on backing up the data...  just getting to grips with the basics of how this system is working. ;)

mcguirepr89, 2022-04-23:
Next time you restart all services, please send me the output it presents just so I can be sure there are no errors therein (if you get the chance).

mcguirepr89, 2022-04-23:
Actually, if things are still failing, and restarting services is failing, and the verbosity has increased, I should be able to see whatever the error is, right? No reason to wait until midnight if it's broken now? Send whatever log output it is creating now and I will find out why it is currently broken (if you'd like)

pentode1, 2022-04-23:
> Actually, if things are still failing, and restarting services is failing, and the verbosity has increased, I should be able to see whatever the error is, right? No reason to wait until midnight if [...]

Oops, I had already wiped it when this came in.  I'll be keeping a closer eye on the logs etc now.  What I can confirm is that after wiping data, it just starts working properly again.

mcguirepr89, 2022-04-23:
No worries, those are good things for me to know. If anything of interest happens between now and midnight, just let me know. Otherwise, I'll check back on this tomorrow and we'll see what is the matter.

My best

pentode1, 2022-04-23:
Once again, the help is very much appreciated.  We've already shared this project with a couple dozen people and will continue to promote it.  Cheers!

mattvlr, 2022-04-23:
I mentioned the time zones in another issue thread as a solution, but I had this happen to me again yesterday. 

I tried: 
-Restarting the pi
-changing system time
-restarted local network
-restarting services from web interface
-changing sensitivity / time record length

it only seems to occur whenever the system time on the pi turns over to midnight.

It’s not really a fix, but I was able to get it to start analyzing again once I used the “wipe all data” option from the admin section.

I’ve been reporting to birdweather so I didn’t feel too bad about wiping my local database. But maybe backup your recordings if you don’t want to lose them in this process.

Thanks,
-Matt

ehpersonal38, 2022-04-23:
> I mentioned the time zones in another issue thread as a solution, but I had this happen to me again yesterday.
>
> I tried: -Restarting the pi -changing system time -restarted local network -restarting [...]

Very strange, 3 people have reported this same issue in the last few days. I'll be curious to see what the solution ends up being. Hopefully @mcguirepr89 can solve this mystery!

mcguirepr89, 2022-04-23:
Yeah, it's weird, and a little tricky to test in a timely fashion. I suppose we'll know more tomorrow?
If anyone else having this problem would care to troubleshoot this issue, please just follow the recommendation in [this comment](https://github.com/mcguirepr89/BirdNET-Pi/issues/218#issuecomment-1107475722) and we'll have more data in the morning.

The command we'll be using anytime after 12:05am will be:
```
journalctl --no-hostname --no-pager --since="$(date +%F --date="yesterday") 23:55:00" --until="$(date +%F --date="today") 00:05:00" -ubirdnet_analysis -ubirdnet_recording
```

Can anyone seeing that issue supply me with that command now? I added the `birdnet_recording` service log output, too, since there is a chance that there is an issue there with new files being created, which would cause the `birdnet_analysis` service to fail to analyze files, but actually succeed as far as systemd is concerned, which we can see in non-verbose log output. I'm leaning towards this being the issue, but again, it's just another guess! Luckily, the `birdnet_recording` services is always super verbose, so we'll have output for that (if there is an error) without needing to enable debugging in the script.

Thank you in advance to anyone able to supply that output.

mcguirepr89, 2022-04-23:
>Yeah, it's weird, and a little tricky to test in a timely fashion. I suppose we'll know more tomorrow?

:thinking: :thinking: I suppose it's almost midnight somewhere??? Anywhere UTC/GMT +6 hours as of the time of this writing. I think I missed the boat on trying things out for that timezone, but I should know something in an hour with UTC/GMT +5.

Wish me luck

ehpersonal38, 2022-04-23:
> Update
I spun up a new installation that is approaching midnight in twenty minutes or so -- we'll see what it does -- coffee break ☕

I'll grab my popcorn!

mcguirepr89, 2022-04-23:
Not able to replicate on my end :/

[image]

I'll try another installation, this time with my true timezone, and will run it overnight and we'll see if I can replicate things that way.

SlavKarach, 2022-04-23:
4 people :) I installed yesterday, just checked, last analyzing was 15 second before midnight...

mcguirepr89, 2022-04-23:
Hi, @SlavKarach -- perhaps you would be willing to run the prescribed troubleshooting command?
```
journalctl --no-hostname --no-pager --since="$(date +%F --date="yesterday") 23:55:00" --until="$(date +%F --date="today") 00:05:00" -ubirdnet_analysis -ubirdnet_recording | sed -e '/CADDY/d;/LATITUDE/d;/LONGITUDE/d;s/\/home\/.*\/Bird/\/home\/user\/Bird/g'
```

SlavKarach, 2022-04-24:
@mcguirepr89 

 ~/BirdNET-Pi$ journalctl --no-hostname --no-pager --since="$(date +%F --date="yesterday") 23:55:00" --until="$(date +%F --date="today") 00:05:00" -ubirdnet_analysis -ubirdnet_recording | sed -e '/CADDY/d;/LATITUDE/d;/LONGITUDE/d;s/\/home\/.*\/Bird/\/home\/user\/Bird/g'
-- Journal begins at Mon 2022-04-04 10:41:41 EDT, ends at Sat 2022-04-23 20:16:30 EDT. --
Apr 22 23:55:02 birdnet_analysis.sh[176436]: analyze.py --i /home/user/BirdSongs/April-2022/22-Friday/2022-04-22-birdnet-23:54:45.wav.csv --lat 34.1246 --lon -84.2406 --week 16 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
Apr 22 23:55:03 birdnet_analysis.sh[176602]: 0.0;3.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
Apr 22 23:55:03 birdnet_analysis.sh[176602]: 3.0;6.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
Apr 22 23:55:03 birdnet_analysis.sh[176602]: 6.0;9.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
Apr 22 23:55:03 birdnet_analysis.sh[176602]: 9.0;12.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
Apr 22 23:55:03 birdnet_analysis.sh[176602]: 12.0;15.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
Apr 22 23:55:03 systemd[1]: birdnet_analysis.service: Succeeded.
Apr 22 23:55:03 systemd[1]: birdnet_analysis.service: Consumed 2.581s CPU time.
Apr 22 23:55:06 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 1058.
Apr 22 23:55:06 systemd[1]: Stopped BirdNET Analysis.
Apr 22 23:55:06 systemd[1]: birdnet_analysis.service: Consumed 2.581s CPU time.
Apr 22 23:55:06 systemd[1]: Started BirdNET Analysis.
Apr 22 23:55:07 birdnet_analysis.sh[176615]: Files loaded
Apr 22 23:55:10 birdnet_analysis.sh[176615]: RECORDING_LENGTH set to 15
Apr 22 23:55:16 birdnet_analysis.sh[176615]: analyze.py --i /home/user/BirdSongs/April-2022/22-Friday/2022-04-22-birdnet-23:55:01.wav.csv --lat 34.1246 --lon -84.2406 --week 16 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
Apr 22 23:55:18 birdnet_analysis.sh[176760]: 0.0;3.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
Apr 22 23:55:18 birdnet_analysis.sh[176760]: 3.0;6.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
Apr 22 23:55:18 birdnet_analysis.sh[176760]: 6.0;9.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
Apr 22 23:55:18 birdnet_analysis.sh[176760]: 9.0;12.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
Apr 22 23:55:18 birdnet_analysis.sh[176760]: 12.0;15.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
Apr 22 23:55:18 systemd[1]: birdnet_analysis.service: Succeeded.
Apr 22 23:55:18 systemd[1]: birdnet_analysis.service: Consumed 2.567s CPU time.
Apr 22 23:55:20 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 1059.
Apr 22 23:55:20 systemd[1]: Stopped BirdNET Analysis.
Apr 22 23:55:20 systemd[1]: birdnet_analysis.service: Consumed 2.567s CPU time.
Apr 22 23:55:20 systemd[1]: Started BirdNET Analysis.
Apr 22 23:55:21 birdnet_analysis.sh[176779]: Files loaded
Apr 22 23:55:25 birdnet_analysis.sh[176779]: RECORDING_LENGTH set to 15
Apr 22 23:55:31 birdnet_analysis.sh[176779]: analyze.py --i /home/user/BirdSongs/April-2022/22-Friday/2022-04-22-birdnet-23:55:15.wav.csv --lat 34.1246 --lon -84.2406 --week 16 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
Apr 22 23:55:33 birdnet_analysis.sh[176915]: 0.0;3.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
Apr 22 23:55:33 birdnet_analysis.sh[176915]: 3.0;6.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
Apr 22 23:55:33 birdnet_analysis.sh[176915]: 6.0;9.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
Apr 22 23:55:33 birdnet_analysis.sh[176915]: 9.0;12.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
Apr 22 23:55:33 birdnet_analysis.sh[176915]: 12.0;15.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
Apr 22 23:55:33 systemd[1]: birdnet_analysis.service: Succeeded.
Apr 22 23:55:33 systemd[1]: birdnet_analysis.service: Consumed 2.712s CPU time.
Apr 22 23:55:35 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 1060.
Apr 22 23:55:35 systemd[1]: Stopped BirdNET Analysis.
Apr 22 23:55:35 systemd[1]: birdnet_analysis.service: Consumed 2.712s CPU time.
Apr 22 23:55:35 systemd[1]: Started BirdNET Analysis.
Apr 22 23:55:36 birdnet_analysis.sh[176939]: Files loaded
Apr 22 23:55:40 birdnet_analysis.sh[176939]: RECORDING_LENGTH set to 15
Apr 22 23:55:46 birdnet_analysis.sh[176939]: analyze.py --i /home/user/BirdSongs/April-2022/22-Friday/2022-04-22-birdnet-23:55:30.wav.csv --lat 34.1246 --lon -84.2406 --week 16 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
Apr 22 23:55:48 birdnet_analysis.sh[177074]: 0.0;3.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
Apr 22 23:55:48 birdnet_analysis.sh[177074]: 3.0;6.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
Apr 22 23:55:48 birdnet_analysis.sh[177074]: 6.0;9.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
Apr 22 23:55:48 birdnet_analysis.sh[177074]: 9.0;12.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
Apr 22 23:55:48 birdnet_analysis.sh[177074]: 12.0;15.0-('Abeillia abeillei_Emerald-chinned Hummingbird', nan)
Apr 22 23:55:48 systemd[1]: birdnet_analysis.service: Succeeded.
Apr 22 23:55:48 systemd[1]: birdnet_analysis.service: Consumed 2.605s CPU time.
Apr 22 23:55:50 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 1061.
Apr 22 23:55:50 systemd[1]: Stopped BirdNET Analysis.
[... cut ...]
Apr 23 00:04:50 systemd[1]: birdnet_analysis.service: Succeeded.
Apr 23 00:04:52 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 1136.
Apr 23 00:04:52 systemd[1]: Stopped BirdNET Analysis.
Apr 23 00:04:52 systemd[1]: Started BirdNET Analysis.
Apr 23 00:04:55 systemd[1]: birdnet_analysis.service: Succeeded.
Apr 23 00:04:57 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 1137.
Apr 23 00:04:57 systemd[1]: Stopped BirdNET Analysis.
Apr 23 00:04:57 systemd[1]: Started BirdNET Analysis.

ryankrage77, 2022-04-24:
I've run into the same (or a very similar) issue. I shut down my pi around 7pm as the battery was near depleted. I did some testing of the microphone inside around 9pm, everything worked fine. I then re-deployed the pi once the battery had recharged at around 2am, at which point the spectogram would no longer update, and the birdnet analysis service stopped outputting information. My log looks basically identical to the latter half of SlavKarach's.
The Pi was not powered on when midnight came and went.

The output of the journalctl command (copy-pasted straight from github) is just
```
-- Journal begins at Mon 2022-04-04 15:41:41 BST, ends at Sun 2022-04-24 03:23:07 BST. --
-- No entries --
```
but that could well be because I had tried rebooting the Pi several times before finding this issue.

mcguirepr89, 2022-04-24:
Hello, all --
I was hopeful my second test installation would allow me to replicate the issue, but I'm afraid this is the second installation that has smoothly transitioned from one day to the next.
I was also hopeful that I would wake up to others' verbose log output to aid in troubleshooting the issue they're seeing, but I don't have that either.

I hope that someone seeing the problem has taken me up on the offer to look through their logs after increasing verbosity. If so, I will look forward to looking through those logs. If not, I can't really continue to help as I'm unable to recreate the issue.

If anyone at all is seeing this issue and can provide the following, it would be very helpful:
- **_ANY AND ALL_** additional installation / post-installation steps taken that deviate from [the installation guide](https://github.com/mcguirepr89/BirdNET-Pi/wiki/Installation-Guide)
- Run ```locale```

mcguirepr89, 2022-04-24:
##HAHAHA
It happened to my system!!! :laughing:

ehpersonal38, 2022-04-24:
> ##HAHAHA It happened to my system!!! 😆

The plot thickens!! *munches popcorn

mcguirepr89, 2022-04-24:
Yeah! I'll finally be able to do some meaningful digging -- I'll let you know what I find!

mcguirepr89, 2022-04-24:
Okay I've fixed it -- I'll let you all know when to run an update -- don't clear all data, as this fix will post-process stuff that's backlogged

mcguirepr89, 2022-04-24:
The issue was that there were corrupted recordings in my directory from yesterday (0 byte audio files without headers). This is likely because my cat unplugged the system several times yesterday. There are likely corrupt audio files in the 4 installations that have reported this issue (again, my clean installation from yesterday that my cat was NOT allowed to unplug is still working without issue).

`birdnet_analysis.service` found those corrupt audio files (it always checks yesterday before processing today) and kept waiting for their length to reach the defined `RECORDING_LENGTH` (15 seconds) to process them. It's default is to check the file's length, and it if has not reached the `RECORDING_LENGTH`, it sleeps for 1 second and then comes back. So, this is the infinite success loop we've seen in the logs that have been shared thus far.

To fix this, I have the service check the file, and if it has no header, it deletes it, assuming that the file itself is corrupt since it would always have a header otherwise.

After putting the fix in place, my system is post-processing the backlog since midnight. After it has caught up and I have ensured that this fix has with it no unintended consequences, I will push the changes and will close this issue.

Thank you all for your help, though no logs will be needed from anyone at this point.

My best regards as always,
Patrick

ehpersonal38, 2022-04-24:
> The issue was that there were corrupted recordings in my directory from yesterday (0 byte audio files without headers). This is likely because my cat unplugged the system several times yesterday. [...]

That does make sense! Although it's interesting, because I've assuredly unplugged my Pi before a few times mid-operation and never ran into this issue. In any case I'm glad to see this issue getting (hopefully) resolved! Good work everyone & thanks to Patrick for all the effort he's put in! 😄

mcguirepr89, 2022-04-24:
>Although it's interesting, because I've assuredly unplugged my Pi before a few times mid-operation and never ran into this issue. 

Corrupting those audio files is a strange/nuanced thing -- they have to be truncated _before_ the header is added, but after the stream has moved onto the next chunk, otherwise they're just truncated audio files, and the header is able to handle the file being a bit shorter than planned. Without the header, however, it's a meaningless file that can't be used. Also, those files are made from a continuous ALSA stream, so the headers are added onto the stream to start new files, but the stream isn't allocating file space prior to writing the file, it's just cutting the stream based on the header and then moves onto the next header. I don't know enough to say for certain, but I can imagine this can cause trouble when the system is preparing a pointer to the storage inode and then when trying to allocate the space, notices there is no header to indicate the start and end of the file, yet the file itself exists.

It truly is a rare thing to happen, but when folks are getting their systems "setup" and moving them around from place to place, this is when those things can happen the easiest.

SlavKarach, 2022-04-24:
Reporting that my system ran without an issue overnight.

SlavKarach, 2022-04-24:
The only difference I could observe is that the first time I didn't plug in a mic until some hours after install and setting it up. I had just ordered it and my Pi came a day before the mic.

mcguirepr89, 2022-04-24:
Thanks @SlavKarach -- This helps confirm what I observed as well. I have not been able to replicate the issue while trying, but my cat unplugged my other installation a few times yesterday and was able to help us out be causing the issue on that installation. After a bit more testing (and [when my system is finished with the backlog of data](https://virginia.birdnetpi.com/log)), I will push the change and you should be able to do an update and not have to worry about it happening again in the future. 

Your system running smoothly after getting established (no moving around, no plugging/unplugging mics) makes sense, so thank you for letting us know!

pentode1, 2022-04-24:
Just for completeness, my system kept running past midnight last night.  I had done a git pull and update and then a data wipe- I'm not sure which if any of those things might have made a difference in my install/issue.

It may also be worth mentioning that a couple days before this issue, I had the exact same symptoms, but mid-afternoon, when someone unaware of my installation flipped a power switch a few times.  Same result-  the system seemed to work perfectly other than it stopped processing.  I assumed it was data corruption, wiped the data, then things went back to normal.

I'm guessing this patch will prevent that sort of problem in the future as well.

Many thanks to @mcguirepr89 for working so diligently at this and for the quick solution and to everyone else in the thread for their contributions as well!

saintaardvark, 2022-04-24:
Just to chime in: I'd come across this issue as well; it resolved before I could troubleshoot it, but re-appeared last night.  This morning I saw *mainly* the same symptoms others had reported: web UI responsive but no new detections showing up, and the BirdNET Analysis log giving endless sequences of "succeeded".  (The one thing that was different was the last spectrogram displayed had a timestamp of 16:09, rather than close to 23:59; I suspect I had an unrelated crash or restart on the Pi around that time.)

I was able to resolve this on the Pi by logging in and running:

```shell
$ cd ~/BirdSongs/April-2022
$ find . -type f -size 0 -ls
   531493      0 -rw-r--r--   1 pi       pi              0 Apr 23 16:09 ./23-Saturday/2022-04-23-birdnet-16:09:46.wav
   531551      0 -rw-r--r--   1 pi       pi              0 Apr 20 19:41 ./20-Wednesday/2022-04-20-birdnet-19:41:30.wav
   531485      0 -rw-r--r--   1 pi       pi              0 Apr 20 19:39 ./20-Wednesday/2022-04-20-birdnet-19:39:06.wav
   531483      0 -rw-r--r--   1 pi       pi              0 Apr 20 07:05 ./20-Wednesday/2022-04-20-birdnet-15:05:08.wav
$ find . -type f -size 0 -delete
```

After that, the logs showed that it was going through the backlog of recorded samples, and the "Currently analyzing" spectrogram on the UI is reasonably recent (~ 30 minutes old as I type this, with the time updating regularly).

@mcguirepr89 I'm wondering if adding something like the `find` command to delete zero-length files would be a good addition to the analysis script.  If you'd like a PR, I'm happy to submit one; alternately, if you already have a solution in mind, I'm happy to test it out.

mcguirepr89, 2022-04-24:
@saintaardvark we think alike!! `find` is where I've decided to solve this. I'll just let 0 byte corrupted files live peacefully on the system and let `find` be the one to ignore them :)

Tentatively, this is all that needs to change in `birdnet_analysis.sh` to fix this bug -- but I'm also optimizing it while making the change cuz why not
```
 get_files() {
 files=($( find ${1} -maxdepth 1 -name '*wav' ! -size 0\
    | sort \
    | awk -F "/" '{print $NF}' ))
    [ -n "${files[1]}" ] && echo "Files loaded"
}
```

adding the `! -size 0` ignores corrupted files (they'll get a header if they're non-zero). Thank you for the great suggestion!

mcguirepr89, 2022-04-24:
I went ahead and pushed that fixing change and will save optimization for another time.

Thanks everyone!

Please update via "Tools" > "System Controls" > "Update" to get the fix.
My best regards,
Patrick
Thank you all for your thanks, as it is reciprocated -- THANKS

closed 2022-04-24 by mcguirepr89

pentode1, 2022-04-25:
Sorry to open this back up again but my system stopped again at midnight last night and I wanted to ask what I did to prevent the fix from being installed on my system.  I ran the update - there was a message about everything being current but I didn't think much of it.  Did I mess things up with the git pull I did before the fix was done? 
I used @saintaardvark 's solution and it seems to be catching up on last night's samples now.

ehpersonal38, 2022-04-25:
> Sorry to open this back up again but my system stopped again at midnight last night and I wanted to ask what I did to prevent the fix from being installed on my system. I ran the update - there was a message about everything being current but I didn't think much of it. Did I mess things up with the git pull I did before the fix was done? I used @saintaardvark 's solution and it seems to be catching up on last night's samples now.
> 
> edit: oh I should probably add that my system crashed yesterday for some unknown reason and I ended up with 1 file of zero size. This is what clued me in that the patch didn't get installed for me with the update function

A `git pull` shouldn't have messed anything up. Can you verify the file manually to see if it truly is updated as per this commit? https://github.com/mcguirepr89/BirdNET-Pi/commit/a8fb4c79a6574bd0272fff1f0b59ce2b2777330e

mcguirepr89, 2022-04-25:
@pentode1 no need to apologize -- I'm sorry you're seeing this issue again.

@ehpersonal38's request is helpful, as the update should have ignored any 0 byte (corrupted) files and verifying that your system has the updated script is essential.

@pentode1 we'll need to figure out why your system is regularly corrupting audio files. Faulty hardware is my one and only guess.

Let us know whether line 51 reads:
```
   files=($( find ${1} -maxdepth 1 -name '*wav' ! -size 0\
```

mcguirepr89, 2022-04-25:
For what it may be woth:
@pentode1 I wonder if the most recent failing was because the `birdnet_analysis.service` was still using the old script when post-processing the data (since you said you actually got it working by manually removing the files, not by updating and then restarting the service).

If that is why you were able to observe the issue again, even after updating the problematic script, that would explain things. To be sure, once the system has "caught up" and is analyzing files close to the current time, run "Tools" > "Services" > "Restart ALL Services" to be sure that your `birdnet_analysis.service` is using the updated `birdnet_analysis.sh` script.

It's that none of the above is applicable, but there is a chance that what I've described happened. It has happened to me many times during development where I will update a script but forget to `sudo systemctl daemon-reload` or `sudo systemctl restart theservice` and get confused when I don't see any changes in the logs.

pentode1, 2022-04-25:
I think there are at least 2 separate issues going on with my setup.  I think the weird interface behavior may just be down to heavy cpu loads-  I'm getting the occasional "database is busy" message in place of certain elements.  The heavy cpu loads might also be exacerbating my power supply issues.  I really pay attention to power supply on the pi and normally I power them directly from the 5V rail, but in this case I'm using a 2A microusb supply, so I'll work on that end of things to make sure the power supply is stable.  I suppose that could explain file corruption too, although I've never had any issues like this in the past.

Your response just came in and I will definitely do as you suggested.  It's 110% possible I screwed something up, no question :)

As I pull all this info together I'll try to get to the bottom of what these other issues are and if I think I have any meaningful info, I'll post it here or some other more relevant place.  

Thanks as always!

Mwisho, 2022-05-09:
I had the same problem after unplugging my Raspberry Pi yesterday, after updating everyting is working as expected again.
Thanks Patrick!