mcguirepr89/BirdNET-Pi #219: Ability to have a flexible design on the daily graph

Svardsten53, 2022-04-24:
I have discussed this before, but would like to have some changes made in daily_plot.py that creates the daily graph. It is possible to change the value "readings = 10" to a higher value, but then the graph looks very strange and compressed because the size of the box is not flexible. It would be good if somebody with python skills could modify the code so that the height of the graph adapts to the number of species that should actually be listed (not what is in the variable "readings"). It would also be good if we could get back the opportunity to list totals for each species.

I have never really understood the idea of only listing the top 10 when it is the more unusual visits at the bottom of the list that are the most interesting.

My idea is to spread BirdNET-Pi to a wider circle in my country, such as bird observatories, and I know that on the overview page they want a quick overview of all the day's detections, not only the most common.

mcguirepr89, 2022-04-24:
@CaiusX I've assigned this to you if you feel like creating a customized overview for Lars. If you would prefer staying focused on Streamlit, that is 100% perfect with me, as that is the direction we should continue to move towards for data visualization and development.

Anyone else willing to put together a customized view for Lars, please let him know here!

My best!

ehpersonal38, 2022-04-24:
> It is possible to change the value "readings = 10" to a higher value, but then the graph looks very strange and compressed because the size of the box is not flexible. It would be good if somebody [...]

Something like this?
```
#Set up plot axes and titles
f, axs = plt.subplots(1, 2, figsize = (10, round(readings/2)), gridspec_kw=dict(width_ratios=[3, 6]), facecolor='#77C487')
```

CaiusX, 2022-04-25:
Hi @Svardsten53 

Hopefully the attached "old skool" code works for you . Rename as daily_plot.py and pop into your scripts directory.

[daily_plot.txt](https://github.com/mcguirepr89/BirdNET-Pi/files/8554973/daily_plot.txt)

You'll have to fiddle with lines 39-41 if you want to fix the number of species & comment out lines 80-92 if you don't want (what I regard as ugly) value on bars

CaiusX, 2022-04-26:
Hi @Svardsten53 - have you had a chance to test the above?  I've had it running on http://birds.naturestation.net for a day without a hitch.

Svardsten53, 2022-04-26:
Thank you for helping me with this. Problem solved! The file you attached was even better than the old one I'm using on my other system. The titles now have the same distance to the graph regardless of the number of species listed.

I had to change the path since we now not are using user "pi" any longer. I also discovered that if you create a new file, it must have execute permission to run. The easiest way is to just copy the new text to the existing file, which already has execute permission.

closed 2022-04-26 by Svardsten53

CaiusX, 2022-04-27:
Hi @Svardsten53 - excellent news, my main system is still running an old OS version, will make sure I'm up to date before I send anything out in future.

Do you think it worth putting your update in a separate branch on GIthub?

Best
AP