mcguirepr89/BirdNET-Pi #220: Cannot view webgui

danmed, 2022-04-24:
So i've finally got this set up to try the RTSP hack posted in another issue.. and i can see it consuming the stream via journalctl -f but i cannot access the gui via the IP address...

It's just a blank page... Any suggestions?

mcguirepr89, 2022-04-24:
Have you tried [http://birdnetpi.local](url) or http://whateveryourhostnameis.local?

danmed, 2022-04-24:
> Have you tried [http://birdnetpi.local](url) or http://whateveryourhostnameis.local?

Thankyou .. that does indeed work.. however.. not ideal for my setup.. how do i make it respond correctly to the IP?

mcguirepr89, 2022-04-24:
It is responding correctly -- as a security feature, `caddy` (the webserver) only responds to addresses configured in its address block, which you can see in "Tools" > "System Info" near the bottom.

By default, the installation creates an mDNS announcement on the network to allow anyone to reach their installation at http://birdnetpi.local (or whatever they set the hostname as)

In "Tools" > "Settings" > "Advanced Settings" you will see the section for the Custom URL an and example there on exactly how to make your installation respond (equally correctly) to http://my.ip.address.octet

Let me know if that doesn't work for you and please let me know if any of that is unclear.

My best,
Patrick

closed 2022-04-24 by mcguirepr89