mcguirepr89/BirdNET-Pi #223: BirdNET analyses running "empty"

thomasgibon, 2022-04-25:
BirdNET ran smoothly on my Raspberry Pi 3B for the past two days. However after restarting it today, it seems the analysis service is not able to pick up the recordings and just runs indefinitely without processing anything (see below).

What is strange is that I can see the .wav files being created, but there is no spectrogram or anything else happening after that. I have tried stopping and restarting all services, to no avail. Any idea of what is happening?

```23:14:34 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 254.
23:14:34 systemd[1]: Stopped BirdNET Analysis.
23:14:34 systemd[1]: Started BirdNET Analysis.
23:14:35 systemd[1]: birdnet_analysis.service: Succeeded.
23:14:38 systemd[1]: birdnet_analysis.service: Scheduled restart job, restart counter is at 255.
23:14:38 systemd[1]: Stopped BirdNET Analysis.
23:14:38 systemd[1]: Started BirdNET Analysis.```

ehpersonal38, 2022-04-25:
Hi @thomasgibon,

This could be related to an issue that was just recently fixed and closed a day or so ago. I'd recommend this:

1. Run `git pull`
2. In the web gui, go to tools > system controls > update

This should apply the latest fixes. Report back if the issue still persists!

artmann23, 2022-04-26:
Hello,

I assume I have the same problem.
My BirdNET-Pi stopped processing right before midnight today and seems to be stucked at the last file covering the daychange (xxx23:59:53.wav).
I ran git pull and triggered the update from the web interface, but with no success. 
Any tips?

nilspupils, 2022-04-26:
@artmann23: Did you reboot after updating? (Please not directly but after a couple of minutes?)

thomasgibon, 2022-04-26:
Thanks @ehpersonal38, I will try it when I'm at home later today – will keep you in touch!

artmann23, 2022-04-26:
> @artmann23: Did you reboot after updating? (Please not directly but after a couple of minutes?)

Yes I did reboot after the update, but maybe a bit quick. ~1...2mins after update finished succesfully.

ehpersonal38, 2022-04-26:
> Hello,
> 
> thanks @ehpersonal38, its working again! i restarted the update (git pull + Tools/System Control/Update + _wait_ + reboot), but with no success. Then I located the "stucked file", which had indeed 0-bytes filesize. After deletion of the file, BirdNET-Pi immediately started to analyse the next file.
> 
> Thanks for your help!

Great! Although I'm sure this will happen again and it'd be best if there was a global fix - it seems like the update isn't doing what it was supposed to.

thomasgibon, 2022-04-26:
I confirm that the following

- `git pull`
- Tools/System Control/Update
- removing the 0 byte files

solved the issue, thanks a lot! I didn't even need to reboot.

I also confirm I have the `-size 0` flag in `birdnet_analysis.sh`.

I lost my birds.db database in the process (my bad), but I kept all processed recordings (in BirdSongs/Processed), is there any way I can rerun the analysis on these past files? (I can file a new issue for that)

ehpersonal38, 2022-04-26:
> I lost my birds.db database in the process (my bad), but I kept all processed recordings (in BirdSongs/Processed), is there any way I can rerun the analysis on these past files? (I can file a new issue for that)

A quick and dirty way would just be to move all the .wavs from BirdSongs/Processed into BirdsSongs/. The script should pick them up and analyze them like normal

thomasgibon, 2022-04-26:
> move all the .wavs from BirdSongs/Processed into BirdsSongs/

It doesn't seem to pick the files up, do they need to be in today's folder? Again, not a big deal, I'm just happy that the whole service works again, thanks a lot for your responsiveness!

artmann23, 2022-04-26:
My output for 
sed -n 45,55p $(which birdnet_analysis.sh)

```

fi

# Create an array of the audio files
# Takes one argument:
#   - {DIRECTORY}
get_files() {
  files=($( find ${1} -maxdepth 1 -name '*wav' ! -size 0\
  | sort \
  | awk -F "/" '{print $NF}' ))
  [ -n "${files[1]}" ] && echo "Files loaded"
}

```

ehpersonal38, 2022-04-26:
> > move all the .wavs from BirdSongs/Processed into BirdsSongs/
> 
> It doesn't seem to pick the files up, do they need to be in today's folder? Again, not a big deal, I'm just happy that the whole service works again, thanks a lot for your responsiveness!

I just tested it and it works for me, in the root BirdSongs folder. Kind of strange that it's not working for you. In view log are there any errors? You're welcome for the responsiveness!

thomasgibon, 2022-04-26:
The root folder worked, I was being too diligent and put each batch back into their timestamped folder. Awesome, thanks! (would it be a good idea to add these few tricks to the FAQ/Wiki?)

ehpersonal38, 2022-04-26:
> The root folder worked, I was being too diligent and put each batch back into their timestamped folder. Awesome, thanks!

Excellent!

> (would it be a good idea to add these few tricks to the FAQ/Wiki?)

Probably, I'm sure someone could find it useful. Patrick did kind of mention this in an older thread: https://github.com/mcguirepr89/BirdNET-Pi/discussions/118#discussioncomment-2089097

I'd be interested in seeing that feature built out more - also maybe a script that automatically backs up birds.db! (I've deleted mine twice since first installing it a few weeks ago >:(

ehpersonal38, 2022-05-02:
Fixed from PR #229, commit 78b1dced396198d019399fb98fc8a308cca5e824.

closed 2022-05-02 by ehpersonal38