mcguirepr89/BirdNET-Pi #225: Bit the bullet a bit too early on a mic

danmed, 2022-04-26:
Seeing as how well the RTSP works (especially with the positioning of my source) i went ahead and ordered a USB microphone.. assuming i can just plug it in to the USB port of the pi... I'm still waiting for it to arrive.

I've since done some more reading and there is talk of sound cards etc..

So.. is the one i've bought going to work at all? It's just a simple clip on USB microphone... I know it wont be the best sound.. but is it actually going to function as a Mic or is a USB soundcard required for a Mic to work on the Pi?

nilspupils, 2022-04-26:
Hi @danmed, 
it could work, but it has to be a USB-mic which is supported by Linux. Was there any info about it?

danmed, 2022-04-26:
ooh i'm guessing it's not going to work then...

https://www.amazon.co.uk/gp/product/B08RP88ZHW/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

nilspupils, 2022-04-26:
It doesn't mention Linux so i guess it will not work.

danmed, 2022-04-26:
OK thanks,

i've cancelled it and ordered one that mentions it works on ubuntu in the
reviews.

ehpersonal38, 2022-04-26:
I am using a mic just like that for my Pi! (although about $25) Actually in the amazon listing it doesn't say anything about linux, just "PC, desktop, and Mac." I figure these things are somewhat universal as long as they don't have specific drivers, but what do I know. 

Here's an audio sample, the bird was probably 40ft away (12m):
https://user-images.githubusercontent.com/103586016/165357998-42d65fec-16b6-45b8-8dbd-6a8a5c1e8434.mp4

danmed, 2022-04-26:
> I am using a mic just like that for my Pi! (although about $25) Actually in the amazon listing it doesn't say anything about linux, just "PC, desktop, and Mac." I figure these things are somewhat [...]

Hey man, don't suppose you have a link to the one yo ubought so i can find a similar one?

ehpersonal38, 2022-04-26:
> Hey man, don't suppose you have a link to the one yo ubought so i can find a similar one?

Here you go!
https://smile.amazon.com/dp/B0176NRE1G

Kestrelio, 2022-04-26:
For what it's worth I'm using an ancient Logitech headset and it's doing the job well, no additional configuration required. I'd wager any Logitech mic works out of the box.

closed 2022-04-28 by ehpersonal38