mcguirepr89/BirdNET-Pi #226: Notifications on platforms beside iOS

ryankrage77, 2022-04-28:
I see birdnet supports pushed.co notification on iOS, but I'm not in the Apple ecosystem so I can't take advantage of this feature.

I took a look at species_notifier.sh, it seems it should be fairly simple to send notifications to other services. For example, you could send a message to a telegram chat with
`$ curl -s -X POST https://api.telegram.org/bot[TOKENID]/sendMessage -d chat_id=[ID] -d text=${NOTIFICATION}`
(example taken from here, https://www.marcodena.it/post/send-a-telegram-notification-when-a-script-is-done/). This would also allow for multiple BirdNet-Pi's to all send messages to the same chat, which would be great for running multiple systems or collaborating with other people.

However, I'm not really clear on how this script interacts with the web UI, so I'm not confident on submitting a pull request just yet. Presumably birdnet.conf could be modified to work with a custom script without breaking anything. I'll tinker with my birdnet this evening (it's currently running outside and I don't want to disturb it) and see what I can come up with.

Additionally, browser notifications via the web UI would be platform agnostic. However, my proficiency stops at kludging simple bash scripts, so I'll let someone else assess the feasability of that.

ehpersonal38, 2022-04-28:
Just adding my 2c here, I'm using http://notify.run for push notifications.

CaiusX, 2022-04-28:
Hi @ryankrage77 & @ehpersonal38 

It would be great if you could collaborate on getting a non-IOS based push feature tested in a separate branch and pulled into main when working!!

Best 
AP

ryankrage77, 2022-04-28:
I know I said I didn't want to disturb the Pi, but I did anyway.

I modified the part that actually sends the notifications in species_notifier.sh like so:
```
if [ ! -z ${PUSHED_APP_KEY} ];then
    #send a notif
    curl -s -X POST https://api.telegram.org/bot[mybotID]/sendMessage -d chat_id="[mychatID]" -d text=${NOTIFICATION}
  fi
fi
```
literally just swapping out the curl command. They keys are in the script, but I've redacted them here.

I then filled in the pushed.co API keys on the web UI as just 'a', so it would detect something and enable the service. When I restarted the service I got a message 'New' from the bot, so that part works at least.

However, new detections don't trigger a notification. Nothing shows in the log to hint at whether it tried to send it & failed.
Perhaps there's some extra login elsewhere checking the API keys I entered via the web UI?

[image]

ehpersonal38, 2022-04-28:
> However, new detections don't trigger a notification. Nothing shows in the log to hint at whether it tried to send it & failed.
Perhaps there's some extra logic elsewhere checking the API keys I entered via the web UI?

AFAIK species_notifier.sh only triggers when there's a new species, not a new detection.

ryankrage77, 2022-04-28:
Ah that explains it, my Pi's already been running for a while so it's not likely to find any new species today.
It looks like this is cumulative and doesn't reset at any point?

ehpersonal38, 2022-04-28:
Okay I've added browser notifications via notify.run to a new test branch "notifications," https://github.com/mcguirepr89/BirdNET-Pi/commit/a9893c519ba6aa362699443f028a0004b4ac99d9

ehpersonal38, 2022-04-28:
> Ah that explains it, my Pi's already been running for a while so it's not likely to find any new species today. It looks like this is cumulative and doesn't reset at any point?

Yeah I'm not exactly how species_notifier works, I don't think it resets though

If you want notifications on new detection you could do something like this in scripts/server.py (haven't tested):
[image]

ryankrage77, 2022-04-29:
> you could do something like this in scripts/server.py

Managed to get this working, eventually. Took me hours to notice I was editing server.py when I was actually using privacy_server.py - but once I edited the correct file & restarted services, the notifications starting flowing.

mcguirepr89, 2022-05-07:
Closing as the direction seems to be towards `apprise`
see issue #228

closed 2022-05-07 by mcguirepr89