mcguirepr89/BirdNET-Pi #228: Feature Request: apprise for notifications

VictorioBerra, 2022-04-28:
Currently only Pushed.co notifications are supported for IOs users, can we have support for this project? https://github.com/caronc/apprise

This would allow anyone to have any sort of notification they want.

CaiusX, 2022-04-29:
Hi @VictorioBerra 

I've never used the notification feature but I'm sure there's a real use case (tell me when species xx is around, tell me when a rare specie is around, etc).

Looks like a neat system to incorporate, maybe hook up with @ehpersonal38  and @ryankrage77 here https://github.com/mcguirepr89/BirdNET-Pi/issues/226#issue-1218547828 and take this forward?

ehpersonal38, 2022-05-05:
@mcguirepr89 I've been putting this off since I'm not sure the best way to add `pip install apprise` into the installation/updating flow, any pointers?

mcguirepr89, 2022-05-05:
It can safely be added to `requirements.txt` and then it will be installed with the system.

To have it added during an update will require a little snippet along these lines:
```
apprise_installation_status=$(~/BirdNET-Pi/birdnet/bin/python3 -c 'import pkgutil; print("installed" if pkgutil.find_loader("apprise") else "not installed")')
if [[ "$apprise_installation_status" = "not installed" ]];then
  ~/BirdNET-Pi/birdnet/bin/pip3 install -U pip
  ~/BirdNET-Pi/birdnet/bin/pip3 install apprise
fi
```

Now, we've just been adding snippet type stuff like this to `update_birdnet.sh`, but I think it may make better sense to have a second file that `update_birdnet.sh` always calls, keeping the sole function of `update_birdnet.sh` just update the repo files through `git`. The second file will do things like adding new variables to users' preexisting `birdnet.conf` files, adding `apt` and `pip` packages to the system, and installing any new services/tools outside of `git`'s duties.

SOOO, we can add that snippet to another file, along with the other little non-`git` snippets, so that we can avoid being forced into the second option, which is appending the snippet to the growing group of snippets at the end of `update_birdnet.sh` and then requiring users run the update a second time to execute the new `update_birdnet.sh` code.

Let me know what you think about that stuff and we'll figure the best way to move forward ~~

mcguirepr89, 2022-05-06:
So what is the verdict on notifications? Should we be shooting for `apprise` or `https://notify.run` for notifications???

VictorioBerra, 2022-05-06:
@mcguirepr89 Those are drastically different are they not?

Take a look at https://github.com/dgtlmoon/changedetection.io#notifications

They leverage appraise to make notification trivial, this way, you have the ability to get notified by ANYTHING in this list https://github.com/caronc/apprise#popular-notification-services

This is a huge win for a one-stop notification support in your app. You drop this in, everyone can configure notifications via tons of different services. I would imagine it would work like this:

[image]

(or just put it in settings)

And then the notification settings are would look like this:

[image]

mcguirepr89, 2022-05-06:
Super cool! I ask because I'm impartial and will not use these notifications, so I just really want to know what folks want to see implemented and not weigh in myself. After looking through the repo, though, it seems `apprise` is certainly the way to go. Thanks for the great suggestion and I'll keep you updated on progress integrating this (and finally getting rid of Pushed.co [even though it will still be available via `apprise`!])

ehpersonal38, 2022-05-06:
Wow, I really feel for the apprise development team. Having to manage integrations and take bug reports for dozens of different services 😬

Jokes aside, thank you for the above screenshots^^ I'll probably work on this soon

mcguirepr89, 2022-05-06:
@ehpersonal38  should I delete the current `notifications` branch that was around notify.run? Or will you use that?

ehpersonal38, 2022-05-06:
> @ehpersonal38 should I delete the current `notifications` branch that was around notify.run? Or will you use that?

Keep please - it has all the framework there for integrating notifications. I'll just need to switch it out with apprise related things.

Also is there an easy way to modify `species_notifier.sh` to also be able to notify on new detection also? I was thinking about just adding something to server.py but I assume there's a reason you made a separate service for species notification and didn't do it in server.py?

mcguirepr89, 2022-05-06:
>Also is there an easy way to modify `species_notifier.sh` to also be able to notify on new detection also? I was thinking about just adding something to `server.py` but I assume there's a reason you made a separate service for species notification and didn't do it in `server.py`?

The simple answer is that I wasn't able to figure out how to implement the notifications into Python when I wanted them, but I knew how to do it in bash, so I did.

But `server.py` somewhere near where it makes the database entry is where the `apprise` notification object stuff should go if you want notifications for every detection.

`species_notifier.sh` just queries the database for distinct species into a tmp tile and compares that to the file `IdentifiedSoFar.txt`. If there are updates, it formats the `diff` as a notification, sends that off with `curl`, and updates `IdentifiedSoFar.txt` with the contents of the tmp file.

One note on notifications for new detections: some installations would definitely hit their API's rate limit at some point. My Pi0W2 installation, for instance, would have sent 676 notifications today. Probably not a big deal, though.

ehpersonal38, 2022-05-06:
Thanks for the helpful pointers! I'm definitely more comfortable implementing it through python (and I [already did for my personal installation](https://github.com/mcguirepr89/BirdNET-Pi/issues/226#issuecomment-1112166151))

> One note on notifications for new detections: some installations would definitely hit their API's rate limit at some point. My Pi0W2 installation, for instance, would have sent 676 notifications today. Probably not a big deal, though.

And my installation would be over 4k some days! Definitely need some kind of "only notify every 5 minutes," or a digest every so often that compiles a periodic list of the latest detections.

mcguirepr89, 2022-05-06:
>And my installation would be over 4k some days! Definitely need some kind of "only notify every 5 minutes," or a digest every so often that compiles a periodic list of the latest detections. (especially for email, I imagine sending hundreds of emails every day wouldn't sit well with most ISPs)

This is why I liked systemd for `species_notifier.sh` because it's easy to adjust the `templates/pushed_notifications.service`'s `RestartSec=`. So if you wrote a little python script to handle notifications, it can take the place of `species_notifier.sh` and you can have set the `RestartSec=300` and you'll get your notifications every 5 minutes. The good thing about having systemd handle it, too, is that it will not be silly about the service restart and will consider resources before triggering the next service cycle, in case it wants to send the notification right when Streamlit is accessing the database and building its data view, the Chart viewer is updating the data in the Overview page, the system is analyzing sound and POSTing to BirdWeather, blah blah blah. It will see when resources are available and then will restart, so it will be every 5 minutes (unless it needs to wait briefly for its turn).

ehpersonal38, 2022-05-07:
Just finished implementing Apprise for species notifications in the `notifications` branch. I haven't tested it fully but it should work... next is to do for each detection

mcguirepr89, 2022-05-08:
@ehpersonal38 it's looking really good -- we'll run it through some tests really soon :1st_place_medal:

ehpersonal38, 2022-05-08:
> @ehpersonal38 it's looking really good -- we'll run it through some tests really soon 🥇

Great! I just finished a bunch of work on it, check it out when you get a chance. Everything seems to be working. except you do have to restart the analysis server after checking or unchecking "Notify each new detection" in the config -- I need to update "thisrun.txt" more frequently in server.py to catch that, right now it only reads that file at the beginning.

mcguirepr89, 2022-05-08:
If you want to clean up your database from those testing results, you should be able to go to "Tools" > "Database Maintenance", then "SQL Command" on the left:

enter this:
```
select * from detections where Com_Name in (
select Com_Name
FROM detections
GROUP BY Com_Name
having MAX(Confidence) < .2 )
```

If that looks like what you want to get rid of, replace "select *" with "delete":
```
delete from detections where Com_Name in (
select Com_Name
FROM detections
GROUP BY Com_Name
having MAX(Confidence) < .2 )
```

Hope that helps :)

ehpersonal38, 2022-05-08:
Thanks Patrick! I ended up doing a similar command, just conf < 0.2. I have a custom confidence exception for crows so they'll still be picked up even at a min confidence of 0.1-- (surprisingly accurate actually, I've only had a few false positives in thousands of crow detection's at this conf- the model can be a little strange)

VictorioBerra, 2022-05-10:
Thank you so much for this!