mcguirepr89/BirdNET-Pi #231: Will this work on a RPi 3B (not +)

robintw, 2022-04-30:
I'm interested in building this project, but I only have a Raspberry Pi 3B, not a 3B+. Pi's are in very short supply at the moment, so I can't easily get a 3B+.

Is there any reason this wouldn't work on a 3B? I notice the requirements say a 3B+, but apart from a slightly increased CPU speed, and some networking improvements, I can't see a massive difference in specs.

See [here](https://www.androidcentral.com/raspberry-pi-3-model-b-vs-3-b) for a specs comparison.

nilspupils, 2022-04-30:
It has not been tested on a Pi 3B so far, and i have the feeling it might not work, but it would be great if you would try it and report back! Happy testing!

clsheppard, 2022-05-01:
Pi 3B was not sufficient.

I just happened to try an install on Pi 3B v1.2 this morning.  The install did not fail, but the webserver was blank.  Moved the same install to a Pi 3B+ (no changes to the install) and the web portal came up as expected.

danmed, 2022-05-01:
I have it running on a 3B but it does hang up atleast once a day resulting in needing a reboot.. This likely explains it

bdoner, 2022-05-03:
I've had an installation running on a Pi 3B for a few days, with no hangs. I've enabled a large swap partition and is actively cooling with a 12V fan on the GPIO pins.

The blank page problem I suspect is due to configurations issues and not the PI's specs.

My setup is currently not high-volume when it comes to recordings, so maybe a higher load gives me problems.

mcguirepr89, 2022-05-03:
It may be worth setting the Recording Length to 30 seconds and maybe add a larger swap space to mitigate some of the hardware constraints. You can see what changes are needed to [install on a Raspberry Pi 0W2](https://github.com/mcguirepr89/BirdNET-Pi/wiki/RPi0W2%20Installation%20Guide), which may be helpful for you all here.

My best,
Patrick

mcguirepr89, 2022-05-03:
>I am wary about how it'll be behaving when summer hits and the temperatures goes up by 20-30 degrees.

Me too -- we're the vanguard for these tests, so we'll see!