mcguirepr89/BirdNET-Pi #233: Cannot Access birdnetpi.local from windows 10 pc

beyerl, 2022-05-01:
At first: Thanks for this really cool project! It was fun building it and its working quite well.

This is not so much a request for help but rather a documentation of a problem I ran into and a solution for it to help other people. Since submitting an issue is mentioned under "Troubleshooting and Ideas", I hope this is the right place for it.

# Problem:
After finishing the installation, the local birdnetpi client website was not opening in my browser. When I used "birdnetpi.local", I got "page cannot be found", when I instead used the IP adress of my raspberry pi, I got an empty website (HTTP Statuscode 200 Ok, Content-Lenght 0). I could connect to my raspberry pi via ssh. Using "curl localhost" inside the shell returned the proper HTML-Code of the website:

```
<title>BirdNET-Pi</title>
<meta name="viewport" content="width=device-width, initial-scale=1">
<style>
body::-webkit-scrollbar {
  display:none
}
</style>
<link rel="stylesheet" href="style.css">
<body>
<div class="banner">
  <div class="logo">
<a href="https://github.com/mcguirepr89/BirdNET-Pi.git" target="_blank"><img src="images/bird.png"></a>  </div>
  <div class="stream">

  <form action="" method="GET">
    <button type="submit" name="stream" value="play">Live Audio</button>
  </form>
  </div>
  <h1><a href="/">BirdNET-Pi</a></h1>
</div>
<iframe src="/views.php"></iframe>
```

# Reason for the problem:
The first problem is, that ".local"-DNS-Adresses often do not seem to be recognized by Windows. This means, the dns adress "http://birdnetpi.local" does not get resolved to the local IP adress of your raspberry pi. This is the reason, why "http://birdnetpi.local" returns "page not found".
The second problem is, that the caddy server, birdnetpi is using always returns "HTTP Statuscode 200 Ok, Content-Lenght 0" for all unmatched routes. [For more info check out this thread](https://github.com/caddyserver/caddy/issues/3226). So the caddy webserver can by accessed by the browser on my windows pc, if I am directly typing in the IP adress of my raspberry pi, but the route I am trying to access is not matched to any content that could be served.

# What do we have so far?
1. The webserver on my raspberry pi is working properly as demonstrated by the "curl localhost"-response describe above
2. I can access the raspberry pi via http as demonstated by the "200 OK"-response when doing "curl <ip.address.of.pi>" (replace <ip.address.of.pi> by the ip you found out with port droid)

This led me to the solution: The caddy server seemed to be misconfigured. It did not resolve the incoming http request to birdnetpi page.

# Solution:
## Find out the IP and local dns address of your raspberry pi
On your phone:
1. open up google play store
2. Search for "port droid". Your are looking for [this app](https://play.google.com/store/apps/details?id=com.stealthcopter.portdroid&hl=gsw&gl=US)
3. Install the app and open it
4. go to "local network".
5. Find your birdnetpi in the device list.
6. Note the local dns address and ip adress. The IP adress is in format x.x.x.x, eg. 192.168.154.3. The local dns adress is to the right of it. If you are using a FritzBox router, like I do, it might be something like "birdnetpi.fritz.box".

## Adjust Caddy Webserver Configuration
On your windows PC, after you have booted up your birdnetpi and it connected to the same wifi as your windwos pc:
1. ssh into your birdnetpi, e.g. 
```
ssh birdnet@<ip.address.of.pi>
```
(replace <ip.address.of.pi> by the ip you found out with port droid)

7. open up caddy server config file for editing 
```sudo nano /etc/caddy/Caddyfile``` [more info](https://pimylifeup.com/raspberry-pi-caddy-web-server/)
In my case, this yielded:
```
http://localhost http://birdnetpi.local  {
  root * /home/birdnet/BirdSongs/Extracted
  file_server browse
  handle /By_Date/* {
    file_server browse
  }
  handle /Charts/* {
    file_server browse
  }
  reverse_proxy /stream localhost:8000
  php_fastcgi unix//run/php/php7.4-fpm.sock
  reverse_proxy /log* localhost:8080
  reverse_proxy /stats* localhost:8501
  reverse_proxy /terminal* localhost:8888
}
```
Now we see the problem: I was accessing the raspberry pi wvia it's local ip adress, but caddy server was not configured to do anything with it.

8. Change the caddy config:
```
http://birdnetpi.fritz.box http://localhost http://birdnetpi.local  {
  root * /home/birdnet/BirdSongs/Extracted
  file_server browse
  handle /By_Date/* {
    file_server browse
  }
  handle /Charts/* {
    file_server browse
  }
  reverse_proxy /stream localhost:8000
  php_fastcgi unix//run/php/php7.4-fpm.sock
  reverse_proxy /log* localhost:8080
  reverse_proxy /stats* localhost:8501
  reverse_proxy /terminal* localhost:8888
}
```
This is just an example. Replace http://birdnetpi.fritz.box with the DNS you found out via port droid.

9. Save file with Ctrl-O
10. Close file with Ctrl-X
11. Reboot birdnetpi with "sudo reboot"
12. Open http://<local-dns-of-your-birdnetpi> in your browser (e.g. http://birdnetpi.fritz.box)
And your done!

mcguirepr89, 2022-05-01:
Hi, @beyerl -- Woah! Great troubleshooting!
I'm taking some time away from BirdNET-Pi, but needed to check on something this morning and saw this post and wanted to address it as well, though it sounds like you were able to find a solution on your own.

Another option would have been:

1. SSH into the BirdNET-Pi
2. Get its IPv4 address ```hostname -I``` or ```ip a```
3. Set your `BIRDNETPI_URL` in `birdnet.conf` to those results, example:
   ```
   BIRDNETPI_URL=http://192.168.1.120
   ```
4. Then run ```sudo update_caddyfile.sh```

Then your installation is accessible at the IP address.

OR another option:
1. [Adjust your Window's `/etc/hosts` file](https://www.groovypost.com/howto/edit-hosts-file-windows-10/) to have birdnetpi.local point to the BirdNET-Pi's IP address (can get the IP address using the same method as above)

Happy birding!

closed 2022-05-01 by mcguirepr89

beyerl, 2022-05-02:
Hi @mcguirepr89,
thx for getting back to me so quickly. Your solution seems a lot more elegant. Thanks for letting me know.
Cheers!