mcguirepr89/BirdNET-Pi #234: Cannot install with Verizon Fios

EricScottN, 2022-05-01:
Receiving the following error when trying to install
```
/etc/birdnet/birdnet.conf: line 13: pool-70-16-33-104.bflony.fios.verizon.net: command not found
+ rm -f /tmp/tmp.Gy7chkqJgA
+ source /etc/birdnet/birdnet.conf
The installation exited unsuccessfully.
```

ehpersonal38, 2022-05-01:
I'd guess the script doesn't like your hostname for some reason, I'd try changing it to something simpler (mine is just "birdnetpi").

EricScottN, 2022-05-01:
> I'd guess the script doesn't like your hostname for some reason, I'd try changing it to something simpler (mine is just "birdnetpi").

I'll give it a shot

ehpersonal38, 2022-05-01:
@EricScottN I just pushed something to main that will hopefully fix this. Could you try running:

`git pull`
Then running `bash newinstaller.sh` again?

EricScottN, 2022-05-01:
Upon further inspection, it seems as though the script is populating an additional line of unnecessary pooling information in the `birdnet.conf` file.

```
################################################################################
#                    Configuration settings for BirdNET-Pi                     #
################################################################################

#--------------------- Required: Latitude, and Longitude ----------------------#

## The shell substitution below guesses these based on your network. THESE NEED
## TO BE CHANGED TO STATIC VALUES
## Please only go to 4 decimal places. Example:43.3984

LATITUDE=[my lat is here but don't want to dox myself]
LONGITUDE=-[my lon in here but don't want to dox myself]
"pool-70-16-33-104.bflony.fios.verizon.net"   <------------------ I DON'T THINK THIS IS SUPPOSED TO BE HERE

#---------------------  BirdWeather Station Information -----------------------#
#_____________The variable below can be set to have your BirdNET-Pi____________#
#__________________also act as a BirdWeather listening station_________________#
```

EricScottN, 2022-05-01:
> 

@ehpersonal38 

```
pi@birdnetpi:~/BirdNET-Pi$ git pull
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint:
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint:
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
remote: Enumerating objects: 4, done.
remote: Counting objects: 100% (4/4), done.
remote: Total 4 (delta 3), reused 4 (delta 3), pack-reused 0
Unpacking objects: 100% (4/4), 379 bytes | 94.00 KiB/s, done.
From https://github.com/mcguirepr89/BirdNET-Pi
   0dd0ec3..cc3817e  main       -> origin/main
Updating 0dd0ec3..cc3817e
Fast-forward
 scripts/install_config.sh | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)
pi@birdnetpi:~/BirdNET-Pi$ bash newinstaller.sh
fatal: destination path '/home/pi/BirdNET-Pi' already exists and is not an empty directory.
The installation exited unsuccessfully.
```

EricScottN, 2022-05-01:
Should I just uninstall and reinstall?

ehpersonal38, 2022-05-01:
Oops, actually I would `rm -rf /home/pi/BirdNET-Pi" and try a fresh install

ehpersonal38, 2022-05-01:
> Upon further inspection, it seems as though the script is populating an additional line of unnecessary pooling information in the `birdnet.conf` file.
>
> ``` [...]

The install script was trying to get your longitude, looking for keyword "lon," but it just so happens that your hostname had the word "lon" in it too, so it pulled that too and messed everything up

EricScottN, 2022-05-01:
> > Upon further inspection, it seems as though the script is populating an additional line of unnecessary pooling information in the `birdnet.conf` file.
> > ```
> > ################################################################################
> > #                    Configuration settings for BirdNET-Pi                     #
> > ################################################################################
> > 
> > #--------------------- Required: Latitude, and Longitude ----------------------#
> > 
> > ## The shell substitution below guesses these based on your network. THESE NEED
> > ## TO BE CHANGED TO STATIC VALUES
> > ## Please only go to 4 decimal places. Example:43.3984
> > 
> > LATITUDE=[my lat is here but don't want to dox myself]
> > LONGITUDE=-[my lon in here but don't want to dox myself]
> > "pool-70-16-33-104.bflony.fios.verizon.net"   <------------------ I DON'T THINK THIS IS SUPPOSED TO BE HERE
> > 
> > #---------------------  BirdWeather Station Information -----------------------#
> > #_____________The variable below can be set to have your BirdNET-Pi____________#
> > #__________________also act as a BirdWeather listening station_________________#
> > ```
> 
> The install script was trying to get your longitude, looking for keyword "lon," but it just so happens that your hostname had the word "lon" in it too, so it pulled that too and messed everything up (@mcguirepr89 I bet you never thought this would happen :)). Should be working now!

All set! What a crazy coincidence :P

closed 2022-05-01 by EricScottN